dresden-elektronik/deconz-rest-plugin #3430: Conbee II disappearing

kmplngj, 2020-10-17:
## Describe the question or issue you are having
I experience similar behavior mentioned in other issues #3410 #3341: After a while of stable operation (mostly 20-30 hours), the Conbee II stick became unstable and drops out of the system. Sometimes it only to reappear some seconds later on a new USB Bus location, sometimes I have to fight with unplugging and rebooting my ESXi host.

I tried with and without USB cable, and different USB ports.

I also tried binding the Conbee II as suggested by @SwoopX in his reply to issue #3341. Does not helped.

And to finally make the issue a question: Any hints on how to fix this?

I really need to get this fixed. It is a bit difficult for my wife and me with our little daughter without light. :/

## Screenshots

## Environment

 - Host system: PC 
 - Running method:  Marthoc Docker container / Virtual Machine
 - Firmware version: deCONZ_ConBeeII_0x26660700.bin.GCF
 - deCONZ version: 2.05.84 / 2.05.85
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs

## Additional context

manutoky, 2020-10-17:
Same here. Running on a Odroid C2 (ARM64) and Hassio.

Smanar, 2020-10-17:
You have 2 problems on your log.

- a disconnection problem.
- a segfault error

For the first one, have you tried to update the firmware on a real machine ? (even the same you already have). Can you install deconz on windows for exemple to test the hardware ?

For the second one, I can't say if the problem is from deconz
```
[  361.496272] wmiir[2171]: segfault at 0 ip 00007fd602f30181 sp 00007fff845b2f08 error 4 in libc-2.28.so[7fd602df6000+148000]
[  361.496281] Code: 84 00 00 00 00 00 0f 1f 00 31 c0 c5 f8 77 c3 66 2e 0f 1f 84 00 00 00 00 00 89 f9 48 89 fa c5 f9 ef c0 83 e1 3f 83 f9 20 77 1f <c5> fd 74 0f c5 fd d7 c1 85 c0 0f 85 df 00 00 00 48 83 c7 20 83 e1
[ 2171.284742] wmiir[2576]: segfault at 0 ip 00007f96463f6181 sp 00007ffea23aaff8 error 4 in libc-2.28.so[7f96462bc000+148000]
[ 2171.284752] Code: 84 00 00 00 00 00 0f 1f 00 31 c0 c5 f8 77 c3 66 2e 0f 1f 84 00 00 00 00 00 89 f9 48 89 fa c5 f9 ef c0 83 e1 3f 83 f9 20 77 1f <c5> fd 74 0f c5 fd d7 c1 85 c0 0f 85 df 00 00 00 48 83 c7 20 83 e1
```

kmplngj, 2020-10-17:
I only other hardware I have are Macs. Windows 10 is also only a VM.

Smanar, 2020-10-17:
No real machine ? except Macs ? even not an old raspberry ?

kmplngj, 2020-10-17:
I always updated with docker without problems. I will have a look of a friend have a raspberry pi for me.

Could the firmware be defect but working for more than a day?

Smanar, 2020-10-17:
Honestly, on my side, this kind of bug generally happen all the time, not after 24h. But it's realy the first test (check) to do
And it will not solve your segmentation error.

For it,you have for exemple https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3427 but not sure you are concerned by it.

kmplngj, 2020-10-17:
I reflashed the stick on a RaspPi4 with a fresh installed Raspian and I upgraded my ESX Host to version 7.0

I will come back when I know more.

kmplngj, 2020-10-18:
Problem is still there.

Smanar, 2020-10-18:
On the second log, I don't see the segmenatation fault, but yes, you have disconnection issue, deconz working and disconnection error again.
You had enought time, to check if you still have too the problem on the raspberry ? (to see if the bug is from the hardware or the OS ?)

If you have an usb HUB , to check power supply ?

Perhaps > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3268#issuecomment-694148401

You have nothing on your OS to configure the virtual usb  port ?

kmplngj, 2020-10-20:
I tried the stick with a Pi 4 and fresh Deconz install. Same issue. 😔

Mimiix, 2020-10-21:
Can you try on the latest beta? There was a fix for a seg fault.

Smanar, 2020-10-21:
Can be the hardware too.
If you have realy updated the firmware, make try on a real machine and have the same problem on different machine. The network works for some time, so not a configuration problem.

kmplngj, 2020-10-26:
Some result with latest beta after 2 days of running.

kmplngj, 2020-10-28:
@Mimiix @Smanar @manup Anything else I could do?

Smanar, 2020-10-28:
No more idea on my side, sorry.
The segmentation error is solved,
We can't test it, but I don't think it s the hardware (20 hours without crash ...)
And for software I never use VM, so don't know them.

The only similar issue I hve found is this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3268#issuecomment-694148401 but not realy a solution.

kmplngj, 2020-10-28:
> And for software I never use VM, so don't know them.
> 
> 
> 
> The only similar issue I hve found is this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3268#issuecomment-694148401 but not realy a solution.
> 
> 
> 
> Or https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3340

I still have this issue on a fresh Pi4 without Docker/VM.

Smanar, 2020-10-29:
And it s the same error issue on the raspberry ? It switch the USB location ?
Or just the USB disconnection but after 2 days ?

Have you tried to "pin" the device ? Using `--dev=.....` in the option file (or command line for test)

kmplngj, 2020-11-01:
Yes same issue on the raspberry.

`[367382.432154] usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
[367382.571246] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[367382.571255] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[367382.571262] usb 1-1.4: Product: ConBee II
[367382.571268] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[367382.571274] usb 1-1.4: SerialNumber: DE1996969
[367382.577926] cdc_acm 1-1.4:1.0: ttyACM1: USB ACM device
[367385.963489] usb 1-1.4: USB disconnect, device number 4
[367386.272183] usb 1-1.4: new full-speed USB device number 5 using xhci_hcd
[367386.411284] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[367386.411293] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[367386.411299] usb 1-1.4: Product: ConBee II
[367386.411305] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[367386.411311] usb 1-1.4: SerialNumber: DE1996969
[367386.417953] cdc_acm 1-1.4:1.0: ttyACM1: USB ACM device
[432578.884472] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[519712.969174] usb 1-1.4: USB disconnect, device number 5
[519712.969563] cdc_acm 1-1.4:1.0: failed to set dtr/rts
[519714.909417] usb 1-1.4: new full-speed USB device number 6 using xhci_hcd
[519715.047788] usb 1-1.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[519715.047796] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[519715.047803] usb 1-1.4: Product: ConBee II
[519715.047809] usb 1-1.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[519715.047815] usb 1-1.4: SerialNumber: DE1996969
[519715.054465] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[519718.394944] usb 1-1.4: USB disconnect, device number 6
[519718.688829] usb 1-1.4: new full-speed USB device number 7 using xhci_hcd
`

closed 2020-12-19 by stale[bot]