dresden-elektronik/deconz-rest-plugin #3432: Deconz showing two devices

silent-303, 2020-10-17:
Hi there,

my deconz showing two devices (Conbee II & RaspBee) but there is only Conbee II plugged in.

When starting it takes some tome to get a connection. Any idea how i can get rid off the "false" RaspBee?

Thanks alot
[image]

silent-303, 2020-10-17:
Also asking two times for a firmware update which i confirmed already-
[image]

Conbee II has 26660700

Smanar, 2020-10-17:
I have it too, I think it s something normal.

SwoopX, 2020-10-17:
And for that reason, I'm closing this one.

closed 2020-10-17 by SwoopX