dresden-elektronik/deconz-rest-plugin #3437: PEQ 3300-P (Centralite) Not Pairing

majorsl, 2020-10-18:
## Describe the question or issue you are having

I'm new to deCONZ and the Conbee II.  I was able to add all my devices except some PEQ (Centralite) 3300-P Door/Window sensors with the Phoscon App.  Pairing with it doesn't happen, it just times out.  I'm confident I'm getting them in pairing mode as I've used them with both a Vera and playing around with the Sonoff ZbBridge flashed with Tasmota.

I might be confused with the relationship of Phoscon and deCONZ. Can I add directly with deCONZ via VNC? Might that be more successful and give more feedback?

These are the only devices I have that are not on the supported list, but I was hoping to get the info here for a supported device request.

## Environment

 - Host system: Raspberry Pi
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.05.85 / 10/15/2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes

## Additional context

Possible because I'm running on channel 25? Do some devices not support it? (I know this is true for channel 26).
These are older sensors, similar to the original Centralite 3300 or 3300-S.

SwoopX, 2020-10-18:
Have you checked https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices if it is supported?

majorsl, 2020-10-18:
> Have you checked https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices if it is supported?

Of course I did, and it isn't a supported device.

However, on this page to Request Device Support https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support it seems to indicate that a non-supported device can be included both by instructions and the included screenshots.

How else would I get the information needed to do a formal request?
How would I include a non-supported device to get this data?

majorsl, 2020-10-18:
Ah, I found another closed issue with the info I needed.  I am going to close this and open up a new issue with the information for a supported device request.

closed 2020-10-18 by majorsl