dresden-elektronik/deconz-rest-plugin #3438: Zenismart ZigBee Smart Curtain Switch ZW-EC-01

psaggiomo, 2020-10-18:
## Device

- Product name: ZigBee Smart Curtain Switch ZW-EC-01
- Manufacturer: Feibit Co.Ltd
- Model identifier: FB56+CUR17SB2.2
- Device type:
  - Switch

## Screenshots

[image]

[image]

Smanar, 2020-10-18:
And this device don't work natively ?

psaggiomo, 2020-10-18:
> And this device don't work natively ?

In Home Assistant works only Close and Stop button. Doesn't work Open button.

Smanar, 2020-10-18:
Ha yes.
You are not in this situation  > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3269 ?
Can you make the same test pls ?

psaggiomo, 2020-10-20:
Hi,
the issue is like #3269 .

The Current Position is Grayed.

[image]

psaggiomo, 2020-10-20:
- cover.close_cover: nothing
- cover.close_cover_tilt: nothing
- cover.open_cover: works
- cover.open_cover_tilt: nothing
- cover.set_cover_position: works
- cover.set_cover_tilt_position: nothins
- cover.stop_cover: works
- cover.stop_cover_tilt: nothing
- cover.toggle: works to open
- cover.toggle_cover_tilt: nothing

Smanar, 2020-10-20:
OK so your device don't support "set position" command, it just use "open" and "close ?

psaggiomo, 2020-10-20:
Yes, I think so.

Smanar, 2020-10-20:
Ok I have added it to my tuya fork (only 1 line edition) > https://github.com/Smanar/deconz-rest-plugin/commit/53b691b4030773e6e3eacebc9e4e8d566109543f

Do you have an unix machine to test it ?

psaggiomo, 2020-10-20:
I have a NAS Synology with Deconz in Docker e Home Assistant in Docker.

Smanar, 2020-10-20:
Ok so it mean "no" ? ^^

No problem, the code will be in next version I think, but without test, you can have to wait for the next one (if I m wrong)

psaggiomo, 2020-10-21:
No problem, I'll wait the next distribution to test. Thank you.

ynnovate, 2020-10-22:
hello
also interested to test and help

Smanar, 2020-10-22:
Hello, do you have an unix machine to compile the code to test it ?

It s is on my fork, tuya branch, I can give you the command to use if you have the good OS.

theslash, 2020-10-23:
Just to add here, I have a shutter controller which identifies as "Heiman | TS0302" and behaves just like this here.
I have a headless setup, where I cant control it, it just is a light where I can set the brightness.

Smanar, 2020-10-23:
@theslash and same than them, it don't work if you use "bri" or "lift" ?

theslash, 2020-10-23:
> @theslash and same than them, it don't work if you use "bri" or "lift" (even those field are present in the json) ?

Yes, nothing happens when I change any of these values.
I'm on 2.05.84 / 26660700 by the  way.

I would suspect there is some kind of calibration needed for the controller. could that be the case?

Smanar, 2020-10-23:
Ha yep, it s possible too, you haven't the device documentation ?

Some tuya device need calibration to have the set position feature working, but other don't have them at all.

Exemple https://blakadder.com/zemismart-zm25tq/ this one use a 433 mhz remote to set min/max position.

If when you use open and close, the device stop itself at good position, you have lot of chance your device don't need calibration.

If you are in the same situation than other I need you device manufacture name, the model id is useless for tuya.

theslash, 2020-10-23:
No there is only a leaflet with notes how to use the "smart life" app to control it.
It works perfectly with the buttons on the device. 
I have this device:
https://www.aliexpress.com/item/4000311591651.html?spm=a2g0s.9042311.0.0.39664c4ddQHTXC

Seems to be a tuya device if I can trust the listing.
Is it normal that it only shows up as a light in the deconz web frontend? I have a brightness slider there but it does nothing.

Smanar, 2020-10-23:
Haaa, I m so stupid ^^, haven't see the word "switch" ^^

So yes, on the switch you have Neutral (N) + phase (L) + up (L1) + down (L2) ?
So no set position command.

So yes I need the manufacture name to add your device.

You still have the "lift"/"bri" field (so still the brightness slider) but the api will convert them to "open" / "close".
And yes phsocon don't manage covering for the moment, better to use your third app.

theslash, 2020-10-23:
Ok, to get the manufacturer I need the Deconz gui, right? will have to setup this first, my setup is headless.

Smanar, 2020-10-23:
You have it too in the device JSON (for all device visible in the API)

theslash, 2020-10-23:
{
  "etag": "e1207f233b15186ee4bf15e8d550760e",
  "hascolor": false,
  "lastannounced": "2020-10-23T10:23:46Z",
  "lastseen": "2020-10-23T17:32Z",
  "manufacturername": "Heiman",
  "modelid": "TS0302",
  "name": "Window covering device 11",
  "state": {
    "bri": 127,
    "lift": 50,
    "on": true,
    "open": true,
    "reachable": true
  },
  "swversion": null,
  "type": "Window covering device",
  "uniqueid": "84:2e:14:ff:fe:fd:9f:bb-01"
}

Hope this helps.

Smanar, 2020-10-23:
Ok so bad luck, you have the bad manufacture name, not your fault, but we have this problem every 2 days.

By hazard, it s not your device ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3269

But you have a different MAC adress, if you have time, can you make a second inclusion ? and hoping for the good manufacturename.

theslash, 2020-10-23:
;-DD really? 
yes  #3269 seems to have the same problem as me and at leas the same model number.
I have a second one here, I will have to set it up and try if it differs.

theslash, 2020-10-23:
Ok, so I had a view at the other issue and PR.
If my manufacturer would be "_TYZB01_dazsid15" then this code should make it work, right:
(from the pull request : https://github.com/dresden-elektronik/deconz-rest-plugin/pull/3316/commits/3a94d2f19febaa09ceeb9456ca1e5ec58d938081)

    //Some device don't support lift, but third app can use it
    if (hasLift)
    {
        if (taskRef.lightNode->manufacturer() == QLatin1String("_TYZB01_dazsid15"))
        {
            hasLift = false;
            hasOpen = true;
            if (targetLiftZigBee > 0)
            {
                targetOpen = false;
            }
            else
            {
                targetOpen = true;
            }
        }
    }

Is there a a way to integrate this device too?

Smanar, 2020-10-24:
Yep, i just need to add your manufacture name, if you haven't the same device.

But to obtain it, you can re-include the device and hoppping to have the good one, or install deconz with the graphical environnement on another machine and switch the conbee the time to have it.

theslash, 2020-10-24:
Ok sorry, I think I don't understand what you mean.
I removed the device, and readded it, but it changed nothing.
Can I change the manufacturer of my device with the GUI?

Smanar, 2020-10-24:
No it s written on the device, but you can read it.
On cluster 0x0000 (basic cluster), attribute 0x0004, you probably need to press the button "read", and make the device awake if it s a sleeping device.

psaggiomo, 2020-11-13:
> Ok so it mean "no" ? ^^
> 
> No problem, the code will be in next version I think, but without test, you can have to wait for the next one (if I m wrong)

Hi, is the code included in v2.5.88?

Smanar, 2020-11-13:
Nope, too sensible, 88 is a stable.
All the tuya stuff will be in the beta in some day if there is no problem.

closed 2020-11-15 by manup

psaggiomo, 2020-11-16:
Hi. I have installed 2.06 and now it works!
The only problem is the current position grayed and is ever 100.
[image]

Smanar, 2020-11-16:
I think it mean the device don't support "set position".
But the returned value in websocket is always 100 too ?

psaggiomo, 2020-11-17:
> I think it mean the device don't support "set position".
> But the returned value in websocket is always 100 too ?

Yes, always 100.

Smanar, 2020-11-17:
And it cause problem on third app ?

Smanar, 2020-11-17:
Haaa yes, your device probably don't use the cluster 0x0102 for return because the attribute used 0x0008 is grayed for you.
So I think it use the cluster 0x0006, can you check the attribute value 0x0000 in the cluster 0x0006 when the device is open or closed.

The deconz code update only "on" and "open" with this cluster.

psaggiomo, 2020-11-18:
Hi, the 0x0006 il grayed also. I see only 0x0000, 0x0007, 0x0010, 0x0011, 0x0012, 0x0013 and 0x0017, but the values do not change.

Smanar, 2020-11-18:
Nope the cluster 0x0006 "on/off", it s possible your device send the notification using this cluster instead of the 0x0102 "covering", because it don't need position management.
The cluster 0x0006 is not present on other covering.

psaggiomo, 2020-11-18:
Yes. I confirm.

Smanar, 2020-11-18:
Ok so I can probably do something.

Not possible prevent the "bri"/"lift" field creation, but I can update the value in same time than "on"/"open".
BTW the "lift"/"bri" is not updated, but the "open" is or only "on" is updated ? (in the device json)

Have you something to compile and test the code ?

HB1300, 2021-12-05:
Hi there,
I also have this relay and would change the "physical_close_limit_lift_cm (id: 0x0001)" is that possible?
so that I can take into account the different lengths and thus when I go to 50% it also opens up to halfway

also the open and closed button is swapped in HA - if it is fully open I can only press the open button that close button unenabeled - can i change that?

please help - thank you

Smanar, 2021-12-06:
>I also have this relay and would change the "physical_close_limit_lift_cm (id: 0x0001)" is that possible?

Depend of device, but here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3438#issuecomment-728255112 it seem this attribute is grayed = not available.

>also the open and closed button is swapped in HA - if it is fully open I can only press the open button that close button unenabeled - can i change that?

ATM using HA, the only way is using "custom widget" (Not sure for name I m not using HA), to use custom requests.
Will be possible soon using DDF, but need to wait too HA use the last deconz version, The next deconz stable is planned around the 15 December

HB1300, 2021-12-07:
hi smanar,
Thank you for your answer
But is there a possibility to reduce the opening / closing time of this relay?
because not all curtains are the same length

HB1300, 2021-12-07:
I cannot choose this function in HA - (Time calibration for Tuya coverage - 0xf003)
Is there another way to access / choose this?

Smanar, 2021-12-08:
How you need to do from the device documentation ?
There is a calibration procedure ? A special remote ?

HB1300, 2022-01-02:
hi,
I have a new hw here I can select the cluster F001 via demoz but unfortunately not via HA can you help me here?

https://github.com/home-assistant/core/issues/46146#

and can I do a firmware update with the old hardware?

thank you for your help
best regards
harald

Smanar, 2022-01-03:
Hello
- what is demoz ?
- this device don't have cluster 0xf001 ?
- wich one hardware, gateway or device ?

HB1300, 2022-01-03:
https://githubmate.com/repo/dresden-elektronik/deconz-rest-plugin/activity
sorry ich meine deconz 

in deconz i hafe this cluster but not in HA 

HW: TS130F Gateway; HA

Smanar, 2022-01-04:
But on your link it's 0xEF00 ? Its the tuya cluster.
You don't make mistake with the attribute 0xf001 on the covering cluster ? You can use the "calibration" request using the API, but not sure this command is visible on HA.
But not all devices support this attribute, if you have the GUI, better to make a try first with deconz.

How you want to update the device firmware ? tuya give firmwares for it ?

>but even if I calibrate it in deconz - if the supply is gone, the calibration reset to the old value

Ha ? this is a new issue, the calibration procedure is working, but not keep in memory ? you have something writen on the documentation ? you don't forget to change again the calibration state ?