dresden-elektronik/deconz-rest-plugin #3440: Moes HY368 (aka Tuya) TS0601

Arquiteto, 2020-10-18:
## Describe the bug

Some thermostats work fine, some have problems reporting current temperature (usually reporting 30deg C), valve position (no updates from some valves) and changing setpoint.

I have currently 6 valves:
**0x842e14fffee5542e - Salon lewy** 
works quite well, except it often reports 30deg as current temperature

**0xbc33acfffe458efd - Salon prawy**
works quite well, except it often reports 30deg as current temperature

**0xbc33acfffe571178 - Sypialnia lewy**
no updates of valve position, stuck at 20%, setpoint working both ways (adjusting from HA, and getting updates of manually changed setpoint), also reports 30deg current temp sometimes
_Update: While writing I was playing around with TRV, I was able to get updates of valve position_

**0x842e14fffee37a93 - Sypialnia prawy**
problems with setting setpoint from HA, changing on TRV is reflected in HA, valve position was working for a moment at midnight, now no updates anymore, 30deg
_Update: While writing I was playing around with TRV, I was able to get updates of valve position, still problems with changing setpoint_

**0xbc33acfffe571fdb - Tomek**
works quite well, expect 30deg reports

**0x842e14fffee55586 - Goscinny**
works quite well, expect 30deg reports

Things I noticed:
- The valve position gets updated no more often than 5 minutes, but it worked like that on Zigbee2MQTT as well
- Changing the setpoint on the valve by few degrees is reflected in HA with 0,5deg increments/decrements every 20-40s until it reaches the temp that was set.

## Steps to reproduce the behavior

1. Connect TRV to deCONZ as sensor
2. Connect deCONZ to HA
3. Use the valves - send setpoint from HA, change setpoint on TRV, check the valves position, check current reported temperatures

## Expected behavior

Expecting no reports of 30deg as they are incorrect, proper reporting on valve positions and ability to reliably set setpoint through HA

## Screenshots

Current data from all thermostats.
Nast = Setpoint
Temp = Current reported temperature by TRV (calibrated manually, but they are about 0,5deg off, becuase calibration allows only full degrees)
Value in upper right corners of rooms = value from external thermometer
[image]

History of a valve postion that doesn't get updated (it's really 0% from last evening as the setpoint is lower). 
[image]

History of setpoint, current temp from TRV and current temp from external sensor (Salon lewy)
[image]

History of setpoint, current temp from TRV and current temp from external sensor (Tomek)
[image]

## Environment

 - Host system: QNAP NAS TS-453Be
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.05.85 / 15.10.2020
 - Device: ConBee II
 - Do you use an USB extension cable: yes

### deCONZ Logs

Container logs filtered with "grep Tuya"
https://pastebin.com/CUUVdVEE

## Additional context

Routers:
Biurko - LXN59-1S7LX1.0 (router for Sypialnia TRV's, but they connected to Conbee as well)
Kuchnia spoty - LXN59-1S7LX1.0 (router for Salon TRV's)
Tomek nocna - FNB56-ZSC01LX1.2 (router for Tomek and Goscinny TRV's)

All TRV's updated to latest firmware using Tuya gateway (v1.1.5)

If you need me to run some more specific tests, let me know. I was using those valves for a few days with Zigbee2Mqtt, but I had large lag on the lights and I was unable to see the map of devices to check the LQI so I decided to check the beta release. I will be happy to assist in improving.

Smanar, 2020-10-18:
Hello , do you have the device model id ? GbxAXL2 ?

Arquiteto, 2020-10-18:
In deCONZ basic cluster it identifies as TS0601 and it was sold as Moes HY368. Via REST API call it returns TS0601 manufactured by Heiman. Should I check somewhere else?

It seems like just this one now, it's the same device id as others. I will try to find error message when I get home.

Smanar, 2020-10-18:
Ok so I think it s this one https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3110
But yes lot of tuya device have the same model id, I realy need the good manufacture name.

And yes if you have "Heiman" as manufacture name, was not good, I think only the no working device have heiman ?  I will disable this part, boring.

BTW I have updated the code, if you can make a try with it ? IDK if you can compile it on your docker installation ?
It will prevent the 30 degré.

Arquiteto, 2020-10-18:
They all advertise the same as Heiman TS0601.
Here is return from REST API

I will try to compile the code but I'm not yet fluent with docker and git, but I think I can do it.

```
    "2": {
        "config": {
            "heatsetpoint": 1900,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true,
            "schedule": {},
            "schedule_on": null,
            "windowopen_set": null
        },
        "ep": 1,
        "etag": "882fbe863c1db852e9b8ce283b2a93ef",
        "lastseen": "2020-10-17T21:56Z",
        "manufacturername": "Heiman",
        "modelid": "TS0601",
        "name": "Sypialnia prawy",
        "state": {
            "lastupdated": "none",
            "lowbattery": false,
            "on": true,
            "temperature": 3000,
            "valve": 15
        },
        "type": "ZHAThermostat",
        "uniqueid": "84:2e:14:ff:fe:e3:7a:93-01-0201"
    },
    "3": {
        "config": {
            "heatsetpoint": 1900,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true,
            "schedule": {},
            "schedule_on": null,
            "windowopen_set": null
        },
        "ep": 1,
        "etag": "e4f07a75f1681c93694799b8f57f2c60",
        "lastseen": "2020-10-17T19:59Z",
        "manufacturername": "Heiman",
        "modelid": "TS0601",
        "name": "Sypialnia lewy",
        "state": {
            "lastupdated": "none",
            "lowbattery": false,
            "on": true,
            "temperature": 2050,
            "valve": 20
        },
        "type": "ZHAThermostat",
        "uniqueid": "bc:33:ac:ff:fe:57:11:78-01-0201"
    },
    "4": {
        "config": {
            "heatsetpoint": 1900,
            "locked": null,
            "mode": "off",
            "offset": 100,
            "on": true,
            "preset": "eco",
            "reachable": true,
            "schedule": {},
            "schedule_on": null,
            "windowopen_set": null
        },
        "ep": 1,
        "etag": "14d12222b5da3198cdf4d2e7973d38fa",
        "lastseen": "2020-10-17T18:27Z",
        "manufacturername": "Heiman",
        "modelid": "TS0601",
        "name": "Tomek",
        "state": {
            "lastupdated": "none",
            "lowbattery": false,
            "on": false,
            "temperature": 3000,
            "valve": 0
        },
        "type": "ZHAThermostat",
        "uniqueid": "bc:33:ac:ff:fe:57:1f:db-01-0201"
    },
    "5": {
        "config": {
            "heatsetpoint": 1800,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true,
            "schedule": {},
            "schedule_on": null,
            "windowopen_set": null
        },
        "ep": 1,
        "etag": "7fab693dbae512c1f843a6a8edd1095f",
        "lastseen": "2020-10-17T22:04Z",
        "manufacturername": "Heiman",
        "modelid": "TS0601",
        "name": "Goscinny",
        "state": {
            "lastupdated": "none",
            "lowbattery": false,
            "on": true,
            "temperature": 3000,
            "valve": 80
        },
        "type": "ZHAThermostat",
        "uniqueid": "84:2e:14:ff:fe:e5:55:86-01-0201"
    },

    "27": {
        "config": {
            "heatsetpoint": 2050,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true,
            "schedule": {},
            "schedule_on": null,
            "windowopen_set": null
        },
        "ep": 1,
        "etag": "1f8271f0d1b11a1d4259832ee1d1c99e",
        "lastseen": "2020-10-17T18:44Z",
        "manufacturername": "Heiman",
        "modelid": "TS0601",
        "name": "Salon lewy",
        "state": {
            "lastupdated": "none",
            "lowbattery": false,
            "on": false,
            "temperature": 2100,
            "valve": 0
        },
        "type": "ZHAThermostat",
        "uniqueid": "84:2e:14:ff:fe:e5:54:2e-01-0201"
    },
    "28": {
        "config": {
            "heatsetpoint": 2050,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "preset": "manual",
            "reachable": true,
            "schedule": {},
            "schedule_on": null,
            "windowopen_set": null
        },
        "ep": 1,
        "etag": "21093b03d7e47a0163caf897db772108",
        "lastseen": "2020-10-17T18:48Z",
        "manufacturername": "Heiman",
        "modelid": "TS0601",
        "name": "Salon prawy",
        "state": {
            "lastupdated": "none",
            "lowbattery": false,
            "on": true,
            "temperature": 2100,
            "valve": 5
        },
        "type": "ZHAThermostat",
        "uniqueid": "bc:33:ac:ff:fe:45:8e:fd-01-0201"
    },

```

Arquiteto, 2020-10-18:
I just used REST API directly to change the setpoint and it worked without any problem. After that I tried to set it with HA again and it worked as well. I'm going to observe it a little bit more and get back to you on that part. Maybe it's an issue with HA deCONZ integration.

I was able to confirm the 30 value comes from boost mode. I've set it to 10minutes on one of TRV's and now it's reporting 10 degrees. I'll get back to you when I manage to try out your code.

Smanar, 2020-10-18:
>I was able to confirm the 30 value comes from boost mode. I've set it to 10minutes on one of TRV's and now it's reporting 10 degrees. I'll get back to you when I manage to try out your code.

I think this part is corrected now, I m using this value only for the siren on last code, so it s disabled for all other device, so for yours too.

But I realy don't understand how your devices can works. I m using the manufacture name for test on the code. Perhaps you add the good value during inclusion (so the device was created normally) , and the code have modified the value at next launch.

Arquiteto, 2020-10-18:
> 
> But I realy don't understand how your devices can works. I m using the manufacture name for test on the code. Perhaps you add the good value during inclusion (so the device was created normally) , and the code have modified the value at next launch.

I have an idea on that one. It says Heiman when calling the REST API, but in deCONZ basic cluster manufacturer id is "_TZE200_ckud7u2l" so the same as in #3110

Smanar, 2020-10-18:
You are able to test the new code ? for the 30 degre problem ?

Arquiteto, 2020-10-19:
> You are able to test the new code ? for the 30 degre problem ?

Sorry, I was working nightshift, I just had time to sit down on it. I wasn't able to setup it properly on my QNAP so I flashed a fresh Raspberry Pi OS on empty sd card, installed deconz-dev then:
```
git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
git checkout -b mybranch HEAD
```

This works ok, but

`qmake && make -j2`

comes out with error
```
de_web_plugin.cpp: In member function ‘void DeRestPluginPrivate::addLightNode(const deCONZ::Node*)’:
de_web_plugin.cpp:1759:68: error: expected primary-expression before ‘||’ token
         node->nodeDescriptor().manufacturerCode() == VENDOR_EMBER) || // atsmart Z6-03 switch + Heiman plug + Tuya stuff
```

Smanar, 2020-10-19:
My bad, typo, just corrected, can make a second try.

Arquiteto, 2020-10-19:
So it seems that the bracket at line 1759 shouldn't be there. I deleted it and it compiled.

It seems like the issue is fixed now. Thank you. In REST API and HA manufacturer changed to "_TZE200_ckud7u2l".

Not sure if anything can be done about the fact that TRV's report most of the values 'null' for quite a long time after connecting to gateway, might confuse new users that they are not working properly. One of the valves didn't report anything to HA for 15 min until I manually changed the setpoint on the valve.

Smanar, 2020-10-19:
> Not sure if anything can be done about the fact that TRV's report most of the values 'null' for quite a long time after connecting to gateway, might confuse new users that they are not working [...]

Was exactly the problem view by another user today > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3381#issuecomment-711421446

Arquiteto, 2020-10-19:
Okay. I've been running new code for a few hours, even changed boost time manually on TRV and didn't get any wrong report, so it seems this problem is fixed.

Smanar, 2020-10-19:
Ok thx, will make the pr in some day.
btw the version you have just compiled have some critic bug (segmentation fault) that was corrected today.
So if you want to use a compiled one, better to wait some days, else the correction will be in a future version.

Mimiix, 2020-10-20:
@Smanar Is this the same issue? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3452

Smanar, 2020-10-20:
No same issue, but same code part, will answer on other issue.

closed 2020-11-15 by manup

RubenKelevra, 2021-02-05:
Hey @Arquiteto, the first screenshot looks interesting. Can you send me a link to how you did that? :)

Arquiteto, 2021-02-05:
> Hey @Arquiteto, the first screenshot looks interesting. Can you send me a link to how you did that? :)

It's made with multiple-entity-row integration, that you can install in HACS for the status and buttons are set with paper-buttons-row