dresden-elektronik/deconz-rest-plugin #3442: No Scene activation for Aqara TW Light ZNLDP12LM

iaktern, 2020-10-18:
Hi,

I think there is a bug with Xiaomi's Aqara TW Light (ZNLDP12LM) if you try to activate a scene. 

## Describe the bug

The Aqara light is the only member of a group. I've created multiple scenes with different colour temperatures for this group via the REST API. I also checked if the scenes are correctly stored, which they are. (I also confirmed the correct number of scenes in the deCONZ GUI via the _Get scene membership_ command.)

The problem: It is not possible to activate the scenes. Neither with the _Recall scene_ command in the GUI nor with the REST API.

## Steps to reproduce the behavior

0. Add a new group and connect it with the Aqara light
1. Set the group state via the REST API: `PUT api/{{apikey}}/groups/<group-id>/action`
1. Store the state in a scene via the REST API: `PUT api/{{apikey}}/groups/<group-id>/scenes/<scene-id>/store`
1. Repeat the previous steps for every scene you want to have and check if it was stored via: `GET api/{{apikey}}/groups/<group-id>/scenes/<scene-id>`
1. try to activate a scene with `PUT api/{{apikey}}/groups/<group-id>/scenes/<scene-id>/recall`

## Expected behavior

The stored scene should be activated.

## Environment

 - Host system: Raspberry Pi 4B
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.05.84
 - Device: ConBee II
 - Do you use an USB extension cable: yes 

### deCONZ Logs

I didn't found any interesting things in the logs (I've looked into _dbg-info_ and _dbg-aps_).

## Additional context

The firmware of the Aqara light seems to be quite old (11-22-2018). Are there any publicly available firmware files?

Thanks for the help!

iaktern, 2020-10-24:
Actually, my goal was to use a [Ikea 5-button remote control](https://www.ikea.com/us/en/p/tradfri-remote-control-00443130/) to change the color temperature of the light. Usually I do this with scenes and let the left/right buttons activate different scenes. (Which does not work for this Aqara light, because I'm not able to activate any scene although it seems that the light has stored the scenes.) 

As a workaround,  I tried to use the Ikea remote control to change the color temperature via the left and right button. I configured this with the Phoscon App. Unfortunately, this seems also not to work... Does anyone have an idea how I can achieve my goal to change the color temperature with the Ikea remote control?

Btw, if someone wants to work on the scene activation issue, I have an Aqara light too much. I could send it to someone.

SwoopX, 2020-10-24:
> Neither with the Recall scene command in the GUI nor with the REST API.

Then I feel chances are low that it will work at all. Please take note that Xiaomi is not a shining example for following and appropriately implementing zigbee specs/standard. It could also be that it is doable via some device specific quirks yet unknown.

However, maybe anybody out there has made some more encouraging experience with them.

ebaauw, 2020-10-25:
> Btw, if someone wants to work on the scene activation issue, I have an Aqara light too much. I could send it to someone.

If it’s an EU bulb, I’ll be happy to have a look at it.  In the meantime, could you post a screenshot of the GUI, showing the endpoints and clusters of this light?

See my comments https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2364#issuecomment-715206300 regarding Zigbee scenes: the standard doesn’t cater for colour temperature, claiming you can use _CurrentX_ and _CurrentY_ instead.  _Store Scene_ seems way more robust than _Add Scene_.  Would be helpful if you have a Zigbee sniffer, to check the payload of the _Enhanced View Scene Response_.

>  I tried to use the Ikea remote control to change the color temperature via the left and right button.

The direct control of lights by the remote uses manufacturer-specific commands for the left and right buttons.  These are understood only by IKEA lights, in factory new state.  As soon as the API includes the lights in `/groups/0`, they no longer accept these commands.

> Does anyone have an idea how I can achieve my goal to change the color temperature with the Ikea remote control?

I would create rules myself, setting `ct` in a group action.  You create these manually using the API.

iaktern, 2020-10-25:
> If it’s an EU bulb, I’ll be happy to have a look at it.

Great, thanks.

>  In the meantime, could you post a screenshot of the GUI, showing the endpoints and clusters of this light?

[image]

> See my comments #2364 (comment) regarding Zigbee scenes: the standard doesn’t cater for colour temperature, claiming you can use CurrentX and CurrentY instead. Store Scene seems way more robust than Add Scene. 

Great explanation. I tried both - _add_ and _store_ scene- but both did not work (although I'm not quite sure if I correctly translated the ct value to x and y). 

Btw, it would be super helpful if such zigbee knowledge is documented inside the wiki. I already often saw some very interesting things inside the issues, but the problem is that it is not that easy to find it again later. 

> I would create rules myself, setting ct in a group action. You create these manually using the API.

Thanks, I did that and described it [here for others.](https://github.com/dresden-elektronik/phoscon-app-beta/issues/176#issuecomment-716197427)

ebaauw, 2020-10-25:
Wow, they put every cluster in there, except _Door Lock_.  This could confuse the hell out of the API plugin.

> If it’s an EU bulb, I’ll be happy to have a look at it.
>> Great, thanks.

Please PM me on Discord for my address.

ebaauw, 2020-11-02:
Received the light this weekend, thanks @iaktern.

Paired with deCONZ on first attempt.  Most clusters only report non-supported attributes.  As for the light features:
- No power on setting support;
- 8 groups, 16 scenes;
- Only `ct`, no `xy` nor `hs` control.  `ctmin` (0) and `ctmax` (0xfeff) values are theoretical;
- `ct` seems to accept any value, which is echoed back (through _Report Attributes_ and _Read Attributes_).  `xy` values are updated, but not reported (only visible through _Read Attributes_).  It also reports attribute 0xf000, u32 with a value that I don't know how to interpret.
- It seems to support _Add Scene_.  It reports _Insufficient Space_ on a _Store Scene_.  It doesn't report the scene state in _View Scene_ and it doesn't support _Enhanced View Scene_.

I managed to create some scenes in the old web app (also in the GUI, but then the API doesn't pick them up).  Subsequently, I can issue a PUT to `/groups/x/scenes/y/lights/z/state` of `{"on": true, "bri": 254}`.  This creates the scene with _OnOff_, _Level Control_, and, surprisingly, _Color Control_ attributes.  Apparently the REST API substitutes what it thinks are the current values here.  I tried adding `xy` to the scene state; the light accepts the _Add Scene_, but it doesn't seem to restore the colour.  Best I can tell, the light doesn't support _Color Control_ with _Scenes_.

iaktern, 2020-11-03:
hmmm, okay - thsnk you very much for the analysis. It's not so nice that the light doesn't save the color with the scene, but I guess there is not much we can do about it...
I have linked the issue in the wiki entry for the Xiaomi/Aqara light. Should the issue be closed? (it's not fixed - so if the issue is closed, it is maybe misleading for someone who doesn't read to the end...)

ebaauw, 2020-11-03:
> I guess there is not much we can do about it...

For API v2, there’s an idea to support (API) scenes for lights that don’t (fully) support Zigbee scenes, see https://github.com/dresden-elektronik/deconz-rest-plugin-v2/issues/5.

Until then you could use scenes of your favourite home automation system.

closed 2020-12-19 by stale[bot]