dresden-elektronik/deconz-rest-plugin #3448: Aqara OPPLE switch WXCJKG13LM pairing but not button events

helipus, 2020-10-19:
## Describe the question or issue you are having
I successfully paired the Aqara OPPLE switch WXCJKG13LM but I don't receive any button events.
What have I done wrong?

## Screenshots
[image]

## Environment
 - Host system: Synology NAS DS716+
 - Running method: Marthoc Docker container latest
 - Firmware version: 26660700
 - deCONZ version: 2.05.85
 - Device: ConBee II 
 - Do you use an USB extension cable: yes 1,5m
 - Is there any other USB or serial devices connected to the host system? No

### deCONZ Logs
Relevant output from following url: 
http://192.168.0.2:8080/api/C009B84873/sensors
```json
"24": {
"config": {
"battery": null,
"on": true,
"reachable": true
},
"ep": 1,
"etag": "1dc874ee9f6ad95e1811735dd41b154a",
"lastseen": "2020-10-19T20:00Z",
"manufacturername": "LUMI",
"mode": 1,
"modelid": "lumi.remote.b686opcn01",
"name": "OPPLE Schalter 3-fach",
"state": {
"buttonevent": null,
"lastupdated": "none"
},
"swversion": "20190730",
"type": "ZHASwitch",
"uniqueid": "04:cf:8c:df:3c:79:73:4b-01-0012"
}
```

SwoopX, 2020-10-19:
I'm afraid you need to wait for version .86 to be available via docker as it contains a fix exactly for that. Should be around the next 1-2 days. Another option, if feasible, use version .82. 3rd option would be to follow this and find the required file https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3412#issuecomment-710541132

closed 2020-10-19 by Mimiix

helipus, 2020-10-20:
The bug was fixed with the version .86! Thanks a lot.