dresden-elektronik/deconz-rest-plugin #3450: Debian update broke RaspBee?

michis0806, 2020-10-19:
I have currently a problem with my Raspbee on a Raspberry Pi 3. 

A few weeks ago, I did a "sudo apt update && sudo apt upgrade -y" to get the latest updates. After this upgrades (that also contained deconz .80) phoscon showed vendor: unkonwn, firmware: not connected in the gateway-screen.
I thought it was a problem with deconz and rolled back the update. I tried the same with deconz .81 and .82 with the same result, connection to RaspBee was broken.
I then tried to update deconz alone with worked.

Now I tried to upgrade all packages again with the .85. This again broke the connection to the RaspBee. 

I'm currently in a state where even a plain Phoscon SD-Image install cannot connect to my RaspBee. 

GCFFlasher -l shows

```
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee
```
sudo GCFFlasher_internal -f deCONZ_Rpi_0x26390500.bin.GCF -x 3 breakes with
```
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
23:16:03:821 using firmware file: deCONZ_Rpi_0x26390500.bin.GCF 
23:16:03:974 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
23:16:03:979 check symlink /dev/ttyAMA0 (/dev/ttyS0)
23:16:03:979 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
23:16:03:980 check symlink /dev/ttyS0 (/dev/ttyS0)
23:16:03:981 dev /dev/ttyS0 (/dev/ttyS0)
Reboot device RaspBee (RaspBee)
23:16:03:986 query bootloader v1 ID after 0 ms
23:16:05:488 query bootloader v1 ID after 1502 ms
23:16:05:989 query deCONZ firmware version
23:16:07:993 uart reset failed, try RaspBee reset
action: reset device RaspBee
wiringPi 2.52 initialized
23:16:08:198 query bootloader v1 ID after 1 ms
23:16:08:212 RX 25 bytes ASCII C?É??[L?SN?MEU?
                                               after 15 ms
23:16:08:212 RX 43a4c3891494cb1c5b0e14074c94534e02cc4d16450355800c after 15 ms
23:16:08:224 RX 9 bytes ASCII ES?O?? after 27 ms
23:16:08:224 RX 14450553b1144f9ac0 after 28 ms
23:16:08:713 RX 8 bytes ASCII ??c?? after 516 ms
23:16:08:713 RX 8d8463cacf340817 after 516 ms
23:16:08:772 RX 9 bytes ASCII ؆p? after 575 ms
23:16:08:772 RX d88670fe00fe00fff8 after 575 ms
23:16:09:287 RX 7 bytes ASCII ؊p??? after 1090 ms
23:16:09:287 RX d88a708004d9f8 after 1090 ms
23:16:10:105 RX 7 bytes ASCII ؎p??? after 1908 ms
23:16:10:105 RX d88e708004c7f8 after 1909 ms
23:16:11:107 query bootloader v3 ID after 2910 ms
23:16:11:107 TX c081027dffc0
```

I have a old image of the card that works but it seems I'm stuck here and cannot make any update to the system any more.

Any Ideas how I can fix this?

Thanks
Michael

Mimiix, 2020-10-20:
Hi. As you removed all contents of the template, i am closing this issue. 

Please open a new issue with template filled correctly.

closed 2020-10-20 by Mimiix

michis0806, 2020-10-20:
*facepalm*