dresden-elektronik/deconz-rest-plugin #3451: Debian update broke RaspBee? / Firmware update fails...

michis0806, 2020-10-20:
## Describe the question or issue you are having
I have currently a problem with my Raspbee on a Raspberry Pi 3. 

A few weeks ago, I did a "sudo apt update && sudo apt upgrade -y" to get the latest updates. After this upgrades (that also contained deconz .80) phoscon showed vendor: unkonwn, firmware: not connected in the gateway-screen.
I thought it was a problem with deconz and rolled back the update. I tried the same with deconz .81 and .82 with the same result, connection to RaspBee was broken.
I then tried to update deconz alone with worked.

Now I tried to upgrade all packages again with the .85. This again broke the connection to the RaspBee. 

I'm currently in a state where even a plain Phoscon SD-Image install cannot connect to my RaspBee. 

GCFFlasher -l shows

```
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee
```
sudo GCFFlasher_internal -f deCONZ_Rpi_0x26390500.bin.GCF -x 3 breakes with
```
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
23:16:03:821 using firmware file: deCONZ_Rpi_0x26390500.bin.GCF 
23:16:03:974 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
23:16:03:979 check symlink /dev/ttyAMA0 (/dev/ttyS0)
23:16:03:979 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
23:16:03:980 check symlink /dev/ttyS0 (/dev/ttyS0)
23:16:03:981 dev /dev/ttyS0 (/dev/ttyS0)
Reboot device RaspBee (RaspBee)
23:16:03:986 query bootloader v1 ID after 0 ms
23:16:05:488 query bootloader v1 ID after 1502 ms
23:16:05:989 query deCONZ firmware version
23:16:07:993 uart reset failed, try RaspBee reset
action: reset device RaspBee
wiringPi 2.52 initialized
23:16:08:198 query bootloader v1 ID after 1 ms
23:16:08:212 RX 25 bytes ASCII C?É??[L?SN?MEU?
                                               after 15 ms
23:16:08:212 RX 43a4c3891494cb1c5b0e14074c94534e02cc4d16450355800c after 15 ms
23:16:08:224 RX 9 bytes ASCII ES?O?? after 27 ms
23:16:08:224 RX 14450553b1144f9ac0 after 28 ms
23:16:08:713 RX 8 bytes ASCII ??c?? after 516 ms
23:16:08:713 RX 8d8463cacf340817 after 516 ms
23:16:08:772 RX 9 bytes ASCII ؆p? after 575 ms
23:16:08:772 RX d88670fe00fe00fff8 after 575 ms
23:16:09:287 RX 7 bytes ASCII ؊p??? after 1090 ms
23:16:09:287 RX d88a708004d9f8 after 1090 ms
23:16:10:105 RX 7 bytes ASCII ؎p??? after 1908 ms
23:16:10:105 RX d88e708004c7f8 after 1909 ms
23:16:11:107 query bootloader v3 ID after 2910 ms
23:16:11:107 TX c081027dffc0
```

I have a old image of the SD-card that works but it seems I'm stuck here and cannot make any update to the system any more.

Any Ideas how I can fix this?

I already tried to flash older versions of the firmware but always geht the bug above...

Thanks
Michael

## Screenshots

## Environment

Deconz: 2.05.85 / 15.10.2020
Firmware: 26390500

 - Host system: Raspberry Pi 3
 - Running method: Raspbian
 - Firmware version: (26390500)
 - deCONZ version: (2.05.85)
 - Device: RaspBee I

### deCONZ Logs

as described above

## Additional context

as described above

Mimiix, 2020-10-20:
HI!

As per documentation](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-raspbian) , Raspbee 1 needs 

sudo GCFFlasher_internal -t 60 -f deCONZ_Rpi_0x26390500.bin.GCF

You have -x 3 ? But i have no clue what that is. 

Can you check with that?

michis0806, 2020-10-20:
@Mimiix -x 3 is the log level...
Flash worked with -t 60 but I still see no valid Vendor/Product/Serial here...

```
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
```

Mimiix, 2020-10-20:
What version of raspbian is running?

Did you set the settings on the serial menu of raspbian? Please see: https://phoscon.de/en/raspbee/install#raspbian

michis0806, 2020-10-20:
I'm running Raspbian Buster Raspbian GNU/Linux 10 and I checked the settings.

My problem is, that it's currently running as expected.

But when update the raspbian with "apt upgrade" it seems that deconz cannot connect to the RaspBee any more. 
The same happens when I try to use my RaspBee on a plain new created SD-Card with the current image...

My guess would be that somethings broken with my RaspBee that also causes the empty Vendor/Product/Serial output in GCFFlasher...

Mimiix, 2020-10-20:
It would be interesting to see what is updated with the apt upgrade. 

Maybe a dependency?

I'll ask Manup on this one.

michis0806, 2020-10-20:
```
base-files/stable 10.3+rpi1+deb10u6 armhf [aktualisierbar von: 10.3+rpi1+deb10u4]
bind9-host/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
bluez-firmware/testing 1.2-4+rpt6 all [aktualisierbar von: 1.2-4+rpt3]
bluez/testing 5.50-1.2~deb10u1+rpt2 armhf [aktualisierbar von: 5.50-1.2~deb10u1+rpt1]
dbus-user-session/stable 1.12.20-0+deb10u1 armhf [aktualisierbar von: 1.12.16-1]
dbus/stable 1.12.20-0+deb10u1 armhf [aktualisierbar von: 1.12.16-1]
deconz/unbekannt 2.05.86-raspbian-buster-beta armhf [aktualisierbar von: 2.05.85-raspbian-buster-beta]
firmware-atheros/testing 1:20190114-1+rpt9 all [aktualisierbar von: 1:20190114-1+rpt7]
firmware-brcm80211/testing 1:20190114-1+rpt9 all [aktualisierbar von: 1:20190114-1+rpt7]
firmware-libertas/testing 1:20190114-1+rpt9 all [aktualisierbar von: 1:20190114-1+rpt7]
firmware-misc-nonfree/testing 1:20190114-1+rpt9 all [aktualisierbar von: 1:20190114-1+rpt7]
firmware-realtek/testing 1:20190114-1+rpt9 all [aktualisierbar von: 1:20190114-1+rpt7]
glib-networking-common/stable 2.58.0-2+deb10u2 all [aktualisierbar von: 2.58.0-2]
glib-networking-services/stable 2.58.0-2+deb10u2 armhf [aktualisierbar von: 2.58.0-2]
glib-networking/stable 2.58.0-2+deb10u2 armhf [aktualisierbar von: 2.58.0-2]
libbind9-161/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
libdbus-1-3/stable 1.12.20-0+deb10u1 armhf [aktualisierbar von: 1.12.16-1]
libdns-export1104/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
libdns1104/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
libgnutls-dane0/stable 3.6.7-4+deb10u5 armhf [aktualisierbar von: 3.6.7-4+deb10u4]
libgnutls30/stable 3.6.7-4+deb10u5 armhf [aktualisierbar von: 3.6.7-4+deb10u4]
libinput-bin/stable 1.12.6-2+deb10u1 armhf [aktualisierbar von: 1.12.6-2]
libinput10/stable 1.12.6-2+deb10u1 armhf [aktualisierbar von: 1.12.6-2]
libisc-export1100/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
libisc1100/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
libisccc161/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
libisccfg163/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
libjson-c3/stable 0.12.1+ds-2+deb10u1 armhf [aktualisierbar von: 0.12.1+ds-2]
liblwres161/stable 1:9.11.5.P4+dfsg-5.1+deb10u2 armhf [aktualisierbar von: 1:9.11.5.P4+dfsg-5.1+deb10u1]
libmailutils5/stable 1:3.5-4 armhf [aktualisierbar von: 1:3.5-3]
libmariadb3/stable 1:10.3.23-0+deb10u1 armhf [aktualisierbar von: 1:10.3.22-0+deb10u1]
libntlm0/stable 1.5-1+deb10u1 armhf [aktualisierbar von: 1.5-1]
libpython3.7-minimal/stable 3.7.3-2+deb10u2 armhf [aktualisierbar von: 3.7.3-2+deb10u1]
libpython3.7-stdlib/stable 3.7.3-2+deb10u2 armhf [aktualisierbar von: 3.7.3-2+deb10u1]
libpython3.7/stable 3.7.3-2+deb10u2 armhf [aktualisierbar von: 3.7.3-2+deb10u1]
libqt5core5a/stable 5.11.3+dfsg1-1+rpi1+deb10u4 armhf [aktualisierbar von: 5.11.3+dfsg1-1+rpi1+deb10u3]
libqt5dbus5/stable 5.11.3+dfsg1-1+rpi1+deb10u4 armhf [aktualisierbar von: 5.11.3+dfsg1-1+rpi1+deb10u3]
libqt5gui5/stable 5.11.3+dfsg1-1+rpi1+deb10u4 armhf [aktualisierbar von: 5.11.3+dfsg1-1+rpi1+deb10u3]
libqt5network5/stable 5.11.3+dfsg1-1+rpi1+deb10u4 armhf [aktualisierbar von: 5.11.3+dfsg1-1+rpi1+deb10u3]
libqt5sql5-sqlite/stable 5.11.3+dfsg1-1+rpi1+deb10u4 armhf [aktualisierbar von: 5.11.3+dfsg1-1+rpi1+deb10u3]
libqt5sql5/stable 5.11.3+dfsg1-1+rpi1+deb10u4 armhf [aktualisierbar von: 5.11.3+dfsg1-1+rpi1+deb10u3]
libqt5widgets5/stable 5.11.3+dfsg1-1+rpi1+deb10u4 armhf [aktualisierbar von: 5.11.3+dfsg1-1+rpi1+deb10u3]
libraspberrypi-bin/testing 1.20200902-1 armhf [aktualisierbar von: 1.20200601-1]
libraspberrypi-dev/testing 1.20200902-1 armhf [aktualisierbar von: 1.20200601-1]
libraspberrypi-doc/testing 1.20200902-1 armhf [aktualisierbar von: 1.20200601-1]
libraspberrypi0/testing 1.20200902-1 armhf [aktualisierbar von: 1.20200601-1]
libx11-6/stable 2:1.6.7-1+deb10u1 armhf [aktualisierbar von: 2:1.6.7-1]
libx11-data/stable 2:1.6.7-1+deb10u1 all [aktualisierbar von: 2:1.6.7-1]
libx11-xcb1/stable 2:1.6.7-1+deb10u1 armhf [aktualisierbar von: 2:1.6.7-1]
mailutils-common/stable 1:3.5-4 all [aktualisierbar von: 1:3.5-3]
mailutils/stable 1:3.5-4 armhf [aktualisierbar von: 1:3.5-3]
mariadb-common/stable 1:10.3.23-0+deb10u1 all [aktualisierbar von: 1:10.3.22-0+deb10u1]
nfs-common/stable 1:1.3.4-2.5+deb10u1 armhf [aktualisierbar von: 1:1.3.4-2.5]
pi-bluetooth/testing 0.1.15 all [aktualisierbar von: 0.1.13]
python3.7-minimal/stable 3.7.3-2+deb10u2 armhf [aktualisierbar von: 3.7.3-2+deb10u1]
python3.7/stable 3.7.3-2+deb10u2 armhf [aktualisierbar von: 3.7.3-2+deb10u1]
qt5-gtk-platformtheme/stable 5.11.3+dfsg1-1+rpi1+deb10u4 armhf [aktualisierbar von: 5.11.3+dfsg1-1+rpi1+deb10u3]
raspberrypi-bootloader/testing 1.20200902-1 armhf [aktualisierbar von: 1.20200601-1]
raspberrypi-kernel-headers/testing 1.20200902-1 armhf [aktualisierbar von: 1.20200601-1]
raspberrypi-kernel/testing 1.20200902-1 armhf [aktualisierbar von: 1.20200601-1]
raspberrypi-sys-mods/testing 20200812 armhf [aktualisierbar von: 20200514]
raspi-config/testing 20200902 all [aktualisierbar von: 20200601]
rpi-eeprom/testing 10.1-1 all [aktualisierbar von: 7.4-1]
```

manup, 2020-10-20:
Can you please check that the serial configuration is still valid?
It might have been changed by the full apt update.

Step 1) from https://phoscon.de/en/raspbee/install#raspbian

michis0806, 2020-10-20:
[image]
Jupp, I tried/checkt this several times...

I'm currently not sure what's the expected result of the GCFFlasher_internal -l? Should there be any other output than my example 0x0000/0x0000/EMPTY ?

manup, 2020-10-20:
The output seems fine, the other fields are just filled for USB devices / ConBee.

What does the follwoing command bring:
`sudo GCFFlasher_internal -r -x3`

Note deCONZ needs to be closed here, if it works it should print the flashed firmware version.

michis0806, 2020-10-20:
```
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
14:58:30:030 ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
14:58:30:031 check symlink /dev/ttyAMA0 (/dev/ttyS0)
14:58:30:031 ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
14:58:30:032 check symlink /dev/ttyS0 (/dev/ttyS0)
14:58:30:032 dev /dev/ttyS0 (/dev/ttyS0)
Reboot device RaspBee (RaspBee)
14:58:30:035 query bootloader v1 ID after 0 ms
14:58:31:537 query bootloader v1 ID after 1502 ms
14:58:32:037 query deCONZ firmware version
14:58:32:045 SLIP RX frame length: 9
deCONZ firmware version 26370500
14:58:32:091 set watchdog ttl status: 0x00
14:58:34:193 query bootloader v1 ID after 1 ms
14:58:34:886 RX 41 bytes ASCII 
RaspBee Bootloader premium
Vers. 1.02
bu after 694 ms
14:58:34:886 bootloader start after 694 ms
RaspBee Bootloader premium
Vers. 1.02
14:58:34:888 GCF_ResetDeviceDone
```

djiwondee, 2020-10-27:
Have the same experience with `apt-get update` and `apt-get dist-upgrade` and have already opened an [issue here](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3451#issue-725272475) for that.

Assuming if there are new kernel-binaries and/or changes in the kernel at all, the kext get broken.
Keen to find a solution for that.

glagolster, 2020-12-18:
Since 3 days I am trying to bring the Raspbee II running, I have currently the same problem. Last night, everything was fine. I shutdown my raspberry to store them at the rigt place. After I turned on again, I am not able to establish any connection.

In my process, to bring them running, I made a firmware downgrade to deCONZ_RaspBeeII_0x26660700.bin.GCF and everything was fine. Now, when I try to downgrade again, I have the following reply.

> pi@phoscon:~ $ sudo GCFFlasher_internal -t 60 -f deCONZ_RaspBeeII_0x26660700.bin.GCF
GCFFlasher V3_16 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
wiringPi 2.52 initialized
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
Reboot device RaspBee (RaspBee)
action: reset device RaspBee
retry, failed

The basement is the official Phoscon_Gateway_2020-11-26 stable.xz image.

closed 2021-03-09 by github-actions[bot]