dresden-elektronik/deconz-rest-plugin #3453: Force Aqara temperature sensor to connect via an OSRAM Smart Plugs

airbone42, 2020-10-20:
I would like to install an Aqara temperature sensor in our basement, which doesn't have a very good Zigbee connection. Therefore I ordered several OSRAM Smart Plugs, which afaik can work as a Zigbee repeater for these devices, to build a bridge to the basement.

Unfortunately, the temperature sensor only connects a Paulmann light which is installed in the top floor. As soon as I remove the sensor away from there it loses the connection.

Is there any way to force a device to connect to a specific repeater?

Mimiix, 2020-10-20:
@airbone42 **Moderation note: As i can answer your question without the info now, i'll proceed. However, please use the template next time. This helps us greatly.**

Aqara sensors are known to stay with their original parent, meaning they won't change their parent at all. Even when its parent goes offline, it doesn't switch.

What you could try is re-pairing it in the basement. Chances are that it would pair to one of the closer ones and thus getting better signal.

airbone42, 2020-10-20:
I'm sorry for missing to use the template. Will definitely take care of that next time.

Thanks for the clarification. I tried to re-pair, having the OSRAM plug directly besides the sensor. But it seems like also the Conbee-stick needs to be near the sensor when pairing. If I put only the plug (which is in the Zigbee network) near the sensor, the sensor can't be re-paired at all. If I also the the Conbee-stick near it pairs then directly to the Conbee-stick, which doesn't help as that has to go back to the top floor.

Am I right that the only chance is to do re-pairing again and again until it luckily connects to the correct master?

Mimiix, 2020-10-20:
Yep! That's basically it. It's really stupid, as the sensor picks it. It's just one of the quirks the Aqara's have. 

Sonoff is a good alternative ;)

MattWestb, 2020-10-20:
Xaiomi sensors is normally picky of the router they is connecting 2 and more or less all OSRAM have old TI chips with buggy stacks (discontinued = no bug fixes from TI) and normally losing the Xiaomi sensors after some time. Also the OSRAM smart plug is famous of having pretty bad antenna so the signal its normally bad or worse (and RaspBee / ConBee is having very good antennas). Also most Xaiomi sensors refuging connecting to some routers then they is not can communicating properly with each other (not Zigbee compatible / certified Xaiomi devices). If you have one more Paulmann light i would putting one in the near the sensor as repeater and trying pairing the sensor thru it (not easy in deCONZ).
With ZHA its possible only open one router for joining devices and its working great. I having for years one Aqara weather sensor in my bath and wos putting one IKEA outlet ther last week, and then opening it for joining and resetting the sensor and it was connecting thru the outlet and have very good signal, reporting good and its saving the battery in the sensor. 

You can also trying HA with deCONZ dongles as they is supported in ZHA and see if you is liking it (and Sonoff Zigbee Bridge is also very good supported in ZHA and is equal or better then DEs NPCs then coming 2 hardware and radio performance).

Mimiix, 2020-10-20:
@MattWestb I have really no clue in what way this would fix. You are acknowledging the osram's aren't good, thus saying the user is having a issue and then telling him to switch to ZHA. That would not fix his issue _at all_. I still feel he should either retry or get another router. ZHA won't fix anything here for anyone.

airbone42, 2020-10-20:
> Yep! That's basically it. It's really stupid, as the sensor picks it. It's just one of the quirks the Aqara's have.
> 
> Sonoff is a good alternative ;)

Thanks just ordered a SONOFF and will test it ;)

MattWestb, 2020-10-20:
Im very sorry Dennis but 
>  If you have one more Paulmann light i would putting one in the near the sensor as repeater and trying pairing the sensor thru it (not easy in deCONZ).

is addressed to the user trying on other router in the near of the sensor and one that hi is knowing is working. 

>Sonoff is a good alternative ;)

And this is ???  
Is the support for the EZSP coming in deCONZ ??
I have helping getting support in for ZHA and Tsamota for the new EZSP so the Sonoff Zigbee Bridge is now working on both.

I was NOT saying "switch to ZHA" only trying if its working better than hi then can forcing which router to joining the device thru with his deCONZ dongle.

manup, 2020-10-20:
The simplest method to force Aqara devices to join to a certain router is to power down other routers during the pairing.
However although not so common for Xiaomi (but have been seen) end-devices may pick any other router as parent afterwards for various reasons. Regardless of the gateway there exists no solution to permanently force end-devices pick a specific parent in Zigbee.

airbone42, 2020-10-20:
> The simplest method to force Aqara devices to join to a certain router is to power down other routers during the pairing.
> However although not so common for Xiaomi (but have been seen) end-devices [...]

Unfortunately, that didn't really work as I anyway need the Conbee-stick near the sensor to allow joining the network.

manup, 2020-10-20:
For Aqara devices it's best to join them at their desired location, pairing works through the mesh so joining doesn't need to be near the ConBee.

airbone42, 2020-10-20:
> For Aqara devices it's best to join them at their desired location, pairing works through the mesh so joining doesn't need to be near the ConBee.

Thanks for the message. Weirdly that didn't work for me. If the Conbee is not near it doesn't find the network, also had this in the past with other devices like doorsensors.

closed 2020-11-24 by stale[bot]