dresden-elektronik/deconz-rest-plugin #3455: Aqara Vibration Sensor stopped working after Update to .86

peer69, 2020-10-20:
## Describe the bug

After Update to .86 and FW 26660700 using ConBee II the Aqara Vibration Sensor stopped working. It did not send any commands, at least none of them were visible in Phoscon or deCONZ GUI. 
I have tried repairing it but it seems like the device is not fully recognized anymore. GUI does recognize it as vibration sensor and it builds the cluster correctly from what I can see but no data is shown via Phoscon or API. I can see the sensor flashing blue in GUI when it is triggered but it never updates. I tried a second sensor but it doesnt work either. 
## Steps to reproduce the behavior

1. Pair sensor via Phoscon GUI
  2. Wait for pairing to complete 
  3. Click on sensor
  4. No data is shown neither is via API
## Expected behavior

 Vibration events should be recognized and shown. 
## Screenshots
 
[image]

[image]

## Environment

 - Host system: Ubuntu 16.04 
 - Running method: ESXI 6.7
 - Firmware version: (26660700)
 - deCONZ version: (2.05.86)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which? no.

### deCONZ Logs

## Additional context

Mimiix, 2020-10-20:
What happens if you click the button on it?

peer69, 2020-10-20:
Device in GUI flashing blue. Nothing to see in phoscon or via API.

peer69, 2020-10-20:
I have just downgraded to .83. It works again. Seems to be a bug with the never build.

Agratam, 2020-10-20:
2.05.84 works too.
I had issues with 2.05.86.

Host system: Ubuntu 20.04
Firmware version: (26660700)
deCONZ version: (2.05.86)
Device: (ConBee II)
Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
Is there any other USB or serial devices connected to the host system? If so: Which? Yes. Aeotec Zwave Stick

Mimiix, 2020-10-20:
@Agratam Do you specifically mean that everything works EXCEPT the vib sensor?

Agratam, 2020-10-20:
Yes, everything else works. i.e.:
Tradfri Lights and Switches
Aqara Temperature and Motion sensors

Mimiix, 2020-10-20:
Alright, thanks :)

Mimiix, 2020-10-20:
I've created a internal issue. I'll have this addressed.

SAOPP, 2020-10-20:
Yes guys still broken vibros...

FunkOdyssey2001, 2020-10-20:
I have this issue in 2.05.85 and 2.05.86.

SAOPP, 2020-10-20:
@FunkOdyssey2001 yes dat right, it's beginb from 95th beta.

FunkOdyssey2001, 2020-10-20:
I am not really a developer, but in 2baf9573f18035fc182fa67fa8c4662e2fa50bd2 there were several lines deleted.
I could not find similar lines in the current code for lumi.vibration. But again: I don't know if this is right or wrong. :-)

nikbr998, 2020-10-21:
Just confirming that I also have the issue. 2.05.85/26390500/Conbee1 (Marthoc container).

Mimiix, 2020-10-21:
**No need for others adding "i have it too". It is known**

ebaauw, 2020-10-21:
For reasons only known by Xioami, this sensor uses the _Door Lock_ cluster.  Probably adding support for one of the locks broke support for the vibration sensor.

manup, 2020-10-22:
I'll make some tests later on with debugger to see where it fails.

Mimiix, 2020-10-22:
Will be fixed in new beta! :)

SAOPP, 2020-10-22:
Thanks guys for quick fix it.

FunkOdyssey2001, 2020-10-22:
Is this fix working for you?
I don't see any new values.

Mimiix, 2020-10-22:
NEed to wait for the new release or compile it yourself @FunkOdyssey2001  ;)

FunkOdyssey2001, 2020-10-22:
That's what I did. 
A current checkout, compiled and copied.

SwoopX, 2020-10-25:
With the change made by @manup, you don't need to compile the plugin. Just replace the `general.xml` from the pulled repo with the version running on your system. Otherwise, that file will just be overwritten with by a new release package.

FunkOdyssey2001, 2020-10-26:
I am sorry. You are absolutely right. It is working fine with the fixed xml. Sorry.

SwoopX, 2020-10-26:
Np, glad we got confirmation.

cracyfloyd, 2020-10-31:
i was going back to version 84 and now it works again.

sophof, 2020-11-10:
Thanks for point me to this thread. I replaced the general.xml file and can confirm that the sensor is currently working. One sensor is not currently picked up by Home Assistant, but that could be due to my previous tinkering, I expect it to work once I re-add it.

timbru31, 2020-11-18:
While my sensors initially showed data again after repairing them (I’ve had the very very binding), the stop working after a while (time ranges from hours to days). It seems the same as #2581 and #2591 - I suspect the sensitivity values changes and thus breaks my use case (window open detection)

They remain available in the GUI and the API shows a lastseen timestamp, however the updated value does not match, nor detects it the correct vibration.

manup, 2020-11-18:
Is this with a version [v2.5.87](https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.5.87) or later?
The vibration sensor got a fix in there, just checked mine with v2.6.0 it is working.

timbru31, 2020-11-18:
I could've sworn that I was on .87, but apt says it's still .86. Going to test v2.6.0 and will report back (once it either works fine for a longer period or when one becomes buggy again).

Btw ConBee II with FW 26660700

aberkeley, 2020-12-21:
Hi i am on 2.0.8 with FW 26680700 Conbee ii. my sensor appears in deconz but isnt recognised nor is it appear in Phoscon (as you would expect)

[image]

[image]

SwoopX, 2020-12-21:
The device is not paired correctly/completely.

aberkeley, 2020-12-21:
> The device is not paired correctly/completely.

thx swoop - removed and re-pair all working now thx

Mimiix, 2020-12-21:
@peer69 As others reported it works again, can you confirm that it was fixed for you too?

aberkeley, 2020-12-22:
Hi - as of today i can report i now have the same issue. Having got the item working (thx swoopX) with a re-pairing. I had to move the device (so i deleted it and then attempted to re-add) - this is successful- now i get the same as the original poster:

[image]

the sensor is fine on deconz vnc view:

[image]

2.0.8 with FW 26680700

SwoopX, 2020-12-23:
Your device node descriptor looks odd. You might want to give this a try:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

bcutter, 2021-01-19:
Still not tracked down/fixed? Will implement one of those sensors shortly and don´t want to spend hours to find out "jap, same issue" :-)

bcutter, 2022-01-18:
Two of those are working just fine related to this issue.

They sometimes trigger without an actual movement/vibration ("phantom vibration" if that issue got not a name yet) which is pretty annoying.

But nothing substantial to add for this issue.

jjdave2002, 2022-01-18:
I do not have that phantom trigger, mine are once triggering and then always stayed in "vibration" mode, so they do NOT reset the vibration detection after the vibration was gone. Everything was fine until the update of the FW (which is neede for the zugbe 3.0 sensors). Any other ideas? Curious as they sensor are named as "door lock", or not?

manup, 2025-05-02:
Closing as the vibration sensor is now handled via DDF and received various fixes.

closed 2025-05-02 by manup