dresden-elektronik/deconz-rest-plugin #3458: Heiman TS0503A (Tuya Zigbee RGB LED Controller)

DerpyDoom, 2020-10-20:
## Device

- Product name: TUYA ZIGBEE RGB RGBW RGBCCT Led Controller
- Manufacturer: _TZ3000_c5drg1wn
- Model identifier: TS0503A
- Device type : Please remove all unrelated device types. 
  - Light

## Screenshots

[image]
[image]
[image]

Smanar, 2020-10-20:
Hello, this device don't work natively ?

DerpyDoom, 2020-10-20:
No it is not responding to color, or brightness changes. I am only able to sporadicly turn it on or off if I set the RGB values to Red.

Mimiix, 2020-10-20:
Missing version information of current deconz version.

DerpyDoom, 2020-10-20:
Sorry! I have not seen the version information required in the template.

Version: 2.05.84 / 14.09.2020

DerpyDoom, 2020-10-20:
I just experimented a little bit, and by setting the Hue on "Move to Hue" to different values between 0-254 I am able to control all the colors that are in the led strip. But not with the interface.

[image]

Smanar, 2020-10-20:
Ha perhaps a bug in reading color capabilies ?
Can you show the device JSON ?

DerpyDoom, 2020-10-20:
> Ha perhaps a bug in reading color capabilies ?
> Can you show the device JSON ?

Could you tell me where I can find it? I'm quite new to under the hood stuff from deconz.

Smanar, 2020-10-20:
On your third app if you have one ?
Else using directly the API https://dresden-elektronik.github.io/deconz-rest-doc/ ?

DerpyDoom, 2020-10-20:
I do have a third party app, well system (Home Assistant) if it's possible with that. Otherwise I have to work with the api.

Smanar, 2020-10-21:
Ok so I have asked, and it depend on deconz integration you are using ^^.

So yes the easier way is using the api

**If you are in local**
You just need an api key
`curl http://127.0.0.1:80/api/ -X POST -d '{"devicetype": "TestKey"}'`

And use the URL
`http://127.0.0.1:80/api/YOUR_API_KEY/lights`

DerpyDoom, 2020-10-23:
Hey Smanar, since I'm using the deconz Home Assistant integration, which IP do I have to use?
Because the one where HassIO is running on is not reacting to port 80.

Smanar, 2020-10-24:
For the ip and port, it s the same you are using for phoscon.

DerpyDoom, 2020-10-26:
Finally managed to get the JSON You wanted. This is the Controller in question.

`{
      "etag":"9260ad2ad3aebb72f86454a5455b53a4",
      "hascolor":true,
      "lastannounced":null,
      "lastseen":"2020-10-24T01:25Z",
      "manufacturername":"_TZ3000_c5drg1wn",
      "modelid":"TS0503A",
      "name":"Color dimmable light 20",
      "state":{
         "alert":"none",
         "bri":25,
         "colormode":"xy",
         "effect":"none",
         "hue":0,
         "on":false,
         "reachable":false,
         "sat":254,
         "xy":[
            0.1166,
            0.798
         ]
      },
      "swversion":null,
      "type":"Color dimmable light",
      "uniqueid":"08:6b:d7:ff:fe:d3:63:ad-01"
   }`

Smanar, 2020-10-26:
Ok, so congratulation, you have find a new "not zigbee compliant chinese stuff", cf https://github.com/Koenkk/zigbee2mqtt/issues/4020

>     Device only work with HS color mode
>     Able to turn it off with genOnOff: offWithEffect
>     Able to turn it on with genLevelCtrl: moveToLevelWithOnOff
>     Able to change color with lightingColorCtrl: moveToHue & moveToSaturation seperately
>     moveToHueAndSaturation does not work
>     Can't turn it on with any genOnOff command
>     No brightness control so far

Need to write a code part just for it ...

DerpyDoom, 2020-10-26:
I'd love to have support for that, I also have some programming experience in C# but i doubt that this will help in this situation :D

The thing is there could be coming more issues with the same device since this is currently the cheapest available (6€ in my case), the others cost around 15€

sonar98, 2020-11-26:
"5": {
		"etag": "d47e9d5b3db50b4459f4ed3e1bd203cc",
		"hascolor": true,
		"lastannounced": null,
		"lastseen": "2020-11-24T13:16Z",
		"manufacturername": "Heiman",
		"modelid": "TS0601",
		"name": "Color dimmable light 5",
		"state": {
			"alert": "none",
			"bri": 2,
			"colormode": "xy",
			"effect": "none",
			"hue": 65018,
			"on": true,
			"reachable": false,
			"sat": 127,
			"xy": [
				0,
				0
			]
		},
		"swversion": null,
		"type": "Color dimmable light",
		"uniqueid": "58:8e:81:ff:fe:db:7e:9a-01"
	}

closed 2020-12-25 by stale[bot]