dresden-elektronik/deconz-rest-plugin #3459: Group for aqara opple

Kristian8606, 2020-10-20:
## Describe the question or issue you are having

Hello i own OPPLE switch 2-gang my question is can these switches create a group like the IKEA remote with 5 buttons ?
I would very much like to replace all the switches in my home with aqara opple and creating a group is important to me so I will be able to control my lighting even if the raspberry pi crashes

## Environment

 - Host system: (Raspberry Pi)
 - Firmware version: (26660700)
 - deCONZ version: (2.05.86)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? No

ebaauw, 2020-10-20:
No, you cannot.

Kristian8606, 2020-10-20:
thank you for your response was disappointing and quick 😁

Kristian8606, 2020-10-20:
@ebaauw which switches create a group?

ebaauw, 2020-10-21:
Wireless switches with client (grey) _On/Off_, _Level Control_, _Color Control_, or _Scenes_ clusters control lights directly.  These clusters can be bound to a group address, so they send commands using groupcasts.  The API plugin eavesdrops on these commands, sets `config.group` in the switch’s `/sensors` resource, creates the `/groups` resource, and maps the command to a `buttonevent` value.

Wireless switches with server (blue) clusters (typically manufacturer specific) send notifications, just like a sensor.  These clusters are bound to the gateway, so they send the notification there.  The API plugin maps the notification to a `buttonevent` value.  The lights are controlled by the gateway, using rules.

The obvious advantage of controllers is that they continue to work when the gateway is down.  The disadvantage is that they only support a limited set of commands.  The commands sent by some pro-grade switches, like ubisys, can actually be configured, but consumer-grade switches  have hard-coded the commands in there firmware.  As the API maps these command to button events, you can still use gateway rules for more elaborate actions.

The advantage of sensors is that they allow for more elaborate button events, like initial press vs release, hold, double click, triple click, etc.

Most wireless switches act as controller, not requiring a gateway to function.  Some act as sensors.  Hue switches actually acts as controller and sensor at the same time.  The Opple can act as either, but not at the same time.  The API configures it as controller, to support the advanced button events.  In theory, we could use `config.mode` to select the mode for the Opple, but sending commands (to change the configuration) to wireless switches is a bitch, as they are only reachable during pairing and shortly after you press a button.  We’d have to use `config.pending`, to hold the change until the switch has become reachable.  And we’d have to add a second button map for controller mode.

Kristian8606, 2020-10-21:
Thanks for the comprehensive answer.  Do you have future plans to configure these opple settings

ebaauw, 2020-10-21:
Me personally?  Probably not, at least not before API v2.

closed 2020-11-24 by stale[bot]

capstan1, 2020-12-09:
Sorry to resurrect this. What is the official way to create controller groups? Ive never seen an option in phoscon. I set them all up using Hue Essentials. Afterwards some groups (Hue dimmer switch and ikea tradfri round remote) appear in phoscon, others not.
For the user this seems like it's not really supported/wanted by the developers.

Also, is there a list somewhere for remotes that support groups?

Mimiix, 2020-12-09:
@capstan1 It's because Aqara / Opple didn't implement Group clusters. So the developers of _those devices_ didn't want them :( . 

The wiki shows a bit, but is not comprehensive.

Kristian8606, 2020-12-10:
Ikea remotes create a group, but aqara ones obviously don't.  I create a group as follows, add a remote and then open the the old API client of the decon and there I see all the added remotes.  I add lights to the group on the remote and then it is visible in the phoscon.  after creating this group, the remote can control the bulbs in this group even without the need for a gateway

burok88, 2020-12-20:
@Kristian8606 this sounds very interessting. do i understand it right, that you are able to create a group where you bind the opple switch and add afterwards the bulbs/lights? Could you give a more detailed "How-To" to since i am also facing the opple group issue.
What do you mean with the old API client exactly?

Kristian8606, 2020-12-21:
I'm sorry I didn't express myself correctly.  Aqara opple cannot create a group.  I'm creating a group for IKEA remotes from the old web gui interface