dresden-elektronik/deconz-rest-plugin #3460: [Question] Which in-wall (German "Unterputz") zigbee actors for 240V (at least 1000W) sockets are supported?

7wells, 2020-10-21:
## Describe the question or issue you are having
Hello,

Are there any in-wall (German "Unterputz") zigbee actors for 240V (at least 1000W) sockets supported? I mean an actor that can be placed in-wall (=flush-mounted) behind the usual socket (if, of course, there is enough room). I have not found any in the compatibility lists, but maybe that does not mean much?

More generally, which zigbee standard/version (3.0?) is currently supported by the RaspBee II module?

I hope it is ok to ask these questions here as an "issue". If not or if this rather belongs to another forum, I kindly ask the moderators to move my post. And thank you for your hints regarding support/compatibility.

## Screenshots
NA

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26650700
 - deCONZ version: 2.05.82
 - Device: RaspBee II

### deCONZ Logs
NA

## Additional context
NA

Mimiix, 2020-10-21:
A quick search through the git repo got me this:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/638

Where someone confirms it works.

cs42, 2020-10-21:
I'm using mutliple D1 and a S4 and they are working quite well. Configuration in Phoscon is a bit strange, though. You assign groups to be switched directly in the switches section and not in the lighting group configuration. The S4 just works as on/off switch (at least I think), for the D1 at least the first button is working with dimming. This one doesn't need to be configured but works directly on the attached load.

Another strange point, that later on in Phoscon you can't see which groups are assigned to the buttons (you can see it in the old webapp, though). I think it might be better to assign all button of the S4 at once. Assigning only one at a time might destroy the configuration of the other buttons. Sometimes the groups "vanish" in Phoscon after assignment, as they are converted to Zigbee groups (visible on the old webapp). I haven't been able to identify, when this happens and when not (maybe if some scenes are defined in Phoscon it doesn't?). But as I've got a running configuration, I don't care too much.

There're many things you could configure in these devices. Ubisys has a quite extensive documentation about all the abilities, but deconz doesn't have a nice interface for that of course (then you'd need their gateway and software I think). Even though they are quite expensive, IMO those devices have a very good quality compared to many of the other ones available...

Mimiix, 2020-10-21:
@cs42 That's all on phoscon and a bit offtopic: But i do agree. Phoscon needs to be more comprehensive in these things. Phoscon removed the groups because "it was confusing" well, now it is even more. So i fully agree with you.  My advice: Use a third party (HA/Iobroker/FHEM/Openhab/ etc) for these things. 

The gateway is exactly the same software as the image you download. So there's no difference there ;)

7wells, 2020-10-21:
Thank you so much to all of you, I really appreciate your helpful and fast feedback! And sorry that I did not thoroughly look for S1 support first. *blush*

closed 2020-10-21 by 7wells