dresden-elektronik/deconz-rest-plugin #3461: Philips TV Firmware Compability + Phoscon

Z-Rick84, 2020-10-21:
Dear Forum-Members,

as there seems to be problems with some Philips-TV Firmwares (Ambilight+Hue) in combination with Phoscon / Conbee, I'd like to share my experience and gain the other user experiences here to complete a list of possibly configurations of Philips- and Phoscon/Conbee-Firmware to beware Users of not downgradable Updates.

This combination works fine for me:
**65OLED804 + TPM191E _ 001.001.114.005 (2019/11/11) + Phoscon (FW: 26330500 / Ver: 2.05.75 / 8.3.2020)**

Is there anyone, who uses a newer Philips-FW and can confirm, that it runs good aswell? Let's find out, which is the latest firmware, which is running well with Phoscon.

Thank you for sharing.

TPM191E _ 101.001.079.005 – Date: 2020/09/15 (? / not tested)
TPM191E _ 001.004.100.001 UK only – Date: 2020/07/27 (? / not tested)
TPM191E _ 001.004.100.000 – Date: 2020/07/01 (? / not tested)
TPM191E _ 001.002.173.010 – Date: 2020/04/20 (? / not tested)
TPM191E _ 001.002.173.008 – Date: 2020/03/04 (? / not tested)
TPM191E _ 001.002.173.004 – Date: 2019/12/02 (? / not tested)
**TPM191E _ 001.001.114.005 – Date: 2019/11/11 (Works / tested)**
TPM191E _ 001.001.114.002 – Date: 2019/09/30 (seems to work / not tested)
TPM191E _ 001.001.037.013 – Date: 2019/09/04 (seems to work / not tested)
TPM191E _ 001.001.037.011 – Date: 2019/08/14 (seems to work / not tested)
TPM191E _ 001.001.037.009 – Date: 2019/07/24 (seems to work / not tested)
TPM191E _ 001.001.017.004 – Date: 2019/05/29 (seems to work / not tested)
TPM191E _ 001.001.017.003 – Date: 2019/05/22 (seems to work / not tested)

Mimiix, 2020-10-21:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3088

Contains some related info!

closed 2020-11-24 by stale[bot]

unstressable, 2022-10-30:
55OLED754/12 with TPM196E_091.004.027.011 does not work. Also tried the latest from 30.10.2022