dresden-elektronik/deconz-rest-plugin #3463: I can't add Aqara Motion / Door sensor

Galiwyx, 2020-10-21:
Hello,

I switched from Raspebrry + Conbee II to a server on Docker (UnRAID).

I added without problem 2 Aqara Temperature Sensor. Works perfectly.

I have some trouble to add Aqara Motion Sensor and Aqara Door Sensor. I tried eveything, long / quick press on reset button. 

On Phoscon App, sensor isn't found : 

[image]

This is my node :

[image]

Smanar, 2020-10-21:
The 0x1DAB is a chinese sensor, perhaps tuya ? Realy generic brand, you can have more information if you clic on the second circle, and check in basic cluster, but I think the field will be empty. If you show the cluster list can be helpfull too.

For the Xiaomi sensor yes, you have some problem, the inclusion is not finished in the API, but in deconz too.

As you have the GUI, something you can try
1 - set phoscon in permit join again (add new sensor)
2 - on deconz clic on the first circle and resend command (read descriptor for exemple) up you have the second circle
3 - when you have the second circle, clic on it, go in basic cluster (0x0000) and read basic attribute again (the button "read")

Problem, thoses sensors are sleeping, si in same time you make a command in deconz you need to wake up the device (short press on button for exemple)

Or make the inclusion normaly and after the blink press shortly the button every second to prevent the device go to sleep.

Mimiix, 2020-10-21:
Hi,

Closing this issue. Not using the template.

closed 2020-10-21 by Mimiix

Galiwyx, 2020-10-21:
> 
> 
> The 0x1DAB is a chinese sensor, perhaps tuya ? Realy generic brand, you can have more information if you click on the second circle, and check in basic cluster, but I think the field will be empty. If you show the cluster list, can be helpfull too.
> 
> For the Xiaomi sensor yes, you have some problem, the inclusion is not finished in the API, but in deconz too.
> 
> As you have the GUI, something you can try
> 1 - set phoscon in permit join again (add new sensor)
> 2 - on deconz clic on the first circle and resend command (read descriptor for exemple) up you have the second circle
> 3 - when you have the second circle, clic on it, go in basic cluster (0x0000) and read basic attribute again (the button "read")
> 
> Problem, thoses sensors are sleeping, si in same time you make a command in deconz you need to wake up the device (short press on button for exemple)
> 
> Or make the inclusion normaly and after the blink press shortly the button every second to prevent the device go to sleep.

Thank you @Smanar, I resetted the gateway before I saw your message and everything is okay now !