dresden-elektronik/deconz-rest-plugin #3464: ETI/Commercial Electric downlights detected as Color dimmable instead of Color temperature

mcalderin, 2020-10-21:
After connecting ETI downlights, the API returns that they're color dimmable (which makes downstream apps like Phoscon) show the RGB selector instead of a simple color temperature selector.

This is reproduceable whenever I add a new ETI downlight of the same model (T62).

In Phoscon, you can see the RGB option, but the color temperature can not be chosen.

[image]

Host is a RaspBee II v2.05.84 w/ firmware 26610700.  No other USB or serial devices are connected.

Saw comments from @ebaauw in Discord, suggesting these issues be added here and that they can be addressed through a whitelist.

ebaauw, 2020-10-21:
Please use the template for asking device support, and include the GUI screenshots.  We need that info to verify that this is indeed the case, and to know how to whitelist the device.

mcalderin, 2020-10-22:
Sorry, I did try to follow the template.

Here's a GUI screenshot showing the light model info.

[image]

If there's anything else I can provide, please let me know.

mcalderin, 2020-10-22:
Same thing is happening w/ Eaton lights.  Here are some more screenshots.

[image]

[image]

[image]

Mimiix, 2020-10-22:
As @ebaauw Asked: Please follow the template.
Closing.

closed 2020-10-22 by Mimiix