dresden-elektronik/deconz-rest-plugin #3465: Lots of nodes on DeCONZ GUI + I can't add sensors

Galiwyx, 2020-10-21:
## Describe the bug
Hello,
I use DeCONZ / Homebridge on docker with Conbee II. I switched from Raspberry 2B. 

Earlier today I had a problem to add Aqara Motion sensor and Door sensor. I resetted the gateway. I added without problem 2 devices.

Now I want to add a third sensor (Aqara door senser). On Phoscon, I had the sensor but it doens't found. 

Moreover, When I check deCONZ GUI, I have a TON of nodes (84 right now, and it's grow every 5-10 seconds), I don't understand why I have a lot of nodes (I've only 5-6 sensors, and 4-5 lights on IKEA Tradfri).

## Expected behavior
Except to add the sensor and not viewing 84 nodes in DeCONZ GUI.

## Screenshots

[image]

## Environment
Version :  2.05.86 / 15/10/2020
Firmware : 26660700

 - Host system: PC on UnRAID
 - Running method: Marthoc Docker container
 - Firmware version: (26660700)
 - deCONZ version: (2.05.86)
 - Device: ConBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? If so: Which? USB dongle for a keyboard

### deCONZ Logs
Some spamming logs : 

```
19:49:18:363 Device TTL 5263 s flags: 0x6
19:49:18:424 0xE8AA nwk changed to 0x337B
19:49:18:581 0x6D3C nwk changed to 0x49EF
19:49:19:461 0x9E77 nwk changed to 0x1B9D
19:49:19:660 0xBC07 nwk changed to 0xBD83
19:49:22:053 0x61BA nwk changed to 0x0B4B
19:49:22:741 0x14DD nwk changed to 0xEDB4
19:49:23:320 0xB8F1 nwk changed to 0xB1A3
19:49:24:247 0xDA57 nwk changed to 0x733E
19:49:25:118 0x4E2E nwk changed to 0xF227
19:49:26:799 0xB718 nwk changed to 0xE4F4
19:49:27:050 0x405F nwk changed to 0x1240
19:49:27:539 0xF702 nwk changed to 0xB14E
19:49:28:339 channel 25 does not match channel mask 0x00000800 (TODO)
19:49:30:658 0x8EE4 nwk changed to 0xDDC9
19:49:36:655 0x00D4 nwk changed to 0x732B
19:49:37:718 0x48D2 nwk changed to 0x4D6F
19:49:38:339 channel 25 does not match channel mask 0x00000800 (TODO)
19:49:39:307 0xFE41 nwk changed to 0x069A
19:49:39:771 0x5809 nwk changed to 0x1807
19:49:42:384 0x44CF nwk changed to 0xB4A3
19:49:42:495 0xBD83 nwk changed to 0x0B60
19:49:44:128 0x58DB nwk changed to 0xDDEB
19:49:44:687 0x14F7 nwk changed to 0xCFDD
19:49:45:328 0xD13C nwk changed to 0x5256
19:49:45:332 0x3780 nwk changed to 0xDBA0
19:49:48:339 channel 25 does not match channel mask 0x00000800 (TODO)
19:49:48:357 0x7D67 nwk changed to 0xD47E
19:49:48:407 0x918D nwk changed to 0x9E6E
19:49:49:989 0x839D nwk changed to 0xDDB5
19:49:53:645 0x2CB3 nwk changed to 0xF612
19:49:55:476 0x62F4 nwk changed to 0xC9E0
19:49:56:528 0xC5E7 nwk changed to 0x1F32
19:49:56:909 0x0B4B nwk changed to 0xB3D6
19:49:58:339 channel 25 does not match channel mask 0x00000800 (TODO)
```

Thank you

Mimiix, 2020-10-21:
Those devices are the ones paired to your gateway before. Therefore, they are now orphaned. You can still see them. 

If you re-pair them, they will disappear.

Galiwyx, 2020-10-21:
Hello mimions,

I never had 85-100 devices ! If I count I have 6 on Conbee and 7-10 on IKEA Tradfri. Is it possible to remove all orphans devices because there a all repaired except 1 Aqara Door sensor.

Smanar, 2020-10-21:
How have you reseted the gateway ?
You don't have a backup ?
Can you compare your setting in deconz with this one https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help ?

Galiwyx, 2020-10-21:
Hello Smanar,
I reseted th gateway on Phoscon web app. I didn't know there are a db file. I just need to delete db file ? No other thing ? 
No I don't have backup, it's not a problem for the moment, I'm learning Docker / deCONZ.

This is my configuration (I don't know if it's sensible data ^^) 

[image]

ebaauw, 2020-10-21:
Am I the only one noticing the suspicious mac addresses?  No clue what could be causing this, but there’s something very weird going on.  @manup, any idea?

manup, 2020-10-22:
This looks indeed very suspicious, like only mac address prefixes are there.
Assuming TC Address and MAC Address are equal the configuration looks fine.

Note while deleting zll.db file deCONZ must not be running.

Can you please post the log while the sensor is paired, to hopefully see which points are reached.

Galiwyx, 2020-10-22:
What, malicious MAC adresses ? It scares me quite a bit.

Yes, I’ll will recreate a new docker container of DeCONZ without any configuration and post logs. 

Thank you for your support.

Galiwyx, 2020-10-22:
This is the log file after a fresh start (no configuration at all) : https://privatebin.net/?598d9159d1997cb5#FRKhAjQtosrePe3GSrYHXwTunvkJFiTUbY975NN58qGB

After pairing an Aqara temperature sensor : https://privatebin.net/?58cda0f121dd40f0#6x5asRAgJTztsJ6r4TEg1BShUZgHMb5UoV16AqEZ3YGR

Smanar, 2020-10-22:
I don't see the bug on your last 2 logs, still have the problem ?

Galiwyx, 2020-10-22:
The the moment, I have only one device paired, this is my deCONZ GUI, is it an excepted mesh ? 

[image]

I'll try to add all my devices.

Galiwyx, 2020-10-22:
I tried to pairing a new device, after I clicked on "Add new sensor", deCONZ GUI was spammed with tons of new nodes (61 right now) : 

Logs: https://privatebin.net/?09134f9f945acab8#CHgmbNj7BebWnFp8jR8VJcYQEPoDjcMnxxiXEv7wkzB5

Galiwyx, 2020-10-23:
By curiosity, I tried to test on my windows 10 computer (familial computer). 

I lunched deCONZ app, opened Phoscon web App and started to add an sensor. 

Same issue, lots of nodes (+50) were discovered and deCONZ app closed itself after few seconds.

Can it be a firmware problem or a defective unit after some months ?

Smanar, 2020-10-23:
IDK, it s realy strange, you have a kind of "flood fake announce" with device using fake MAC adress.

>mac capabilities 0x8E

If the problem is from a device, it s probably a router, try to turn off your powered device.

Galiwyx, 2020-10-23:
I unplugged all my bulbs, battery from devices, unplugged my IKEA gateway. I just let my routeur from my ISP to connect to the local adress ton control Phoscon App.

I tried to add an Aqara sensor, nothing happens. 

I tried to add an IKEA bulb, 30 nodes appeared on deCONZ Gui : 
```
22:07:03:339 send permit join, duration: 59
22:07:03:339 send GP proxy commissioning mode
22:07:19:877 APS-DATA.indication from unknown node 0x47ED080000000000
22:07:19:878 new node - ext: 0x47ed080000000000, nwk: 0x9B2E
22:07:19:878 device announce 0x47ED080000000000 (0x9B2E) mac capabilities 0x8E
22:07:19:878 device announce 0x47ED080000000000 (0x9B2E) mac capabilities 0x8E
22:07:24:675 APS-DATA.indication from unknown node 0x3FEE080000000000
22:07:24:676 new node - ext: 0x3fee080000000000, nwk: 0x8A40
22:07:24:676 device announce 0x3FEE080000000000 (0x8A40) mac capabilities 0x8E
22:07:24:676 device announce 0x3FEE080000000000 (0x8A40) mac capabilities 0x8E
22:07:25:879 0x3FCE nwk changed to 0x9819
22:07:25:879 device announce 0x5862090000000000 (0x9819) mac capabilities 0x8E
22:07:25:879 device announce 0x5862090000000000 (0x9819) mac capabilities 0x8E
22:07:26:687 APS-DATA.indication from unknown node 0x3420080000000000
22:07:26:688 new node - ext: 0x3420080000000000, nwk: 0x3162
22:07:26:688 device announce 0x3420080000000000 (0x3162) mac capabilities 0x8E
22:07:26:688 device announce 0x3420080000000000 (0x3162) mac capabilities 0x8E
22:07:27:096 0x49C7 nwk changed to 0x33D2
22:07:27:096 device announce 0x43CB080000000000 (0x33D2) mac capabilities 0x8E
22:07:27:096 device announce 0x43CB080000000000 (0x33D2) mac capabilities 0x8E
22:07:27:856 APS-DATA.indication from unknown node 0x13EE080000000000
22:07:27:857 new node - ext: 0x13ee080000000000, nwk: 0x7FC7
22:07:27:857 device announce 0x13EE080000000000 (0x7FC7) mac capabilities 0x8E
22:07:27:857 device announce 0x13EE080000000000 (0x7FC7) mac capabilities 0x8E
22:07:30:180 APS-DATA.indication from unknown node 0x9ECC080000000000
22:07:30:181 new node - ext: 0x9ecc080000000000, nwk: 0x85D4
22:07:30:181 device announce 0x9ECC080000000000 (0x85D4) mac capabilities 0x8E
22:07:30:181 device announce 0x9ECC080000000000 (0x85D4) mac capabilities 0x8E
22:07:30:205 0x32E1 nwk changed to 0xC0E7
22:07:30:205 device announce 0x9DEF080000000000 (0xC0E7) mac capabilities 0x8E
22:07:30:205 device announce 0x9DEF080000000000 (0xC0E7) mac capabilities 0x8E
22:07:30:353 Current channel 25
22:07:30:361 Device TTL 2564 s flags: 0x7
22:07:30:462 device announce 0x43CB080000000000 (0x33D2) mac capabilities 0x8E
22:07:30:462 device announce 0x43CB080000000000 (0x33D2) mac capabilities 0x8E
22:07:30:770 device announce 0x13EE080000000000 (0x7FC7) mac capabilities 0x8E
22:07:30:770 device announce 0x13EE080000000000 (0x7FC7) mac capabilities 0x8E
22:07:32:185 APS-DATA.indication from unknown node 0xE660090000000000
22:07:32:186 new node - ext: 0xe660090000000000, nwk: 0x1619
22:07:32:186 device announce 0xE660090000000000 (0x1619) mac capabilities 0x8E
22:07:32:186 device announce 0xE660090000000000 (0x1619) mac capabilities 0x8E
22:07:33:176 APS-DATA.indication from unknown node 0x40EB080000000000
22:07:33:177 new node - ext: 0x40eb080000000000, nwk: 0x5307
22:07:33:177 device announce 0x40EB080000000000 (0x5307) mac capabilities 0x8E
22:07:33:177 device announce 0x40EB080000000000 (0x5307) mac capabilities 0x8E
22:07:36:132 APS-DATA.indication from unknown node 0x3F6E090000000000
22:07:36:132 new node - ext: 0x3f6e090000000000, nwk: 0xDD86
22:07:36:133 device announce 0x3F6E090000000000 (0xDD86) mac capabilities 0x8E
22:07:36:133 device announce 0x3F6E090000000000 (0xDD86) mac capabilities 0x8E
22:07:39:511 0xD2DC nwk changed to 0x95DA
22:07:39:511 device announce 0xE8EB080000000000 (0x95DA) mac capabilities 0x8E
22:07:39:511 device announce 0xE8EB080000000000 (0x95DA) mac capabilities 0x8E
```

It's demoralizing, I don't know what to do anymore.

ebaauw, 2020-10-24:
deCONZ does get device announcements, so that explains the nodes.  The issue is not in deCONZ, but in the Zigbee network, in the ConBee, or in the device you’re adding.  My money would be on the latter.

Galiwyx, 2020-10-24:
Hello @ebaauw,
I literally removed all batteries from my devices, removed bulbs and plugs. All my zigbee devices can’t send a signal due to their offline state.

I downgraded the firmware to the 26650700 version but the problem is still present. 

I think I’ll bought a second Conbee stick for test.

ebaauw, 2020-10-24:
My guess is that a single Zigbee device has gone crazy.  It is sending bogus device announcements.  These cause deCONZ to see bogus devices.  As soon as you open the network in search of new devices, deCONZ creates the nodes for these bogus devices.  You want to confirm that it is indeed a single device gone crazy, and then discard that device.

Once created, the bogus nodes will be persisted in the database and, worse, the bogus devices will still be in the ConBee’s neighbour table.  Meaning the bogus nodes will re-appear automagically after you delete them.  You need to stop deCONZ, and delete the database.  I would hope that  removing and re-inserting the ConBee will clear its neighbour table, but I’m not sure.  If the nodes re-appear after starting deCONZ, we’ll know it doesn’t.

Starting from a clean installation, using the 2nd ConBee, add the devices one by one.  Power on the device, and search for new devices.  The node should appear.  Reset the device if needed.  If it looks normal, wait a couple of minutes (for the database to be updated), shutdown deCONZ, and make a backup copy of the database.  If not, power down the device, shutdown deCONZ, remove and re-insert the ConBee, and restore the database from the last working backup copy.  Then proceed adding the other devices.  Hopefully you’ll end up with a normal network with all your devices, except the one gone crazy.

Galiwyx, 2020-10-24:
Hello @ebaauw, 
Thank you for you detailed answer, it helps me !

So, I unplugged the conbee stick, created a new docker, put the conbee stick and launched the container.

This is the what I see on deCONZ GUI, it's a good start !

[image]

After, I clicked on "Add new sensor", but **not powered any device** ! After 2min30, he found a device : 

[image]

I checked on the net the code for the manufacturer : **0x117c**. It can be the remote of IKEA Tradfri. I forgot to remove battery.
So I stopped the container. deleted the **zll.db**. Removed the conbee stick, and put again. Launched the container. **The unknown device disappeared** ! I started again the procedure to add a new device without power any device. I waited 3 minutes, and nothing was found !

Okay, now I add devices one by one ! Fingers crossed !!

Galiwyx, 2020-10-24:
Quick update, I added most of my Aqara devices. I have a problem with 2 on 3 Aqara Window / Door Sensor. The first one added without problem, but others two are found on deCONZ GUI but not found on Phoscon App : 

The light blinks in Blue one time then in Green when I push the button on the Sensor.

[image]

I tried to keep pushing the button every 2 seconds without success on Phoscon App.

Smanar, 2020-10-24:
>I tried to keep pushing the button every 2 seconds without success on Phoscon App.

Better to make short press (every 1/2 s), after the blink

Galiwyx, 2020-10-24:
Okay, I succeeded for the 2nd but I'm still trying for the 3rd! My poor fingers :D

Galiwyx, 2020-10-24:
1. Okay, I added my first **IKEA Tradfri bulb**, and I was spammed with weird nodes. 
2. Restored my backup, unplug / plug the conbee stick, weird nodes disappeared. 
3. I added a different IKEA bulbs, same thing, spammed with nodes.
4. Tested without power on the bulb, just searching new devices, spammed with nodes.

ebaauw, 2020-10-24:
What was the last device you added before add ing the Trådfri bulb?  If you just search for new devices, but don’t power on any lights, do you still get bombarded with bogus nodes?

Galiwyx, 2020-10-24:
I added an Aqara Window  / Door Sensor (very hard to detect, I tried like 30 times). 

I removed battery, started a new pairing discover : 

- Sensor > IKEA > ok (nothing was found)
- Sensor > Other > spamming with nodes then deCONZ crashed

Tested 2 times with same behaviour (restoration, unplug/plug between new pairing).

ebaauw, 2020-10-24:
This is beyond me. As far as I know there’s only one API call to search for new devices, and searching for IKEA vs Other in Phoscon shouldn’t make a difference.

Galiwyx, 2020-10-24:
Oh okay, that's really, really weird. If you ever want me to do further testing, let me know.

SwoopX, 2020-10-27:
@Galiwyx Did deconz cease with a segmentation fault again?

Galiwyx, 2020-10-27:
Not for the moment, 
Question ? Can it be a defective battery which create some crazy devices ? Because a tested all batteries and one is 0,72V and down to 0 V after 15-20 sec.

SwoopX, 2020-10-27:
No clue. What happens if you just take swap batteries for some devices? That should give us a hint if it could be related.

I've recently also seen a comparable behaviour described somewhere.

Galiwyx, 2020-10-27:
I didn't think of testing the bad battery.
I received a new battery today, I tried to add sensors and lights and I didn't get any weird devices. I'll do more testing.

Galiwyx, 2020-10-27:
Bad news, I added a Hue dimmer, and spammed with nodes (crashed after 10 sec) : 

```

18:40:01:354 [INFO] - No button handler for: lumi.sensor_motion.aq2 endpoint: 0x01 cluster: 0x0400 command: 0x0A payload[0]: 000
18:40:01:354 ZCL attribute report 0x00158D00045F9452 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
18:40:01:362 [INFO] - No button handler for: lumi.sensor_motion.aq2 endpoint: 0x01 cluster: 0x0406 command: 0x0A payload[0]: 000
18:40:01:362 ZCL attribute report 0x00158D00045F9452 for cluster: 0x0406, ep: 0x01, frame control: 0x18, mfcode: 0x0000
18:40:02:339 Set sensor check interval to 100 milliseconds
18:40:52:353 Current channel 25
18:40:52:361 Device TTL 834 s flags: 0x7
18:41:22:723 [INFO] - No button handler for: lumi.sensor_motion.aq2 endpoint: 0x01 cluster: BASIC (0x0000) command: 0x0A payload[0]: 001
18:41:22:723 ZCL attribute report 0x00158D00045F9452 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F
18:41:22:723 0x00158D00045F9452 extract Xiaomi special attribute 0xFF01
18:41:22:723 01 battery 2985 (0x0BA9)
18:41:22:723 03 temperature 26 °C
18:41:22:723 04 unknown 17320 (0x43A8)
18:41:22:723 05 RSSI dB (?) 32 (0x0020)
18:41:22:723 06 LQI (?) 34360131584 (0x0800060000)
18:41:22:723 0a unknown 0 (0x0000)
18:41:22:723 64 on/off 0
18:41:22:723 0b lightlevel 8 (0x0008)
18:41:31:439 sensor 5 (lumi.sensor_motion.aq2): disable presence
18:41:31:739 Set sensor check interval to 1000 milliseconds
18:41:52:352 Current channel 25
18:41:52:360 Device TTL 774 s flags: 0x7
18:42:13:339 send permit join, duration: 59
18:42:13:339 send GP proxy commissioning mode
18:42:14:455 APS-DATA.indication from unknown node 0x89CA080000000000
18:42:14:459 new node - ext: 0x89ca080000000000, nwk: 0x0D34
18:42:14:460 device announce 0x89CA080000000000 (0x0D34) mac capabilities 0x8E
18:42:14:460 device announce 0x89CA080000000000 (0x0D34) mac capabilities 0x8E
18:42:14:523 APS-DATA.indication from unknown node 0x97EB080000000000
18:42:14:523 new node - ext: 0x97eb080000000000, nwk: 0x8D1E
18:42:14:523 device announce 0x97EB080000000000 (0x8D1E) mac capabilities 0x8E
18:42:14:523 device announce 0x97EB080000000000 (0x8D1E) mac capabilities 0x8E
18:42:15:205 APS-DATA.indication from unknown node 0x4369090000000000
18:42:15:205 new node - ext: 0x4369090000000000, nwk: 0x7D16
18:42:15:205 device announce 0x4369090000000000 (0x7D16) mac capabilities 0x8E
18:42:15:205 device announce 0x4369090000000000 (0x7D16) mac capabilities 0x8E
18:42:15:671 APS-DATA.indication from unknown node 0x7D69090000000000
18:42:15:671 new node - ext: 0x7d69090000000000, nwk: 0xD740
18:42:15:671 device announce 0x7D69090000000000 (0xD740) mac capabilities 0x8E
18:42:15:671 device announce 0x7D69090000000000 (0xD740) mac capabilities 0x8E
18:42:16:119 ZDP discovery done in 1660 ms18:42:16:745 APS-DATA.indication from unknown node 0xE530080000000000
18:42:16:745 new node - ext: 0xe530080000000000, nwk: 0xFA4E
18:42:16:745 device announce 0xE530080000000000 (0xFA4E) mac capabilities 0x8E
18:42:16:745 device announce 0xE530080000000000 (0xFA4E) mac capabilities 0x8E
18:42:16:934 APS-DATA.indication from unknown node 0xB569090000000000
18:42:16:934 new node - ext: 0xb569090000000000, nwk: 0xA6BA
18:42:16:934 device announce 0xB569090000000000 (0xA6BA) mac capabilities 0x8E
18:42:16:934 device announce 0xB569090000000000 (0xA6BA) mac capabilities 0x8E
18:42:17:074 APS-DATA.indication from unknown node 0x1162090000000000
18:42:17:074 new node - ext: 0x1162090000000000, nwk: 0x85F7
18:42:17:074 device announce 0x1162090000000000 (0x85F7) mac capabilities 0x8E
18:42:17:074 device announce 0x1162090000000000 (0x85F7) mac capabilities 0x8E
18:42:18:532 APS-DATA.indication from unknown node 0x43CB080000000000
18:42:18:532 new node - ext: 0x43cb080000000000, nwk: 0xE6A4
18:42:18:532 device announce 0x43CB080000000000 (0xE6A4) mac capabilities 0x8E
18:42:18:532 device announce 0x43CB080000000000 (0xE6A4) mac capabilities 0x8E
18:42:18:641 APS-DATA.indication from unknown node 0x47ED080000000000
18:42:18:641 new node - ext: 0x47ed080000000000, nwk: 0x28FC
18:42:18:641 device announce 0x47ED080000000000 (0x28FC) mac capabilities 0x8E
18:42:18:641 device announce 0x47ED080000000000 (0x28FC) mac capabilities 0x8E
18:42:18:764 APS-DATA.indication from unknown node 0x9C31080000000000
18:42:18:764 new node - ext: 0x9c31080000000000, nwk: 0x8681
18:42:18:764 device announce 0x9C31080000000000 (0x8681) mac capabilities 0x8E
18:42:18:764 device announce 0x9C31080000000000 (0x8681) mac capabilities 0x8E
18:42:19:655 APS-DATA.indication from unknown node 0x9F5F090000000000
18:42:19:739 APS-DATA.indication from unknown node 0xC239
[... 50 lines cut ...]
18:42:34:320 new node - ext: 0x40eb080000000000, nwk: 0xC390
18:42:34:320 device announce 0x40EB080000000000 (0xC390) mac capabilities 0x8E
18:42:34:320 device announce 0x40EB080000000000 (0xC390) mac capabilities 0x8E
18:42:34:519 new node - ext: 0x962f080000000000, nwk: 0xB36F
18:42:34:653 APS-DATA.indication from unknown node 0xA6B6
18:42:34:853 APS-DATA.indication from unknown node 0x47EE080000000000
18:42:34:854 new node - ext: 0x47ee080000000000, nwk: 0x075C
18:42:34:854 device announce 0x47EE080000000000 (0x075C) mac capabilities 0x8E
18:42:34:854 device announce 0x47EE080000000000 (0x075C) mac capabilities 0x8E
18:42:34:955 APS-DATA.indication from unknown node 0x3B66090000000000
18:42:34:956 new node - ext: 0x3b66090000000000, nwk: 0xB5C3
[... cut ...]
18:42:35:430 APS-DATA.indication from unknown node 0xF0EC080000000000
18:42:35:430 new node - ext: 0xf0ec080000000000, nwk: 0x8D44
18:42:35:430 device announce 0xF0EC080000000000 (0x8D44) mac capabilities 0x8E
18:42:35:430 device announce 0xF0EC080000000000 (0x8D44) mac capabilities 0x8E
18:42:37:218 APS-DATA.indication from unknown node 0x2E30080000000000
18:42:40:790 APS-DATA.indication from unknown node 0x53CB080000000000
18:42:40:900 APS-DATA.indication from unknown node 0xF56C090000000000
Segmentation fault
```

[image]

manup, 2020-10-27:
Would be interesting to see a sniffer log on this. Currently I'd say the dimmer (if that's the only device which causes this) might have corrupted memory or something like that and doesn't send a valid mac address.

Haven't seen this problem before, I could add a filter in core to ignore all messages with an invalid mac address, but still this seems weird.

Galiwyx, 2020-10-27:
I bought the dimmer today, I think it's not the problem. 
I'll try again to remove all batteries from devices and power-off all bulbs / plugin.

SwoopX, 2020-10-27:
You got a Raspi to use? That could really help since it could allow us to eventually analyse the segfault and I can make you some code available turning deconz into a rudimentary sniffer.

Galiwyx, 2020-10-27:
I’m afraid I don’t understand. 
I use a Conbee 2 with the official docker. I have also a raspberry 2b if needed.

SwoopX, 2020-10-27:
The 2b would do. While preparing it, you should a) ensure it is able to generate core dumps ([see here)](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Enable-core-dumps-on-Raspbian) and b) use an alpha version of .86 [here](http://deconz.dresden-elektronik.de/raspbian/alpha/deconz-2.05.86-raspbian-buster-beta~git.c97006.a83c8c_armhf.deb).

The core dump should hopefully give us some further hints what might have gone wrong and where the segfault is coming from.

Galiwyx, 2020-10-27:
Thank you @SwoopX for your support, 

So if I understand correctly : 
- I install raspbian buster 
- I enable dumps
- I install deconz alpha
- I let the magie happens when I add a sensor 
- I share core dumps

Right ?

SwoopX, 2020-10-27:
Jup, that's the theory.

Galiwyx, 2020-10-27:
Okay, 
I’m sorry but what should I do ? 

[image]

SwoopX, 2020-10-27:
Dunno, why are you installing unraid and another deconz version? If it's about the messages, I guess that was only relevant for Raspbee

Galiwyx, 2020-10-27:
No, UNRAID is just the name of my USB key :D
deconz.deb is your file I downloaded.

Seems good, I started to add a devices, spammed with nodes. This is the coredumps :  

[core-deCONZ-sig11-user1000-group1000-pid1626-time1603835904.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5448326/core-deCONZ-sig11-user1000-group1000-pid1626-time1603835904.zip)

SwoopX, 2020-10-27:
Cool, thanks. Will have a look in a bit

Galiwyx, 2020-10-27:
Thank you again for your support, I really appreciate !

SwoopX, 2020-10-27:
Which device was it exactly you were trying to add at that time?

Here's the backtrace:

[image]

Galiwyx, 2020-10-27:
This time I tried to add a Hue Dimmer. But problems occurs with lots of devices.

SwoopX, 2020-10-27:
ok, I asked internally for this to get some attention. Meanwhile, could you please repeat the crash (no need for the core dump) with some more elaborate debug output?

Just have deconz started as follows `/usr/bin/deCONZ --dbg-error=2 --dbg-info=2 > debug.txt`. This generates more information and redirects it to a file which you can just share here.

Galiwyx, 2020-10-28:
Huh...hum, do you know where the file is created ?

Galiwyx, 2020-10-28:
First debug log : deCONZ crashed when I click on "Connect" : [debug1.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5448802/debug1.txt)

Second, I added a Hue Dimmer, discovered ~45 nodes and crashed : [debug2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5448806/debug2.txt)

SwoopX, 2020-10-28:
Thanks. I already have a gut feeling on this. Could you please install and try version .79 [here](http://deconz.dresden-elektronik.de/raspbian/beta/deconz-2.05.79-qt5.deb)?

Galiwyx, 2020-10-28:
It seems better ! No spamming nodes. Do you know where is the daatabase on raspbian ? Because deCONZ keeps 6 nodes (I added nothing), I want to try with deCONZ from scratch to be sure.

SwoopX, 2020-10-28:
Sounds promissing and contributes to my theory. It is under `$home/.local/dresden-elektronik/deCONZ/zll.db`

Galiwyx, 2020-10-28:
Thank you ! 
I tested from scratch to add : 
- a light
- a switch
- a sensor

Nothing was found and no spam ! 
I added after a Hue Dimmer without problem. It's a good news !!

Smanar, 2020-10-28:
So what is the mystery ?

Galiwyx, 2020-10-29:
> 
> 
> Thanks. I already have a gut feeling on this. Could you please install and try version .79 [here](http://deconz.dresden-elektronik.de/raspbian/beta/deconz-2.05.79-qt5.deb)?

Hello,
Is it the same version ([amd64-2.05.79](https://hub.docker.com/layers/marthoc/deconz/amd64-2.05.79/images/sha256-9de19eeec9cc5e2df0b55f7d0646d33d6b11a7aa81c0f2d92056e6136062e97e?context=explore)) on docker ?

SwoopX, 2020-10-29:
Codewise, it should.

Galiwyx, 2020-10-29:
hum, tried but spammed with nodes. weird.

Galiwyx, 2020-11-01:
Hello, 
Any update from the team ? 
Thank you :)

manup, 2020-11-13:
Can you please check with the latest version v2.5.88. There have been a bunch of bug fixes some related to memory corruption.
I'm not sure if it helps here but worth a try.

Galiwyx, 2020-11-13:
Hello,
Thank you for your answer, unfortunatly, the problem persists.

[image]

manup, 2020-11-13:
Very odd, I can only think of a rouge device which sends invalid MAC addresses, since it also is within the device announce frame.

I'll add a filter to ignore invalid MAC addresses in the upcoming beta (checking that the last for bytes of the MAC address ain't 0).

[79 more replies]

closed 2021-03-05 by github-actions[bot]