dresden-elektronik/deconz-rest-plugin #3467: Danfoss Ally: Setting operation mode via REST API?

CaCu15, 2020-10-22:
Hello deconz team, 

after I saw that you already implemented support for the new Danfoss Ally thermostats with https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2983 I just bought a  thermostat to test it and updated to the latest deconz version as well.
I'm using DECONZ REST API in a Node-Red based (self developed) MQTT gateway.  It works fine so far and I'm receiving status update events and I'm able to set the heating setpoint using the API. So first of all: Thanks for the quic support of new ZigBee devices!

But I've 1 question: According to  https://dresden-elektronik.github.io/deconz-rest-doc/sensors/#changeconfig thr API does support setting of the operation mode (e.g. off, auto, cool, heat, emergency heating, precooling, fan only, dry, sleep) using `PUT /api/<apikey>/sensors/<id>/config` with a body containing the desired `mode` at least for Tuya thermostats.   If I understand correctly, than the Danfoss Ally does support similar operation modes as well. Currently I receive an error response:

```
{"error":
        {"address":"/sensors/28/config/mode",
         "description":"parameter, mode, not available",
         "type":6}
}
```

 **Is there any chance that the API can support setting of the operation mode for Danfoss Ally Thermostats as well?** 

I could work around this in Node-Red by implementing the logic to maintain the temparature settings for each mode in Node-Red as well, but it would be a lot easier to use the capabilities of the thermostat device itself. 

## Environment
 Host System: Raspberry 3
 Running Method:  Rasbian Stretch
 Device: RaspBee 1
 Deconz Version:  2.05.86 / 15.10.2020
 Firmware: 26390500

CaCu15, 2020-10-22:
BTW: 
Just noticed, that the "preset" option is what I really want. I confused the options I think.

SwoopX, 2020-10-22:
> Is there any chance that the API can support setting of the operation mode for Danfoss Ally Thermostats as well?

No, because the device simply doesn't support it. Check out the REST API for the device to learn what it supports.

closed 2020-10-22 by SwoopX