dresden-elektronik/deconz-rest-plugin #3469: Philips Hue bulb don't upgrade

PrimaryInc, 2020-10-22:
Hi, 

I've downloaded Philips Hue motions sensor otua files and added them to Deconz.
Using OTAU Plugin I query the sensor(s) and they recognise firmware file converter -zigbee by deconz.
File Sensor-ATmega_6.1.1.27575_0012.sbl-ota
Update starts but stops in the middle and falls back to either queued or Idle.

This is what I see:
[image]

Stauts been stuck like this for about an hour or so, no new progress.
Motion sensor current firmware

[image]

DeConz version 2.05.86
Conbee I firmware 26390599
Running on windows 2019 server

Any ideas?

Mimiix, 2020-10-22:
You need them stay awake! :) 

Next time, use the template please (:

PrimaryInc, 2020-10-23:
Okay. Thanks :)

closed 2020-10-23 by PrimaryInc