dresden-elektronik/deconz-rest-plugin #3471: Blitzwolf BW-SHP13 still not showing up, despite it marked as in the .79 beta here #2988

fribse, 2020-10-22:
## Describe the bug

According to  #2988 it was solved as included in the .79 beta. I'm on 2.05.84, and it is showing as a smartplug in the VNC interface, but nothing is showing up in phoscon, not as a light, and not as a sensor

## Steps to reproduce the behavior

## Expected behavior

It should show up as a light and two sensors.

## Screenshots

[image]

## Environment

 - Host system: Intel NUC i5
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26580700)
 - deCONZ version: 2.05.84
 - Device: ConBee II
 - Do you use an USB extension cable: yes, usb3
 - Is there any other USB or serial devices connected to the host system? If so: aeotech z-wave stick

fribse, 2020-10-22:
Ok, this is odd, after watching it showing up as 0xAD4A, and not being able to find it in the REST API, I tried pairing it again (not resetting it before I did that), and it changed the name of the entity to Smart plug 9, and it showed up in Deconz???
It shows both 'smart plug' and 'sensors', so it is working now...

closed 2020-10-22 by fribse

carmelo42, 2020-11-27:
Hello, 
I juste paired a Blitzwolf BW-SHP13 on my deconz (latest .deb installed). I t appears as light, and I can control it, but nothing in sensor page. (in Phoscon Web App or with GET method : 

{
  "etag": "828ad70ec4d6a6c26c8c87de75e4408c",
  "hascolor": false,
  "manufacturername": "Heiman",
  "modelid": "TS0121",
  "name": "Smart plug 42",
  "state": {
    "alert": "none",
    "on": true,
    "reachable": true
  },
  "swversion": null,
  "type": "Smart plug",
  "uniqueid": "84:2e:14:ff:fe:36:68:19-01"
}

LukasA83, 2020-11-29:
Same here with actual beta version. Once I start the search for new sensors and query via vnc, the measuring will show up, but needs manual binding for updates.

SwoopX, 2020-11-29:
Yes, that's exactly the solution. Unfortunately, that's a device thing where we can do nothing about it. It used a not so standard way of reporting power values in predefined intervalls. If that report didn't get emitted or caught in the first pairing attempt, the corresponding sensors are not created.

So rule of thumb: If any sensor did not get created, just do a new sensor search in Phoscon (without resetting the device), but try to trigger the function which is missing (turn it on, apply a load, etc.).

Glad it works now!

LukasA83, 2020-11-30:
I understand, thanks. Is there a way to increase the reporting interval or trigger the "read" from Rest API?

SwoopX, 2020-11-30:
> Is there a way to change the reporting interval or trigger the "read" from Rest API?

Unfortunately, no based on any feedback provided. That'S the circumstances I meant earlier.

LukasA83, 2020-11-30:
Why not enable the Rest API to trigger read per node cluster? I believe that might be useful in other areas, too.

LukasA83, 2020-11-30:
I see in Zigbee2mqtt they are actively polling this plug to get frequent updates. Not sure what works best with deconz, but I can add some polling via rest api myself and this would be useable in other areas as well. Is that feasible @SwoopX ?

LukasA83, 2020-12-03:
Any chance to look into it?

SwoopX, 2020-12-03:
Thought about it at least. It would probably require some hack of the poll mechanism. Apart from that, I currently do see 0 benefit from that.

Polling is used if a device doesn't support reporting and therefore, try to receive status updates by the given frequency. So in other words: No polling -> unknown device state. Now, as far as I know, the plug supports autonomous reporting at a fixed rate of 2 mins. If you just received a report and then poll every 5 secs, you just get spammed by identical values and put some more load on your network until the next report (and then if repeats).

The only exception would be from my perspective, if the plug would internally refresh its values more frequently and provides them for reading.

LukasA83, 2020-12-03:
I can confirm the 2 minutes, not 30s. By pressing the read button in VNC I did receive new values like every 5s.

carmelo42, 2020-12-04:
> I can confirm the 2 minutes, not 30s. By pressing the read button in VNC I did receive new values like every 5s.

Can you explain what did you mean with "By pressing the read button un VNC" ? I can't figure out how to get consumption measure in phoscon webapp / jeedom / home assistant :(

Mimiix, 2020-12-04:
@LukasA83 Mine does a refresh every 30~ seconds , by itself. I've tested this yesterday for @SwoopX , got a video if you'd like.

LukasA83, 2020-12-04:
@carmelo42: Connect to VNC, select the respective node, use cluster "Electrical Measurement Cluster" and click "read" as shown in the screenshot.
[image]

@Mimiix: I did check it and mine is updating every 2 minutes. I'm using deconz (docker) ver. 2.07.00, see screenshot around basic information from the plug:
[image]

Any difference here?

LukasA83, 2020-12-09:
Any idea how to increase polling or trigger from the REST API

mhoogenbosch, 2021-01-15:
Am interested in this too. Doesn't really refresh for me at all.

userMak, 2021-01-18:
Hi all
I have 4 plugs BW-SHP13 with Deconz. I have connected the 3 of them in Home Assistant and I can see the readings and turn them on and off. The 4th although Deconz find it, I can only turn it on and off. No other entities created for watts & kwh. In Phoscon (of Home Assistant) I can see the readings in VPN). I tried also to search the sensors in Deconz (I had it plugged in hair dryer and one more time in a floorlight) but no entities created.  Since it is a zigbee device I only have Deconz in HA so I can not download the blitzwolf app to test it. Any help what more I can do to make the sensors work?

Mimiix, 2021-01-18:
I am locking this one, as we seem to have 2 issues on these now. 

Please all continue in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3788#issuecomment-762074302 

I'll get manuel to check it.