dresden-elektronik/deconz-rest-plugin #3472: ETI/Commercial Electric and Eaton downlights detected as Color dimmable instead of Color temperature

mcalderin, 2020-10-22:
## Describe the bug

After connecting ETI and Eaton downlights, the API returns that they're color dimmable (which makes downstream apps including Phoscon and homebridge-hue) show the RGB selector instead of a simple color temperature selector.

## Steps to reproduce the behavior

This is reproduceable whenever I add a new ETI or Eaton downlight.  Process to add lights:
1. Go to menu
2. Choose Lights
3. Choose Add new lights
4. Unpaired lights are automatically detected as expected
5. Lights are added as Color dimmable

## Expected behavior

Lights should be configured to allow for color temperature selection and not RGB.

## Screenshots

[image]

[image]

[image]

[image]

[image]

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian (HOOBS image)
 - Firmware version: 26610700
 - deCONZ version: 2.05.84
 - Device: RaspBee II
 - Do you use an USB extension cable: (yes / no) -- n/a
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

### deCONZ Logs
Sorry, not sure how to pull this info.  The Wiki link in the template leads to creating a new page.

## Additional context

Not sure this is relevant, but there's also a Wink hub and Hue w/ Homekit hub on premises.  The Hue bridge is used only for Hue lights and switches/sensors.  I've been transitioning everything away from the Wink hub; at this point, the only Zigbee device remaining there is a Smartthings door sensor.

Mimiix, 2020-10-22:
@mcalderin I've edited the link in the template! Good catch!
@ebaauw suggested opening a Device Request: 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Device+Request&template=device-request.md&title=Device+name

mcalderin, 2020-10-23:
OK, closing this to open device requests

closed 2020-10-23 by mcalderin