dresden-elektronik/deconz-rest-plugin #3473: Problem Mijia RTCGQ01LM Motion Sensors

jbustdel, 2020-10-22:
## Describe the question or issue you are having

## Screenshots

[image]

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

### deCONZ Logs

## Additional context

Mimiix, 2020-10-22:
You are missing routers in your network.  Xiaomi mijas tend to drop off once they are directly connected to the coordinator and if there are not routers.

jbustdel, 2020-10-22:
> You are missing routers in your network. Xiaomi mijas tend to drop off once they are directly connected to the coordinator and if there are not routers.

And what kind of router can i introduce?. Thanks.

Mimiix, 2020-10-22:
Should be fine, but others complain about that one. The goal is a healthy mesh.

jbustdel, 2020-10-22:
Is there any updated list?.  Thanks.

Mimiix, 2020-10-22:
Anything that is on 220V :)

Smanar, 2020-10-23:
Hello,
How many time the power off ?
You loose them too, if you just done a reboot ?
Only the same device and all of them ?

jbustdel, 2020-10-23:
> Hello,
> How many time the power off ?
> You loose them too, if you just done a reboot ?
> Only the same device and all of them ?
> 
> And BTW the device disapear in deconz ? No node at all ?

Hi,

The problem is in any case the raspbeerry is out of energy, in deconz's update or Os update. In hassio's reboot all is ok. the loss of devices is random, although motion sensors seem more prone. In deconz some nodes disappear completely and other nodes appear but without connection.

I think xiaomi devices go into sleep mode when they lose network and are not incorporated in any way once reset, only by resetting the sensor. In that case, the device is directly re-incorporated without having to be added.

Smanar, 2020-10-23:
>I think xiaomi devices go into sleep mode when they lose network and are not incorporated in any way once reset, only by resetting the sensor. In that case, the device is directly re-incorporated without having to be added.

Exactly, they can wait 1 hour before sending  a report, so they can be "unreachable" during this time, without connexion, it s normal.

But not normal node disapear, even if they leave the network, or if the API filter them (because a bug), they are still in the GUI.

Jaguaza, 2020-10-24:
I join this error.

Problem: some device cant not persistent after restart deconz container

Homeassisant Version: 0.116.4
Deconz Firmware:  26660700
Addons Version: 6.4.1
Purchase Date: 12/10/2020

Item:

I have 7 Hue bulbs, 6 Xiaomi temperature sensors, 10  xiaomi magnetic door / window, 2 flood sensors

Only xiaomi magnetic door / window nodes disappear.

Log:
[log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5432734/log.txt)

Smanar, 2020-10-24:
BTW what is exactly the "zombie mode" in deconz ?

Jaguaza, 2020-10-24:
> So they are marked as "unreachable" but are still present on deconz (the GUI, if you have it) ?
> 
> BTW what is exactly the "zombie mode" in deconz ?

I don't know what is zombie mode. 
In UI it appears but does not update the status. 
In deconz nodes disappear after reboot. 
In deconz it is not that they appear disconnected, it is that they simply are not.

jbustdel, 2020-10-25:
In the end, I went to ZHA and zero problems. Now I can update, shutdown and reboot and all devices come back quickly.

Jaguaza, 2020-10-25:
> In the end, I went to ZHA and zero problems. Now I can update, shutdown and reboot and all devices come back quickly.

ZHA???

Jaguaza, 2020-10-25:
I don't know, here we are reporting an error and nobody checks what is happening. Trust deconz, but maybe it's time to look at other alternatives and return my conbee2

jbustdel, 2020-10-25:
Zigbee Home Automation. It's a Hassio'integration. Deconz it's too difficult to understand. In ZHA only install integration, pair devices and go. Conbee 2 is compatible.

Jaguaza, 2020-10-25:
Thanks, we will have to go getting other alternatives, a shame that we also talk about conbee and then lose the nodes after a restart and do not worry about looking why this happens.

jbustdel, 2020-10-25:
Try ZHA, for me its ok now. If you have problems with Xiaomi devices, ask me, if I cant i hep you. The problem is initial pair, then all is ok.

Jaguaza, 2020-10-25:
> Try ZHA, for me its ok now. If you have problems with Xiaomi devices, ask me, if I can i help you. The problem is initial pair, then all is ok.

Thank you

joaoubaldo, 2020-12-02:
I am having the same issue, with Xiaomi motion sensors disappearing after reboots and so on. This makes the sensor network too fragile. Looks like some sensors tend to disconnect indefinitely if they lose connectivity with the parent node for a while.

Smanar, 2020-12-03:
>Looks like some sensors tend to disconnect indefinitely if they lose connectivity with the parent node for a while

Yep Xiaomi do that, but it need somes hours for that, generally if device disapear after a reboot it s mean the inclusion is not finished.

Jaguaza, 2020-12-03:
> Yep Xiaomi do that, but it need somes hours for that, generally if device disapear after a reboot it s mean the inclusion is not finished.

The problem that having the device running for several hours, you reboot and the node disappears.  But it is not xiaomi's problem, it is deconz's problem since zha and zigbee2mqtt, this does not happen.  I expected @manup to give a solution, but it never comes.

jbustdel, 2020-12-03:
With ZHA integration, no problems at all. Works perfect without issues.

Smanar, 2020-12-03:
> In the Phoscon interface they appear but do not detect movement, but in Deconz they disappear directly.

Ha ok, I have bad read, you mean the device disapear in deconz itself ? no node at all ?

Jaguaza, 2020-12-03:
> > En la interfaz Phoscon aparecen pero no detectan movimiento, pero en Deconz desaparecen directamente.
> 
> Ja, está bien, he leído mal, ¿quieres que el dispositivo desaparce en la propia desconcisión? ¿ningún nodo en absoluto?

The problem is that the node disappears from deconz after a reboot and never comes back.

MattWestb, 2020-12-03:
deCONZ have problem with direct chields can / is disappearing.
If having some routers its still happens but the network is working as one mesh network that is not doing without routers.
In Z2M and ZHA Rasb/CornBee have problem getting LIQ from direct connected childs and its well documented in both. I think the neighbor table is bad implemented in the router part of the firmware.

SwoopX, 2020-12-26:
Solution:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

closed 2020-12-26 by SwoopX