dresden-elektronik/deconz-rest-plugin #3475: Deconz 86 and Mijia Button

SAOPP, 2020-10-22:
At Deconz 86 hotfix, xiaomi mijia button is mistaken a long click like a single click.

Mimiix, 2020-10-23:
Closed. Not using correct template.

closed 2020-10-23 by Mimiix

SAOPP, 2020-10-23:
@Mimiix It was written via smartphone so it was a short info 'about issue. Now I can share more info.

```
9:44:
{
    "event_type": "deconz_event",
    "data": {
        "id": "hall_sofa_mi_button",
        "unique_id": "00:15:8d:00:01:6d:86:28",
        "event": 1002
    },
    "origin": "LOCAL",
    "time_fired": "2020-10-23T06:44:41.543487+00:00",
    "context": {
        "id": "3a7b419f14fb11ebb2f0d92301fc637c",
        "parent_id": null,
        "user_id": null
    }
}
9:44:
{
    "event_type": "deconz_event",
    "data": {
        "id": "hall_sofa_mi_button",
        "unique_id": "00:15:8d:00:01:6d:86:28",
        "event": 1001
    },
    "origin": "LOCAL",
    "time_fired": "2020-10-23T06:44:41.440208+00:00",
    "context": {
        "id": "3a6b7fb614fb11eb8bd8c382bdb861cb",
        "parent_id": null,
        "user_id": null
    }
}
9:44:
{
    "event_type": "deconz_event",
    "data": {
        "id": "hall_sofa_mi_button",
        "unique_id": "00:15:8d:00:01:6d:86:28",
        "event": 1000
    },
    "origin": "LOCAL",
    "time_fired": "2020-10-23T06:44:40.587226+00:00",
    "context": {
        "id": "39e957aa14fb11ebbc63afd4b24c1713",
        "parent_id": null,
        "user_id": null
    }
}
```
Here is a long click log above, from down to up.

Long click --> 1000 --> 1001 --> **1002**

1002 is a single click event, long click it should be 1001 and release will be 1003. But this does not happen, it is a mistake.The release now works like a single click.

SAOPP, 2020-10-23:
And one more thing... This button never had an event 1000 - I don't know what this event is.

Mimiix, 2020-10-23:
Doesn't change a thing. You still need to create a new bug report .