dresden-elektronik/deconz-rest-plugin #3476: Tradfri Color Light

darkmc80, 2020-10-23:
Hi
I'm using Tradfri Devices with Conbee Stick and therefore Deconz Binding for Openhab.

A was experimenting a lot on how to change colors and brightness. I think, but I'm not sure because I'm not a Pro, that there might be a problem with the binding. The brightness scale in color goes from 0 to 100 but the light brightens only to about 50% - either in rules or with the controller in sitemap. When adjusting color and brightness in Phoscon App, brightness changes normally from 0 to 100%.

Is this an issue?

Thank you

Jan

Version of the Binding: 231 │ Active │  80 │ 2.5.9                   │ openHAB Add-ons :: Bundles :: Dresden Elektronik d
Version of Openhab: openHAB 2.5.9-1 (Release Build)
Running on: Raspbian GNU/Linux 10 (buster), Kernel = Linux 5.4.51-v7+, Platform = Raspberry Pi 3 Model B Rev 1.2

Conbee: 2.05.84 / 14.9.2020, FW 26390500

Context: Not using Tradfri Gateway anymore because of stability issues... possibly in binding, restarting every 3 minutes

## Describe the question or issue you are having

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

### deCONZ Logs

## Additional context

Mimiix, 2020-10-23:
Hi @darkmc80 

I think you need to post this within the Openhab repo, as phoscon seem to work fine and they both use the same API.

Also: please fill the template correctly next time.

darkmc80, 2020-10-23:
Hi Mimiix Agien
Yes sorry, I thought that I was in the git Hub for Deconz Binding for Openhab...
Ok, I'm gonna search for it.
Jan

darkmc80, 2020-10-23:
Addressed in Openhab

closed 2020-10-23 by darkmc80