dresden-elektronik/deconz-rest-plugin #3480: Xiaomi Mijia round single key wireless WXKG01LM - Deconz 86 issue

SAOPP, 2020-10-23:
## Describe the bug
 1000 --> 1001 --> **1002**

1002 is a single click event, long click it should be 1001 and release will be 1003. But this does not happen, it is a mistake.The release now works like a single click.
-->

## Environment

 - Host system: (PC)
 - Running method: (Marthoc Docker container)
 - Firmware version: (264A0700)
 - deCONZ version: (2.05.86)
 - Device: (ConBee II / RaspBee I)
 - Do you use an USB extension cable: (no)

## deCONZ Logs
```
9:44:
{
    "event_type": "deconz_event",
    "data": {
        "id": "hall_sofa_mi_button",
        "unique_id": "00:15:8d:00:01:6d:86:28",
        "event": 1002
    },
    "origin": "LOCAL",
    "time_fired": "2020-10-23T06:44:41.543487+00:00",
    "context": {
        "id": "3a7b419f14fb11ebb2f0d92301fc637c",
        "parent_id": null,
        "user_id": null
    }
}
9:44:
{
    "event_type": "deconz_event",
    "data": {
        "id": "hall_sofa_mi_button",
        "unique_id": "00:15:8d:00:01:6d:86:28",
        "event": 1001
    },
    "origin": "LOCAL",
    "time_fired": "2020-10-23T06:44:41.440208+00:00",
    "context": {
        "id": "3a6b7fb614fb11eb8bd8c382bdb861cb",
        "parent_id": null,
        "user_id": null
    }
}
9:44:
{
    "event_type": "deconz_event",
    "data": {
        "id": "hall_sofa_mi_button",
        "unique_id": "00:15:8d:00:01:6d:86:28",
        "event": 1000
    },
    "origin": "LOCAL",
    "time_fired": "2020-10-23T06:44:40.587226+00:00",
    "context": {
        "id": "39e957aa14fb11ebbc63afd4b24c1713",
        "parent_id": null,
        "user_id": null
    }
}
```
-->

SwoopX, 2020-10-23:
You might be right on this, however, that must be ancient then. Assuming, the zigbee modelID starts with "lumi.sensor_switch", single and long press have the same parameters and that has been in there presumably for years.

```
                [1, "0x01", "ONOFF", "0x0a", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal release"],
                [1, "0x01", "ONOFF", "0x0a", "1", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Long release"],
```

To actually verify this, someone owning the device should remove that button map temporarily and share the corresponding debug output.

ebaauw, 2020-10-23:
The Mija button would be `lumi.sensor_switch`; the 2006 Aqara button `lumi.sensor_switch.aq2`.  There's also a variant with built-in vibration sensor, I think `lumi.sensor_switch.aq3` - I ordered those, but the freaking webshop on AliExpress delivered the wrong model (or rather: advertised the wrong description) and wouldn't even acknowledge that. And there is a 2008 Aqara button with the same commercial name as the 2006 but with a different _Model Identifier_, `lumi.remote.b1acn01`.  All of these support different button actions and use different ways of reporting these.

I only have the `lumi.sensor_switch.aq2`.  It reports single press (1002), double press (1004), and triple press (1005).  It uses attribute reporting on the server _On/Off_ cluster:
- On release it reports the _OnOff_ attribute (0x0000) - the report actually contains two records in the same message, one for _Off_ followed by one for _On_.  It doesn't matter how long the button was held before release;
- On release after double press, it reports value 2 for attribute 0x8000;
- On release after triple press, it reports value 3 for attribute 0x8000.

Note to self: looks like 0x8000 is missing from `general.xml`.

As I said, the other models use a different way of reporting the button actions.  I don't think the Mija reports long press; @manup hacked the code to measure the time between press and release and invalidate the match for the first entry, so it would fall back to the second entry.  That probably broke with the move to external button maps.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/cbef8cc540c5f7e4258b45f396fbf48a5cb21852/de_web_plugin.cpp#L3964-L3990

kooskaspers, 2020-10-23:
Same issue here. Since Deconz 86 I can't get a '1003' anymore (Button long release).

SwoopX, 2020-10-26:
Tbh, I don't see what the issue should be here. Other than changing the variable name and type, the code has remained untouched. As I see it, it even gets entered, otherwise you wouldn't see the `hold` event.

I wonder if it could be a matter of changed timing...

manup, 2020-10-27:
Fixed in the PR. Note that the out-commented check for an empty button map might cause other issues as well since `updateSensorNode(const deCONZ::NodeEvent &event)` and the button maps work in parallel.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/de_web_plugin.cpp#L7575-L7576

kooskaspers, 2020-11-04:
> Tbh, I don't see what the issue should be here. Other than changing the variable name and type, the code has remained untouched. As I see it, it even gets entered, otherwise you wouldn't see the `hold` event.
> 
> I wonder if it could be a matter of changed timing...

The issue seems to be resolved here. And I have t changed the container image of deconz. Very, very strange. I guess home-assistant plays a role here. I’ve upgraded hass a few days ago and that seem to have solved the problem.

closed 2020-12-19 by stale[bot]