dresden-elektronik/deconz-rest-plugin #3482: Cannot add any new switches

JasperAlgra, 2020-10-23:
## Describe the bug
I'm trying to add new switched to an already existing setup.
I've tried adding them via
-     Phoscon beta
-     normal webinterface
-     old interface

Three types of hardware:

-     IKEA E1743 (rectangular on / off remote) (brand new, don't know firmware)
-     IKEA E1524 (round remote) (latest firmware)
-     Xiaomi Oplle 3-gang (WXCJKG12LM)

I've already have a few of the E1743 and similar xiaomi switches joined and working fine. But I cannot add any new ones.

Trying to join fails in similar ways, the device can be seen in Deconz GUI but not in phoscon.

## Steps to reproduce the behavior
1. open phoscon
2. go to devices -> switches
3. press 'add new switch' select right type
4. reset switch (ikea; press button 4x)
5. wait and get error

## Expected behavior
I would expect switches to join the network and be available in phoscon

## Screenshots
result in phoscon:
[image]
Devices are showing in deconz (but not in phoscon):
[image]
[image]

firefox console of phoscon;
[image]
[image]

## Environment
- Version: 2.05.86 / 10/15/2020

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26580700
 - deCONZ version: 2.05.86
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
[20201023-adding_switch.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5428822/20201023-adding_switch.log)

## Additional context

If I can get more debug info or other helpfull info please let me know. I'm available to test whatever needs to be done.

JasperAlgra, 2020-10-23:
as @Mimiix saw errors regarding a 'blitzwolf bw-shp13 smartplug' in the log I've powered down this plug and tried again to pair without the plug present.
Attached is a new logfile:
[20201023-adding_switch_without_blitzwolf.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5428986/20201023-adding_switch_without_blitzwolf.log)

MattWestb, 2020-10-23:
My last experience of pairing the IKEA E1524 (round remote) is not using switch type "IKEA" then its not working and instead using "other" and after start the paring moving / twisting the device little now and then so its not "falling in sleep".
Its one of the tricky one getting working in deCONZ.

Your "ghost" devices is not successful paired devices and is best deleting then in deCONZ GUI before trying paring them again.

GeoCyberwolf, 2020-10-24:
I am having the same issue since updating to 2.05.86. I upgraded because I was unable to add my sonoff switches to groups, and now I cant pair them at all.

JasperAlgra, 2020-10-24:
@GeoCyberwolf I'll try to downgrade. Let you know if that helps

Would really like to get it working again. Other then lights I cannot add anything to the network at the moment.

JasperAlgra, 2020-10-24:
@GeoCyberwolf  Ok that helped! 
I'm now at deconz-2.05.83-qt5.deb with firmware 26660700 and it's back to normal.
@Mimiix this probably something interesting for the devs. I suspect something is broken regarding adding switches/sensors after 2.05.83

I've added a logfile of the successful attempt to join a ikea  E1743 to the network.
[adding_switch_on_2.05.83.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5433057/adding_switch_on_2.05.83.log)

GeoCyberwolf, 2020-10-24:
@JasperAlgra 

Interestingly I just managed to add a switch on .86. No idea how. I tried adding it through deconz gui first, which didnt work. I then closed the network in deconz gui, and tried adding a switch through the webapp. That then worked. So something about opening and closing the network in the deconz gui first changed something? I don't have logs I am afraid, as I am running deconz without logging as I dont want to kill yet another raspberry pi SD card.

Sonoff SNZB-01 still cant be added to groups in .86. though I have a separate issue open for that bug.

JasperAlgra, 2020-10-24:
I might have been celebrating to early.. I only got the add the one ikea switch and now it's back to the earlier state. Coudn't add another one (the E1743, E1524 or xiaomi opple).

If you want logging just stop  the deconz service 
` sudo systemctl stop deconz`

and start deconz by hand: 
`deCONZ --dbg-info=1 --dbg-zcl=1 --dbg-zdp=2 --dbg-error=1`.
You'll get the logging in the terminal so your SD-card is safe. This is the way I do it for logging. After that I'll stop deCONZ and restart the service.

Smanar, 2020-10-24:
Hello, can you show your setting panel in deconz, key F9, or compare with this one https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-and-configuration-restore-does-not-help pls.

SwoopX, 2020-10-25:
#3502

JasperAlgra, 2020-10-25:
@Smanar 
sure: 
[image]
[image]

That looks pretty similar to the screenshot in the link you've given.

SwoopX, 2020-10-25:
@JasperAlgra Your network/devices configuration seems to be a bit fishy. You got quite some entries like this:

`neigbor 0x00212effff0593d4 in different pan 0xDDDDDDDDDDDDDDDD`

You used z2m before? To me it looks like all your routers give that in return. Maybe leaving and joining via deconz GUI can resolve that.

JasperAlgra, 2020-10-25:
First of all; thanks for looking into this.

I've seen the messages but I couldn't figure out if these errors are bad or not. I did indeed use z2m before, but nearly all devices are on the new network. I'll try tomorrow to reset of shutdown any devices in my "box full of zigbee stuff" to see if that helps.

As a sidenote: I have to say, and this can be coincidence of course, but after downgrading to 2.05.83 the adding of sensors works pretty much flawlessly at the moment.

I'll keep you posted about the wrong PAN

JasperAlgra, 2020-10-27:
So I did a little more digging and I find it hard to figure out what's going on.

Unique occurrences of "in different PAN" logs:
```
32256846:Oct 27 12:02:55 raspberrypi deCONZ[309]: 12:02:40:390 neigbor	 0x000b3cfffef0b641 	in different pan 0xDDDDDDDDDDDDDDDD	switch  IKEA TRADFRI control outlet	woonkamer1                     000b3cfffef0b641-01  True  2.0.022    2020-10-27T11:03.00 Off
32069186:Oct 27 11:54:36 raspberrypi deCONZ[309]: 11:54:21:524 neigbor	 0x000b3cfffef7adb1 	in different pan 0xDDDDDDDDDDDDDDDD	switch  IKEA TRADFRI control outlet     woonkamer-3                    000b3cfffef7adb1-01  True  2.0.024    2020-10-27T10:57.00 Off
31835467:Oct 27 11:46:20 raspberrypi deCONZ[309]: 11:46:14:855 neigbor	 0x00158d0002f08b44 	in different pan 0xDDDDDDDDDDDDDDDD	sensor  Xiaomi lumi.remote.b286acn01    kantoor-dubbel                 00158d0002f08b44-01  True  20180809   2020-10-27T11:00.00
31640195:Oct 27 11:39:37 raspberrypi deCONZ[309]: 11:39:09:609 neigbor	 0x00158d00041df996 	in different pan 0xDDDDDDDDDDDDDDDD	sensor  Xiaomi lumi.sensor_cube.aqgl01  lumi.sensor_cube.aqgl01 16     00158d00042f2035-03  True  20160704   2020-10-27T10:53.00
32263400:Oct 27 12:03:11 raspberrypi deCONZ[309]: 12:02:55:779 neigbor	 0x00158d00042f2035 	in different pan 0xDDDDDDDDDDDDDDDD	sensor  Xiaomi lumi.sensor_cube.aqgl01	lumi.sensor_cube.aqgl01 16     00158d00042f2035-03  True  20160704   2020-10-27T10:53.00
31814419:Oct 27 11:45:53 raspberrypi deCONZ[309]: 11:45:44:058 neigbor	 0x00212effff0593d4 	in different pan 0xDDDDDDDDDDDDDDDD	?
31834953:Oct 27 11:46:20 raspberrypi deCONZ[309]: 11:46:14:855 neigbor	 0x04cf8cdf3c75c5e1 	in different pan 0xDDDDDDDDDDDDDDDD	sensor  Xiaomi lumi.remote.b286opcn01   OPPLE switch 1-gang            04cf8cdf3c75c5e1-01  True  20190730   2020-10-27T10:20.00
31632917:Oct 27 11:38:53 raspberrypi deCONZ[309]: 11:38:38:791 neigbor	 0x086bd7fffe5db459 	in different pan 0xDDDDDDDDDDDDDDDD	switch  IKEA TRADFRI control outlet     espresso                       086bd7fffe5db459-01  True  2.0.024    2020-10-27T10:58.00 On
32256333:Oct 27 12:02:55 raspberrypi deCONZ[309]: 12:02:40:390 neigbor	 0xccccccfffeb2f6be 	in different pan 0xDDDDDDDDDDDDDDDD	switch  IKEA TRADFRI control outlet     buitenlampjes                  ccccccfffeb2f6be-01  True  2.0.022    2020-10-27T10:59.00 Off
31637475:Oct 27 11:39:37 raspberrypi deCONZ[309]: 11:38:54:156 neigbor	 0xccccccfffeb4ada6 	in different pan 0xDDDDDDDDDDDDDDDD	switch  IKEA TRADFRI control outlet     woonkamer-2                    ccccccfffeb4ada6-01  True  2.0.022    2020-10-27T10:56.00 Off
```

All my working devices (from API):

```
switch  SONOFF BASICZBR3                         spiegelverwarming              00124b001da526f5-01  False None       2020-10-27T10:17.00 On
sensor  Xiaomi lumi.remote.b186acn01             schakelaar-spiegel             00158d00040e1259-01  True  20180809   2020-10-27T10:17:08
sensor  Xiaomi lumi.remote.b286opcn01            OPPLE switch 1-gang            04cf8cdf3c75c5e1-01  True  20190730   2020-10-27T10:20.00
sensor  IKEA TRADFRI on/off switch               schakelaar-espresso            588e81fffe3ec5d2-01  True  2.2.010    2020-10-27T10:26.00
sensor  IKEA TRADFRI on/off switch               schakelaar-kledingkast         680ae2fffeaa1bc8-01  True  2.2.010    2020-10-27T10:31.00
sensor  Xiaomi Temperature/Humidity (Round)      temp-humi-5-humidor            00158d0004066edd-01  True  20160516   2020-10-27T10:35:11 19.34C, 72.04%
sensor  Xiaomi Temperature/Humidity (Round)      temp-humi-2-slaapkamer         00158d00041025f6-01  True  20160516   2020-10-27T10:37:34 65.14%, 17.99C
sensor  Xiaomi Temperature/Humidity (Round)      temp-humi-4-buiten             00158d00041df92d-01  True  20160516   2020-10-27T10:41:56 12.93C, 83.05%
sensor  Xiaomi Temperature/Humidity (Round)      temp-humi-6-woonkamer_voor     00158d000413274a-01  True  20160516   2020-10-27T10:48:01 15.84C, 76.01%
sensor  Xiaomi Temperature/Humidity (Round)      temp-humi-1-kantoor            00158d00044b489c-01  True  20160516   2020-10-27T10:48:52 19.68C, 56.85%
sensor  Xiaomi Temperature/Humidity (Round)      temp-humi-3-entree             00158d00041df996-01  True  20160516   2020-10-27T10:49:51 17.77C, 62.73%
sensor  Xiaomi Temperature/Humidity (Round)      temp-humi-8-woonkamer_achter   00158d000410e17a-01  True  20160516   2020-10-27T10:50:15 18.68C, 67.01%
sensor  Xiaomi lumi.sensor_cube.aqgl01           Mi Magic Cube                  00158d00042f2035-02  True  20160704   2020-10-27T10:53.00
sensor  Xiaomi lumi.sensor_cube.aqgl01           lumi.sensor_cube.aqgl01 16     00158d00042f2035-03  True  20160704   2020-10-27T10:53.00
switch  IKEA TRADFRI control outlet              woonkamer-2                    ccccccfffeb4ada6-01  True  2.0.022    2020-10-27T10:56.00 Off
switch  IKEA TRADFRI control outlet              woonkamer-3                    000b3cfffef7adb1-01  True  2.0.024    2020-10-27T10:57.00 Off
switch  IKEA TRADFRI control outlet              espresso                       086bd7fffe5db459-01  True  2.0.024    2020-10-27T10:58.00 On
switch  IKEA TRADFRI control outlet              buitenlampjes                  ccccccfffeb2f6be-01  True  2.0.022    2020-10-27T10:59.00 Off
sensor  Xiaomi lumi.remote.b286acn01             kantoor-dubbel                 00158d0002f08b44-01  True  20180809   2020-10-27T11:00.00
[... cut ...]
switch  IKEA TRADFRI control outlet              woonkamer1                     000b3cfffef0b641-01  True  2.0.022    2020-10-27T11:03.00 Off
sensor  IKEA TRADFRI remote control              schakelaar-slaapkamer-1        000d6ffffe4a7dac-01  True  2.3.014    2020-10-27T11:03.00
sensor  IKEA TRADFRI on/off switch               schakelaar woonkamer           90fd9ffffea7f69c-01  True  2.2.010    2020-10-27T11:04.00
```

So all but one of the devices mentioned in the "different PAN" log are normal working devices. 
Can this message be the cause of anything or am I chasing ghosts?

SwoopX, 2020-10-27:
Not really my turf, but according to [this article](https://www.eetimes.com/zigbee-applications-part-3-zigbee-pans/), it could be part of the problem, at least theoretically.

> Extended PAN IDs (EPIDs) are 64-bit numbers that uniquely identify a PAN. ZigBee communicates using the shorter 16-bit PAN ID for all communication except one. The beacon response issued as the result of a beacon request contains an Extended PAN ID to allow a node that wishes to join a network to pick exactly the right one. 

@manup Maybe you can share your thoughts around this?

manup, 2020-11-13:
Can you please try again with latest version v2.5.88, quite some fixes in there.

closed 2020-12-19 by stale[bot]

popy2k14, 2021-08-31:
Hey Guys,

exact same issue here with my conbee II network running newest:

2.12.06 / 19.8.2021
26720700

Just cant repair an Shutter switch (same as On/Off) after it's battery was running out.
Yes, the same switch was connected to netwrok before and working as it should.
I am pressing 4 times the button in 5 secs and the led of the switch begins to breath red.

A question regarding the first screenshot, i just have those:
[image]

And no "On/Off" Switch. Is this supposed to be so in the new version?

Any hint's?
Which log do you need to get behind the issue?

thx

Smanar, 2021-08-31:
And you have no message in phoscon ? (the picture is just an indication, don't take care at it) when you make the procedure ?

popy2k14, 2021-09-01:
Sure, it say's "Verbindung fehlgeschlagen" (german for: connection failed) after 3 minutes (network open time).
I tried re-pair a few times without success.
Then i deleted the old device and tried again 10 times.
No chance, it just does'nt re-pair.

Any hints?

Smanar, 2021-09-01:
Are you using the GUI, to check for model id ?
Else can you share some logs with "info" and "info-l2" ?

popy2k14, 2021-09-25:
@Smanar I have treid i multiple times a week without success (also with restarting deconz, RPI, ...)
Today i stopped deconz-gui again, but started it with debug info level 2 and piped it into a file.

What should i say, open network as always in phoscon and it paired instantly!?

Don't know the issue was but now it's paired and working again.
Sorry, i can't provide an debug log with the issue in it.

Smanar, 2021-09-26:
Lol
NP, if now it s working, it s the principal.

Cyriuz, 2022-03-28:
I have also switched from z2m and get spammed with "in different pan", did anyone figure out how to solve it? I've tried re-pairing the devices but it still set on using `0xDDDDDDDDDDDDDDDD` not sure what else I need to reset.

Mimiix, 2022-03-28:
> I have also switched from z2m and get spammed with "in different pan", did anyone figure out how to solve it? I've tried re-pairing the devices but it still set on using `0xDDDDDDDDDDDDDDDD` not sure what else I need to reset.

Check the forum , there's some people having the same :)

popy2k14, 2022-03-28:
Can you please provide a link to the thread?
I am also interested in this topic.
Maybe i have that issue again sometime :-)

Mimiix, 2022-03-28:
https://forum.phoscon.de

Mimiix, 2022-03-28:
Feel free to open an own topic.  This is not really the place to discuss this 😅