dresden-elektronik/deconz-rest-plugin #3485: Add touchlink support on conbee II

zoic21, 2020-10-23:
## Feature request type

 Can you add touchlink support on conbee II like conbee I ?

## Description

In conbee I we can make reset off touchlink device but it's not possible on conbee II, can you re-add this feature ?

## Considered alternatives

## Additional context

MattWestb, 2020-10-23:
Please also for RaspBee II need this funktion for working with lights and the great working old web app.

Larsn1, 2020-10-24:
That would be great

JakobTewes, 2021-01-07:
would love that too ;-)

JakobTewes, 2021-01-07:
Found [this issue](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/337), in which someone states, that LivingColors or TouchLink inclusion is not possible.
Seems to have something to do with philips proprietary protocol and missing encryption keys...

manup, 2021-01-14:
The Touchlink support for ConBee II / RaspBee II is on the back log, it doesn't have a priority and ETA yet since we currently focus on fixing and improving other parts which are considered more urgent.

Touchlink is mostly useful for Hue mains powered devices since these don't provide a reset sequence. As workaround the Hue Dimmer switch can be used to send the reset command — this has also the benefit of being mobile, so the light doesn't need to be near the gateway.

SteGe88, 2021-01-28:
+1 went through a lot of trouble to get my Living Color connected to mijn ConBee II, before I saw that the touchlink function is not working on the Conbee II.

klodeckl, 2021-01-30:
+1

I have this plug: https://www.amazon.de/Philips-6916531PH-Living-Whites-Leuchten-Adapter/dp/B005ER7MX2
I think it also uses touchlink, right?
I want to buy a Conbee 2, but need that feature.
Do I also need it for Phlips Hue Iris and Bloom?

Smanar, 2021-01-31:
IDK for this device, but for me, if you have philips stuff, the only method that work at 100% is the touchlink feature, but you can too just buy a philips remote to do that.

JT089, 2021-01-31:
@Smanar, did you get (for example the wallplug, @klodeckl mentioned) the „pre-hue“ stuff to work with phoscon via touchlink remote? jow did you include the remote?

ebaauw, 2021-01-31:
You guys are mixing two things: touchlink search/reset vs touchlink linking.
- Touchlink reset, as the name suggests factory-resets a Zigbee device over touchlink.  This is supported by deCONZ, and by the REST API plugin for the RaspBee and ConBee.  As Manuel indicates, it’s on the backlog to add support for the RaspBee II and ConBee II.  I wouldn’t expect support in Phoscon before that;
- The first-generation pre-Hue Philips Living Color and Living White devices don’t support Zigbee ZLL nor ZHA in full.  Note that these devices are over 10 years old; this does **not** apply to any Hue-branded version.  In particular, they don’t support joining a Zigbee network the regular way.  They could be touchlink-paired to the Hue bridge, using the included remote, similar to how the Trådfri hub needs a controller to pair lights.  deCONZ does **not** support touchlink pairing, and afaik there are no plans to change that.

Smanar, 2021-01-31:
> The first-generation pre-Hue Philips Living Color and Living White devices don’t support Zigbee ZLL nor ZHA in full. Note that these devices are over 10 years old; this does not apply to any [...]

Right, sorry I m reading the manual of the plug, nothing about the touchlink RESET, reproducible with the philips remote.

systemcrash, 2021-03-26:
+1 pretty pleazzzeeeee 

> The Touchlink support for ConBee II / RaspBee II is on the back log, it doesn't have a priority and ETA yet since we currently focus on fixing and improving other parts which are considered more [...]

Hmm - if I cannot connect my Hue device to the ConBee2 in the first place, it seems that a reset sequence is more important than other functionality. Don't have a Hue Dimmer at hand... ☹️ 

Arrived here via [old forum thread](https://community.smartthings.com/t/dr-hue-is-in-and-the-bulbs-are-back-reset-hue-bulbs-to-factory/9856/7). I assumed that if the first iteration of the devices had this, then surely the second iteration would.

Is this touchlink process the same as this?:

https://www.nxp.com/docs/en/user-guide/JN-UG-3115.pdf#G17.1050870

systemcrash, 2021-03-26:
Or wait - is Touchlink init just a command sequence for the CB2?

Dreaming-Codes, 2021-07-21:
I would absolutely love that

systemcrash, 2021-08-16:
Arrived here again and see that I already wrote on this issue 😆 

But, please, being able to reset one's lights is important, no?

Mimiix, 2021-08-17:
https://forum.phoscon.de/t/feature-request-touchlink-support-on-conbee-ii/340

Created a feature request on it for the forum :)

systemcrash, 2021-08-18:
@Mimiix - could you ask that the request is to implement in the conbee2 firmware, as a start. The CB2 are in use all over the place outside of deconz stuff, so I don't mind testing and using without waiting for it to be implemented elsewhere.

systemcrash, 2022-01-19:
Any update on this feature request?

mcsrobert, 2022-05-07:
I also want this feature for my ConBee II.

Et3rnel, 2022-08-19:
I also would like to reset my Philips devices with Touchlink reset from my ConBee 2 ! 🙏🏻
Any progess on this feature request ?

dave-sausages, 2022-09-10:
+1 for the feature request

Bejjje, 2022-09-23:
Give us please

cedricnet, 2022-12-22:
Would be quite interested in the touchlink feature on raspbee II, useful for old philips living colors lamps.

Smanar, 2022-12-23:
No return on my side, faster to use the remote for that for the moment.

bitfl0wer, 2023-10-10:
> Would be quite interested in the touchlink feature on raspbee II, useful for old philips living colors lamps.
> 
> Is there any ETA now?

Afaik this exists on the RaspBee 2.

systemcrash, 2024-04-04:
*sigh*

cedricnet, 2024-04-05:
I bought a raspbee II and I can’t neither command old Philips living Colors
with it :-(

Smanar, 2024-04-05:
The remote is not working to reset them ?

systemcrash, 2024-04-05:
What remote 😉

ebaauw, 2024-04-06:
deCONZ doesn’t support device pairing over touchlink, not even on the original RaspBee and ConBee.  The old, pre-Hue Philips Living Colors (and Living White) lights and remotes do not work, and will not work with deCONZ.

deCONZ does support device discovery and reset over touchlink, but only on the original RaspBee and ConBee, as later devices lack firmware support for this.  This is an ancient feature, exposed only through the old web app.  Note that this feature is only useful, if the device is close enough (< 2m) to the RaspBee or ConBee.  You might need to power the Raspberry Pi with the RaspBee from a power bank, or use a 5m USB extension cable for the ConBee.

Note that not all devices support reset over touchlink.  And if they do, only for a short period after powering on.  I’ve only ever needed this for Hue lights, since they cannot be reset through some power on sequence.
You can issue a reset over touchlink, using a Hue dimmer switch.  Hold the On/Off and Hue buttons (On and Off buttons on the old model) for 10s, with the dimmer close (< 10cm) to the light.  Be patient and continue to hold when the light starts reacting, until it remains steady at full brightness.

systemcrash, 2024-04-06:
Yep, and that's what people in this thread are asking for in the conbee 2 firmware, likely because they don't have those remotes. I haven't.

Mimiix, 2024-04-08:
Hi guys,

I am closing the issue here, as we stopped collecting Feature requests on the Github and moved those to the forums (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/5113). 

The forum version can be found here, with a response of Manup: https://forum.phoscon.de/t/feature-request-touchlink-support-on-conbee-ii/340/5

Please proceed this feature request there :)

Closing and locking.

closed 2024-04-08 by Mimiix