dresden-elektronik/deconz-rest-plugin #3488: Installation problem - debian - status=218/CAPABILITIES - Operation not permitted

Rob1983dev, 2020-10-23:
## Describe the question or issue you are having

Hello All!
I tried to install deconz for several hours.. but know I don't know what to do - I need your help!
Result is always the same, message:

```
sudo journalctl -u deconz

Oct 23 14:47:38 iobroker-pve systemd[1]: deconz.service: Service RestartSec=100ms expired, scheduling restart.
Oct 23 14:47:38 iobroker-pve systemd[1]: deconz.service: Scheduled restart job, restart counter is at 1237.
Oct 23 14:47:38 iobroker-pve systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Oct 23 14:47:38 iobroker-pve systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Oct 23 14:47:38 iobroker-pve systemd[44205]: deconz.service: Failed to apply ambient capabilities (before UID change): Operation not permitted
Oct 23 14:47:38 iobroker-pve systemd[44205]: deconz.service: Failed at step CAPABILITIES spawning /usr/bin/deCONZ: Operation not permitted
Oct 23 14:47:38 iobroker-pve systemd[1]: deconz.service: Main process exited, code=exited, status=218/CAPABILITIES
Oct 23 14:47:38 iobroker-pve systemd[1]: deconz.service: Failed with result 'exit-code'.
```

## Screenshots

FOR THE BOT!!! NO PICTURES! Hopefully he/she/it is happy now!! :-)

## Environment

 - Host system: Debian 10 (buster)
 - Running method: CT, Proxmox
 - Firmware version: not checked
 - deCONZ version: 2.05.69 - but I tried several
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? no!

## deCONZ Logs
```
Oct 23 14:47:38 iobroker-pve systemd[1]: deconz.service: Service RestartSec=100ms expired, scheduling restart.
Oct 23 14:47:38 iobroker-pve systemd[1]: deconz.service: Scheduled restart job, restart counter is at 1237.
Oct 23 14:47:38 iobroker-pve systemd[1]: Stopped deCONZ: ZigBee gateway -- REST API.
Oct 23 14:47:38 iobroker-pve systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Oct 23 14:47:38 iobroker-pve systemd[44205]: deconz.service: Failed to apply ambient capabilities (before UID change): Operation not permitted
Oct 23 14:47:38 iobroker-pve systemd[44205]: deconz.service: Failed at step CAPABILITIES spawning /usr/bin/deCONZ: Operation not permitted
Oct 23 14:47:38 iobroker-pve systemd[1]: deconz.service: Main process exited, code=exited, status=218/CAPABILITIES
Oct 23 14:47:38 iobroker-pve systemd[1]: deconz.service: Failed with result 'exit-code'.
```

## Additional context
```
lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

I know what I should create an overwrite file - but only to be sure that this is not an issue - I changed (only port is changed):

```
sudo nano /lib/systemd/system/deconz.service

[Unit]
Description=deCONZ: ZigBee gateway -- REST API
Wants=deconz-init.service deconz-update.service

[Service]
User=1000
ExecStart=/usr/bin/deCONZ -platform minimal --http-port=8080
Restart=on-failure
StartLimitInterval=60
AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME

[Install]
WantedBy=multi-user.target
```

Further details:
```
root@iobroker-pve:~# groups:
root dialout
root@iobroker-pve:~# id
uid=0(root) gid=0(root) groups=0(root),20(dialout)

```
Created a "pi" user, too - because I thought.. maybe that is the problem:
```
root@iobroker-pve:~# id pi
uid=1001(pi) gid=1001(pi) groups=1001(pi),20(dialout),27(sudo)
```

Installation process (done with user root!):
```
wget http://deconz.dresden-elektronik.de/ubuntu/beta/deconz-2.05.69-qt5.deb
sudo dpkg -i deconz-2.05.69-qt5.deb
sudo apt --fix-broken install
sudo gpasswd -a $USER dialout
sudo systemctl enable deconz
sudo service deconz start
```

Tried the official repo, too - but .. same error message.

Hopefully it is only a small issue.

Thanks a lot!!!

Best regards
Rob

SwoopX, 2020-10-25:
Just real quick because it's late:

1. Why do you install version .69? We're currently at .86, so you're more than 16 versions behind 
2. Proxmox and deconz are not best friends
3. Systemd is trying to start the service with UID 1000, your user Pi as UID **1001** and presumably not in `dialout` and/or `sudo` group
4. You can start deconz with `/usr/bin/deCONZ -platform minimal --http-port=8080`, if it's just about getting it going.

Rob1983dev, 2020-10-25:
Hello @SwoopX ,

I think.. I found "the problem".
It was not a problem because of the user - it was a problem because of the using Proxmox with an LXC!
The message was always` "status=218/CAPABILITIES"` and something with `"Failed at step CAPABILITIES spawning /usr/bin/deCONZ: Operation not permitted"`.

After searching a lot I understood, that there might be problem permission problems between Proxmox and LXC. 
deconz needs: `AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME`
But (and I don't know why) the LXC with deconz missed the last one - CAP_SYS_TIME.
So I set the CAP_SYS_TIME manually:
`setcap "CAP_SYS_TIME=+ep" /usr/bin/deCONZ`

After that - the service was suddenly starting!
I did some more changes on the Proxmox host for the LXC in the config file - I added:
```
lxc.apparmor.profile: unconfined
lxc.cgroup.devices.allow: a
```
BUT! I am not an expert - I was a little bit desperate because nothing was working. So it might be enough to do setcap, but maybe it is necessary to add these small lines to the config of the LXC, too.

Anyway - deconz service started!

Next problem was: USB conbee2 not viewable in contains. Viewable with lsusb - but not with GCFFlasher_internal -l!
.. only if someone else has the same problem..
Searched a lot.. again.. :-/
Working solution:
https://doc.turris.cz/doc/en/public/deconz_lxc_howto - check the part "Tweak container environment"!
It is necessary to create a new dir in the container and add some data:

`mkdir -p /run/udev/data/`

```
cd /run/udev/data/
echo "E:ID_VENDOR_ID=1cf1
E:ID_MODEL_ID=0030" > c166\:0
```

After that - deconz is working with conbee2 in the LXC container!

Rob1983dev, 2020-10-25:
and yes, @SwoopX 
Now I am using the version .86!
Totally correct, Proxmox and deconz are not best friends - but now they must be! :-)

closed 2020-10-25 by Rob1983dev