dresden-elektronik/deconz-rest-plugin #3490: Sonoff SNZB-01 Unable to Pair or Add to group

GeoCyberwolf, 2020-10-23:
## Describe the bug
Unable to add Sonoff SNZB-01 Wireless switch to group in 2.05.84. I have updated to the latest beta version 2.05.86 and still could not add the switch to a group. I also can't pair the switches in 2.05.86, I can see the device announce in the logs when I reset the switch but it does not pair.

## Steps to reproduce the behavior
Try to add the switch to a group, switch does not show in the list of available switches. 

Try to pair the switch, switch does not pair.

## Expected behavior
Switch is available to add to groups
Switch pairs.

## Screenshots
[image]
## Environment
 - Host system: Rapberry Pi
 - Running method: Raspbian
 - Firmware version: 26520700
 - deCONZ version: 2.05.86
 - Device: RaspBee II
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
`pi@deconz:~ $ deCONZ --dbg-info=1
QCoreApplication::arguments: Please instantiate the QApplication object first
libEGL warning: DRI2: failed to authenticate
libpng warning: iCCP: known incorrect sRGB profile
19:10:45:739 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
19:10:45:753 CTRL. 3.27.219:10:45:988 dev /dev/ttyS0 (/dev/ttyS0)
19:10:45:988 ZCLDB init file /home/pi/.local/share/dresden-elektronik/deCONZ/zcldb.txt
19:10:46:522 parent process bash
19:10:46:523 gw run mode: normal
19:10:46:523 GW sd-card image version file does not exist: /home/pi/.local/share/dresden-elektronik/deCONZ/gw-version
19:10:46:523 sd-card cid: 035344534333324780ffffffff014509
19:10:46:555 DB sqlite version 3.27.2
19:10:46:559 DB PRAGMA page_count: 41
19:10:46:559 DB PRAGMA page_size: 4096
19:10:46:559 DB PRAGMA freelist_count: 2
19:10:46:559 DB file size 167936 bytes, free pages 2
19:10:46:560 DB PRAGMA user_version: 7
19:10:46:560 DB cleanup
19:10:46:565 DB create temporary views
18:10:46:660 started websocket server at port 443
18:10:46:674 [INFO] - Found file containing button maps. Parsing data...
18:10:46:682 [ERROR] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
18:10:46:697 [INFO] - Button maps loaded.
18:10:46:702 dlg action: Read binding table
18:10:46:703 found node plugin: libde_rest_plugin.so - REST API Plugin
18:10:46:714 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
18:10:46:777 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
18:10:46:837 dev /dev/ttyS0 (/dev/ttyS0)
18:10:47:213 New websocket 10.13.37.22:60295 (state: 3) 
18:10:47:258 New websocket 10.13.37.22:60297 (state: 3) 
18:10:47:930 dev /dev/ttyS0 (/dev/ttyS0)
18:10:47:933 COM check bootloader
18:10:49:941 COM timout query bootloader, assume application
18:10:50:011 Device firmware version 0x26520700 RaspBee II
18:10:50:147 unlocked max nodes: 200
18:10:50:329 Device protocol version: 0x010B
18:10:50:367 new node - ext: 0x00212effff052e3b, nwk: 0x0000
18:10:50:573 Tuya debug 7 : Missing manufacture name for 0x00124b002228c125
18:10:50:573 Tuya debug 7 : Missing manufacture name, till missing in DB.
18:10:50:574 SensorNode 3 set node 0x00124b002228c125
18:10:50:577 SensorNode 2 set node 0x00124b002228c125
18:10:50:581 Tuya debug 7 : Missing manufacture name for 0x00124b002228c125
18:10:50:581 Tuya debug 7 : Missing manufacture name, till missing in DB.
18:10:50:589 Tuya debug 7 : Missing manufacture name for 0x00124b002228d384
18:10:50:589 Tuya debug 7 : Missing manufacture name, till missing in DB.
18:10:50:589 SensorNode 5 set node 0x00124b002228d384
18:10:50:592 SensorNode 4 set node 0x00124b002228d384
[... 522 lines cut ...]
18:16:41:806 dev /dev/ttyS0 (/dev/ttyS0)
18:16:41:808 GW firmware version: 0x26520700
18:16:45:745 Current channel 11
18:16:45:768 Device TTL 5292 s flags: 0x7
18:16:51:156 ZCL attribute report 0x680AE2FFFEE3CA83 for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:16:51:239 ZCL attribute report 0x680AE2FFFEE3CA83 for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:16:51:293 ZCL attribute report 0x680AE2FFFEE3CA83 for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:16:51:776 dev /dev/ttyS0 (/dev/ttyS0)
18:16:51:777 GW firmware version: 0x26520700
^A18:17:00:878 ZCL attribute report 0x680AE2FFFEE34F1E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:17:01:790 dev /dev/ttyS0 (/dev/ttyS0)
18:17:01:794 GW firmware version: 0x26520700
18:17:10:556 ZCL attribute report 0x680AE2FFFEE34F1E for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:17:11:778 dev /dev/ttyS0 (/dev/ttyS0)
18:17:11:780 GW firmware version: 0x26520700
18:17:21:787 dev /dev/ttyS0 (/dev/ttyS0)
18:17:21:790 GW firmware version: 0x26520700
18:17:23:029 ZCL attribute report 0x680AE2FFFEE34C9B for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:17:23:081 ZCL attribute report 0x680AE2FFFEE34C9B for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:17:23:120 ZCL attribute report 0x680AE2FFFEE34C9B for cluster: 0x0300, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
18:17:28:078 [INFO] - No button handler for: TH01 endpoint: 0x01 cluster: 0x0402 command: 0x0A payload[0]: 000
18:17:28:078 ZCL attribute report 0x00124B002228D384 for cluster: 0x0402, ep: 0x01, frame control: 0x18, mfcode: 0x0000`

## Additional context
Zigbee network has a mixture of various brands of bulb, and a mix of xaiomi mijia and sonoff sensors, switches

GeoCyberwolf, 2020-10-24:
Update: after opening the network in deconz gui, trying to add the switch(which didnt work) then closing it. I attempted again to add the switch in the webapp, which then worked. So something about opening and closing the network in the gui allowed the switch to be added. 

I am still unable to add it to a group to control lights though. I am happy to send one of these switches to a developer for testing if it helps get these switches working. I have 4 of them and they are currently useless.

closed 2020-11-08 by manup

accelle17, 2020-11-14:
@GeoCyberwolf were you able to add the snzb-01 to a group after #3502 or 2.05.87?

GeoCyberwolf, 2020-11-19:
@accelle17 No, unfortunately it doesn't show on the list when I go to add a switch to a group. I tried removing and re-adding the switch since updating to .88

SwoopX, 2020-11-21:
Please check the old webapp and have a look if it shows up there.

accelle17, 2020-11-22:
> Please check the old webapp and have a look if it shows up there.

Great. can see it on the old web app. Should this be the behavior?

GeoCyberwolf, 2020-11-22:
@SwoopX How do I access the old webapp? I have just been browsing to the hostname of the raspberry pi deconz is running on.

SwoopX, 2020-11-22:
Should be http://YOURIP/login.html

@accelle17 yes. What you see there are the zigbee groups. By using them, the coordinator must not be online. Never understood why they are not properly exposed via Phoson though.

GeoCyberwolf, 2020-11-22:
@SwoopX Ah, that works, thanks. I can add lights to the switch group in the old webapp, and the switch will control those lights. 

Is there any way to get the switch working in the new webapp, like my hue dimmer switches already do?

sunshineautomate, 2021-04-24:
Guys, This bug is still an issue. Is there any change I it can get some attention please.

GeoCyberwolf, 2021-04-24:
> 
> 
> Guys, This bug is still an issue. Is there any chance I it can get some attention please.

You can do it in the old deconz webui, just not the phoscon webapp (Oh my god this is such a fragmented piece of software!!). Try going to http://[ip address]/login.html. Should give you the old WebUI where you can link the switch.

Though I would suggest opening another issue to get support properly added to the phoscon webapp.