dresden-elektronik/deconz-rest-plugin #3493: Device name

mcalderin, 2020-10-23:
## Device

- Product name: Commercial Electric Smart LED Downlight T62
- Manufacturer: ETI
- Model identifier: Zigbee CCT Downlight
- Device type:
  - Light

## Screenshots

[image]
[image]
[image]
[image]
[image]

mcalderin, 2020-10-23:
Note: this is not a color dimmable light.  It's a color temperature light.

ebaauw, 2020-10-24:
No it’s not; the HA profile doesn’t define a _Color Temperature Light_, so its type is _Color Dimmable Light_.  Notwithstanding that the API should expose `ct` and not `xy`, `hue`, and `sat`.  The light might need to be whitelisted for that.  How does the API expose it crrently?

mcalderin, 2020-10-24:
Thanks.  Let me know if I can provide any other info.

ebaauw, 2020-10-24:
What does the _Color Capabilities_ attribute in the _Color Control_ cluster report?  Make sure to read the attribute.

mcalderin, 2020-10-24:
[image]

closed 2020-10-29 by manup

TheAlphaLaw, 2020-12-26:
I would also like these lights to be addend correctly. Previously these added fine and were routers. Now they report as Color changing lights, with no temperature control and are not able to mesh with other devices. Even when they're 5 feet from the Conbee II, they won't mesh.