dresden-elektronik/deconz-rest-plugin #3495: add ikea Tradfri shortcut button support

joten75, 2020-10-24:
## Feature request type

add support for ikea shortcut button
## Description

## Considered alternatives

## Additional context

[image]
[image]
[image]
[image]

ebaauw, 2020-10-24:
Missed that one.  Google reveals a number of news articles from around June, mentioning that the shortcut button was spotted in the Netherlands online store, but I cannot find it.  Where (in which country) did you buy it?  I've been trying to get a standalone open/close remote (for my Xiaomi curtains controller); IKEA customer service mentioned it's listed in their database, but not yet available to customers.  They seem to have some challenges in getting these devices ready for mass distribution.

I'd prefer to get my hands on this device myself, but, in the meanwhile, can you provide the following info:

1. The new items suggest it only supports a single action, to recall a scene.  It is physically like the on/off switch and open/close remote, with two pushable areas, or indeed just a single pushable area?
2. Could you please read each attribute in the _Basic_ cluster (using the _Attribute Editor_ popup window if needed) and make another set of screenshots.  I'm particularly interested in _ZCL_Version_, _Date Code_, _Product Code_, and_SW Build ID_.
3. The device fingerprint is a spitting image of the on/off switch, except that the server 0xFC7C cluster is missing.  Unfortunately, it gives no clue how the button works.  It's missing a client _Scenes_ cluster as well as a server cluster for attribute reporting.  My guess is it would use _On/Off_ and/or _Level Control_ commands and direct bindings to the Trådfri Hub.  The Hub then generates the commands for enacting the (non-Zigbee !) scene, similar to how the SYMFONISK controller works: the Hub translates _Move to Level_ to Sonos API calls.  The concept would be very similar to deCONZ and Hue bridge rules.  Do you have a Trådfri Hub?  If so, how does the IKEA Home Smart app configure the "scene"?  Could you post any screenshots?
4. Could you run deCONZ with `--dbg-info=2` and see what "No button event" messages appear, including the `asdu` of the preceding APS-DE.indication message?  Make sure to try single click (press and release quickly), press & hold, and release after holding.

joten75, 2020-10-24:
it's a single click button. like the xiaomi round button. I don't have ikea hub sorry.
[image]
[image]

Mimiix, 2020-10-26:
Swapped labels.

MattWestb, 2020-10-26:
Was looking on ikea.it but cod not finding it online :-(
Not one big problem then can't going over border without 14 days "forced extra hallyday".

The PCB is one slimmed (one button less) of the On/Off dimmer / Open / close switch from [FCC-ID](https://fccid.io/FHO-E1812/Internal-Photos/E1812-Shortcut-button-Internal-photo-AA-2189372-1-4489068).

The PCB layout is nice like the the big brothers and can easy being modded to one Zigbee sniffer or one EZSP Coordinator in 5 minutes :-)))))

joshua-schnabel, 2020-11-02:
I got the button (or even three) at IKEA Germany (Karlsruhe). The button is not listed by Ikea. It is also not listed on the OTA page (http://fw.ota.homesmart.ikea.net/feed/version_info.json). @ebaauw below you will find the information you are looking for. This is a different tool I used but I stumbled over the issue during my research and thought I share my information. 
```
Zigbee2MQTT:debug 2020-11-02 21:46:56: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"modelId":"TRADFRI SHORTCUT Button"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:46:56: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"modelId":"TRADFRI SHORTCUT Button"}'
Zigbee2MQTT:debug 2020-11-02 21:46:58: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"IKEA of Sweden"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:46:58: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"IKEA of Sweden"}'
Zigbee2MQTT:debug 2020-11-02 21:47:00: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"powerSource":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:47:00: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":3}'
Zigbee2MQTT:debug 2020-11-02 21:47:02: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:47:02: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
Zigbee2MQTT:debug 2020-11-02 21:47:04: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"appVersion":33}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:47:04: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":33}'
Zigbee2MQTT:debug 2020-11-02 21:47:06: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"stackVersion":98}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:47:06: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":98}'
Zigbee2MQTT:debug 2020-11-02 21:47:08: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:47:08: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
Zigbee2MQTT:debug 2020-11-02 21:47:10: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"dateCode":"20190715"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:47:10: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":"20190715"}'
Zigbee2MQTT:debug 2020-11-02 21:47:12: Received Zigbee message from '0x680ae2fffe34c590', type 'readResponse', cluster 'genBasic', data '{"swBuildId":"2.3.015"}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-11-02 21:47:12: No converter available for 'E1812' with cluster 'genBasic' and type 'readResponse' and data '{"swBuildId":"2.3.015"}'
Zigbee2MQTT:debug 2020-11-02 21:50:30: Received Zigbee message from '0x680ae2fffe34c590', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":587290161,"imageType":4550,"manufacturerCode":4476}' from endpoint 1 with groupID 0
```

```
type 'commandOn', cluster 'genOnOff'
type 'commandMoveWithOnOff', cluster 'genLevelCtrl'
type 'commandStopWithOnOff', cluster 'genLevelCtrl'
```

ebaauw, 2020-11-02:
Thanks, @joshua-schnabel.  Could you also get the commands the device sends when the button is pressed?

@joten75, could you please read the attributes before making a screenshot.  Double-click on each attribute for the _Attribute Editor_ popup window.

edvaldocastro, 2020-11-14:
Hi @ebaauw , I'm facing the same issue, so instead of opening another issue, I'm trying to contribute on this one. Please check if the below is the information you are looking for. (It looks like I cannot read the attributes, but I hope it helps).

"I'm particularly interested in ZCL_Version, Date Code, Product Code, and SW Build ID."

[image]

**ZCL_Version**
[image]

**Date Code**
[image]

**Product Code**
[image]

**SW Build ID**
[image]

Unfortunately it looks like most of reading from the attributes have failed, but some of them worked, please see the full picture below.
[image]

I hope this helps.

Edvaldo Castro

SwoopX, 2020-11-14:
Unfortunately, it doesn't. However, try pushing the button and then immediately read the attributes. The device must be awake the get the read requests.

edvaldocastro, 2020-11-14:
Unfortunately I couldn't make it work.. all the readings are failing, I'll keep trying... Let's see if others have more success on it.

Mimiix, 2020-11-16:
@edvaldocastro Keep clicking the button when its reading (Both the read button and the button itself).

t0bse, 2020-12-08:
any news on this? got two shortcut buttons but can't read them in deconz.

letszb, 2020-12-09:
The button is online at ikea DK, Available at 3 of 5 stores and for shipping.
https://www.ikea.com/dk/da/p/tradfri-genvejskontakt-hvid-40356381/

mtjoernelund, 2020-12-09:
Just bought a couple of these myself. I run Deconz headless, but will try to fire it up in x11 remote and see if I can provide information. First time trying to help implement new devices, so I am a little unsure if it is all information that is still missing or if it is something specific?

Lemocin, 2020-12-14:
I manages to read some of the values by deleting and repairing. If you need some other values, please let me know, i will try to read them out!

[image]
[image]

ebaauw, 2020-12-22:
Finally got my hands on one, thanks to Kevin.  It's one of those devices that goes to sleep pretty quickly after pairing, and won't wake up, unless you created a binding within that time.
[image]

On pressing the button, it sends an _On_ command to the bound group.  The client _Level Control_ and _Window Covering_ cluster seem to be there just for show.  _Battery Percentage Remaining_ is reported in full percentage, as with other IKEA devices.  It supports attribute reporting.

ebaauw, 2020-12-22:
> It's really half an on/off switch; we can probably re-use the buttonmap.

This must have been the easiest device I've ever integrated.

```
$ ph get /sensors/10
{
  "config": {
    "alert": "none",
    "battery": 87,
    "group": "11",
    "on": true,
    "reachable": true
  },
  "ep": 1,
  "etag": "729f204a8ff1827197ab1abb52bc59b0",
  "lastseen": "2020-12-22T21:53Z",
  "manufacturername": "IKEA of Sweden",
  "mode": 1,
  "modelid": "TRADFRI SHORTCUT Button",
  "name": "TRADFRI SHORTCUT Button ",
  "state": {
    "buttonevent": 1002,
    "lastupdated": "2020-12-22T21:53:06.523"
  },
  "swversion": "2.3.015",
  "type": "ZHASwitch",
  "uniqueid": "68:0a:e2:ff:fe:34:91:ec-01-1000"
}
```
And the group:
```
$ ph get /groups/11
{
  "action": {
    "alert": "none",
    "bri": 127,
    "colormode": "hs",
    "ct": 0,
    "effect": "none",
    "hue": 0,
    "on": true,
    "sat": 127,
    "scene": null,
    "xy": [
      0,
      0
    ]
  },
  "devicemembership": [
    "10"
  ],
  "etag": "0034c83da5355a6a42c2b6a996456718",
  "id": "11",
  "lights": [],
  "name": "TRADFRI SHORTCUT Button ",
  "scenes": [],
  "state": {
    "all_on": false,
    "any_on": false
  },
  "type": "LightGroup",
  "uniqueid": "68:0a:e2:ff:fe:34:91:ec"
}
```
Not sure why the endpoint is missing from the group's (probably non-unique) `uniqueid`.

closed 2020-12-23 by manup

martinimarcello00, 2021-02-11:
Hi, I just installed the update of the latest version (2.09.03) of deCONZ. Theoretically this version should allow the pairingof the Tradfri Shortcut Button by the phoscon webapp (in my case installed on home assistant). I repeatedly tried to connect the new device but every time the connection time runs out and the error "Failed to connect" appears.What can I do in order to pair the device? Thanks for the support.

mtjoernelund, 2021-02-12:
I’m on version 2.09.00. I can’t get the shortcut button to show in the Phoscon webapp, however they pair just fine using the old webapp (open wireless light control). Only issue has been that I struggled to rename them in the old webapp. Easily solved using the ph command that comes with ebaauw’s Homebridge-hue plugin.

al-wi, 2021-02-13:
Can confirm that pairing is not possible in the webapp with a new shortcut button in version 2.09.03. However you'll find the node in the deconz app:
[image]

snookyly, 2021-02-13:
Same here. Already paired shortcut button is not visible in Phoscon webapp.

andreassave, 2021-02-14:
I also paired a new shortcut button using phoscon webapp which seemed to fail according to the ui. However the button seemed to have paired anyway.. It shows up in deconz app (2.9.3).

I have a problem though. I use Openhab which listens to buttonevents and every single press seems generate two 1002 events. I use other Trådfri remotes the same way without any problem.

Log items in deconz when shorcut button is pressed once:

`12:09:45:746 [INFO] - Button 1002 - TRADFRI SHORTCUT Button, broadcast to: 0x0013, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 93'
'12:09:45:808 [INFO] - Button 1002 - TRADFRI SHORTCUT Button, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 93`

This can be compared to single press on Trådfri remote (5 button):

`12:10:01:616 [INFO] - Button 1002 - TRADFRI remote control, broadcast to: 0x0011, endpoint: 0x01, cluster: ONOFF (0x0006), action: Toggle, payload: None, zclSeq: 65`

Unfortunately my deconz skills are not in a level where I can draw any conclusions on this, other than the fact that I get two rows in the log for the shortcut button.

Does anyone have a clue why I get two events?

hugosbnarciso, 2021-02-14:
@andreassave i'm using home assistant, what i do is filter by the unique id, not sure if you can do that from OpenHab.

hugosbnarciso, 2021-02-14:
my node-red config:

[image]

hope it helps,
H

andreassave, 2021-02-14:
Thanks. 

I gather Openhab is different in this regard. As I understand my rule in Openhab gets triggered by the buttonevent and the type of event is all that is "known", i e in this case 1002. For now it gets fired two times. Which in my case turns on and off my lights at once :-)

I tried debug level 2 and that reveal two websocket updates which probably is some kind of evidence that this is "caused" by deconz.

```
14:27:40:169 [INFO] - Button 1002 - TRADFRI SHORTCUT Button, broadcast to: 0x0013, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 102
14:27:40:172 Websocket 192.168.1.xx:57808 send message: {"e":"changed","id":"36","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-02-14T13:27:40.169"},"t":"event","uniqueid":"58:8e:81:ff:fe:29:b8:ad-01-1000"} (ret = 165)
14:27:40:245 [INFO] - Button 1002 - TRADFRI SHORTCUT Button, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), action: On, payload: None, zclSeq: 102
14:27:40:248 Websocket 192.168.1.xx:57808 send message: {"e":"changed","id":"36","r":"sensors","state":{"buttonevent":1002,"lastupdated":"2021-02-14T13:27:40.246"},"t":"event","uniqueid":"58:8e:81:ff:fe:29:b8:ad-01-1000"} (ret = 165)

```

I wonder what "unicast to" means? That line does not exist when watching my other Trådfri remotes. Other than the fact that this is the first time I use the Shortcut button it is also the first time the pairing was not performed ok in phoson app. Can that have caused this? Do I have to do more configuration in the deconz app?

hugosbnarciso, 2021-02-15:
i'm sorry i'm not versed in OpenHab :(
hopefully someone here can help you.
H

Mimiix, 2021-02-15:
@j-n-k Can you help out?

J-N-K, 2021-02-15:
The problem seems to be that the same message is sent twice by deconz (according to you log one time as "broadcast" and one time as "unicast" message). From the websocket message itself it is impossible to tell which one is which one (so we can't ignore one of them). Since there is nothing like an "event-id", it's also impossible to tell if these are really two clicks on the button or if the second one s a duplicate. IMO the deconz-plugin should not send the same event twice.

If you trigger a rule in openHAB, you could "debounce" by storing the last time the rules was triggered and check if it is (e.g.) at least 1s in the past and exit early otherwise.

ebaauw, 2021-02-15:
Afaik the button should only send broadcasts to the group when configured by the API.  Did you manually create a binding to the coordinator?

trevorbenyack, 2021-02-18:
Hi, I'm also seeing that a short press is sending two 1002 events (in the deCONZ logs and to HA). Also not showing up in Phoscon.

trevorbenyack, 2021-02-18:
> Hi, I'm was also seeing that a short press was sending two 1002 events (in the deCONZ logs and to HA). -- Something happened where it got disconnected, so I did the reset on the device (4 short presses of the pairing button) and re-scanned in phoscon and now it is sending only 1002 for a short press. So maybe try repairing it?

Okay, now it's back to sending double 1002 events... really strange.

andreassave, 2021-02-18:
> Afaik the button should only send broadcasts to the group when configured by the API. Did you manually create a binding to the coordinator?

I have not done any manual configuration at all. My pairing was done in Phoson. The pairing failed according to the ui but the shortcut button is listed in deconz app ui and in api (.../sensors) so it seemed to have paired anyway. Besides the pairing no additional configuration have been made.

etcg, 2021-02-28:
I have the same problem with the double events. My pairing was also done only in Phoscon. I tested it with ZHA integration in Home Assistant and there it were even 5-6 events on 1 button press. My logs look exactly like that.

> Thanks.
>
> I gather Openhab is different in this regard. As I understand my rule in Openhab gets triggered by the buttonevent and the type of event is all that is "known", i e in this case 1002. For [...]

Knuspel, 2021-03-01:
I'm getting this error with version 2.10.01. 
Anything I can add or change to mitigate that?

```
21:43:57:350 [INFO] - No button handler for: TRADFRI SHORTCUT Button, unknown, endpoint: 0x01, cluster: 0x0003, command: 0x01, payload: None, zclSeq: 2
```

Knuspel, 2021-03-01:
I thought I did. 
I guess I forgot to mention that this message in the post comes up while pairing.

```
21:43:57:350 [INFO] - No button handler for: TRADFRI SHORTCUT Button, unknown, endpoint: 0x01, cluster: 0x0003, command: 0x01, payload: None, zclSeq: 2
```

SwoopX, 2021-03-01:
There is no error, but an information. So nothing to do in this regard.

Knuspel, 2021-03-01:
Fair. But that's the only info about the button I can find in the logs. 
What am I doing wrong?
I seem to not even get the messages the others reported.

SwoopX, 2021-03-02:
It might help if you enter search for arbitraty switches and then press the button(s). I'd place my bet that the cluster bindings haven't been set up which would explain the missing info messages.

Knuspel, 2021-03-02:
I did that, there is now a bit more in the logs. 
The button still shows up in deCONZ but doesn't show battery or switches anything.
It also doesn't show up in Phoscon.
```
09:39:30:094 DeviceAnnce of SensorNode: 0xEC1BBDFFFE8FF1EC [1],
09:39:30:092 0x66BA nwk changed to 0xE33B,
09:39:36:731 [INFO] - No button handler for: TRADFRI SHORTCUT Button, unknown, endpoint: 0x01, cluster: 0x0003, command: 0x01, payload: None, zclSeq: 18,
09:39:36:731                       ...     (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:36:730 FP indication 0x0104 / 0x0003 (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:34:530 discard double entry in binding queue (size: 1) for for 0xEC1BBDFFFE8FF1EC, cluster 0x0006,
09:39:34:524 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) cluster 0x8004,
09:39:34:524                       ...     (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:34:524 FP indication 0x0000 / 0x8004 (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:34:519 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) clear timeout on cluster 0x8004,
09:39:34:519 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) cluster 0x8004,
09:39:34:518                       ...     (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:34:518 FP indication 0x0000 / 0x8004 (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:33:329 wait response fastEnddeviceProbe() 0xEC1BBDFFFE8FF1EC, elapsed 78 ms,
09:39:33:324 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) cluster 0x8005,
09:39:33:324                       ...     (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:33:323 FP indication 0x0000 / 0x8005 (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:33:292 wait response fastEnddeviceProbe() 0xEC1BBDFFFE8FF1EC, elapsed 40 ms,
09:39:33:287 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) cluster 0x8005,
09:39:33:286                       ...     (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:33:286 FP indication 0x0000 / 0x8005 (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:33:251 [3] get simple descriptor 0x01 for 0xec1bbdfffe8ff1ec,
09:39:33:245 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) clear timeout on cluster 0x8005,
09:39:33:245 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) cluster 0x8005,
09:39:33:244                       ...     (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:33:244 FP indication 0x0000 / 0x8005 (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:32:797 Device TTL 1247 s flags: 0x7,
09:39:32:793 CTRL got nwk update id 1,
09:39:32:789 Current channel 11,
09:39:32:232 [2] get active endpoints for 0xec1bbdfffe8ff1ec,
09:39:32:227 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) clear timeout on cluster 0x8002,
09:39:32:227 ZDP indication search sensors 0xEC1BBDFFFE8FF1EC (0xE33B) cluster 0x8002,
09:39:32:227                       ...     (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:32:226 FP indication 0x0000 / 0x8002 (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:31:818 ZCL attribute report 0xBC33ACFFFEAF479F for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 ,
09:39:30:974 [1] get node descriptor for 0xec1bbdfffe8ff1ec,
09:39:30:097 discard double entry in binding queue (size: 1) for for 0xEC1BBDFFFE8FF1EC, cluster 0x0006,
09:39:30:095 device announce 0xEC1BBDFFFE8FF1EC (0xE33B) mac capabilities 0x80,
09:39:30:095                       ...     (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:30:095 FP indication 0x0000 / 0x0013 (0xEC1BBDFFFE8FF1EC / 0xE33B),
09:39:30:094 set fast probe address to 0xEC1BBDFFFE8FF1EC (0xE33B),
09:39:30:094 device announce 0xEC1BBDFFFE8FF1EC (0xE33B) mac capabilities 0x80,
09:39:30:094 	nwk address changed 0x0000 -> 0xE33B [2]
```

Knuspel, 2021-03-02:
Funnily enough I used the Phoscon beta Webversion and paired the button and that worked.

Elvys8, 2021-03-05:
Hi everyone. I need help pairing the shortcut button. I have a Synology NAS running HombeBridge and marthoc-dezonz Docker containers. I’ve tried putting the button in pairing mode pushing the link button four times and a red light blinks, I select “other” on the scan assistant but nothing happens. Any suggestions? Thanks in advance!
[image]

Mimiix, 2021-03-05:
@Elvys8  You need to update as you are on 2.05.79.

Elvys8, 2021-03-05:
@Mimiix Thanks for the quick reply. That "The version is up to date" message totally fooled me…
I've been looking on Google for a way to update the Phoscon software on my docker container but couldn't find a tutorial.
I'm assuming it's quite an obvious process but my basic skills don't go that far I'm afraid.
Any idea on how to update the software?
Thanks again!

Mimiix, 2021-03-05:
> @Mimiix Thanks for the quick reply. That "The version is up to date" message totally fooled me…
> I've been looking on Google for a way to update the Phoscon software on my docker container but [...]

I have no clue either. Check with the docker repo.

Knuspel, 2021-03-05:
> @Mimiix Thanks for the quick reply. That "The version is up to date" message totally fooled me…
>
> I've been looking on Google for a way to update the Phoscon software on my docker container but [...]

It is properly explained in the docker repo. Basically you need to stop the container and then run the update script once. 

As to pairing the shortcut button. As said before I only was successful with the Phoscon beta accessible through their webservice. 

And even then it did not work as a light switch for certain things. I needed it to activate scenes that then turned the light on. Just as an FYI.

Smanar, 2021-03-05:
No one with the double event bug can try this PR ? https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4265

Elvys8, 2021-03-08:
I managed to update the docket container. I’m on version 2.10.02 now.
Still no luck, as mentioned by others paring is not working with the Phoscon web app: "Failed to connect"

ebaauw, 2021-03-08:
> It does somehow shows up on my Home app via homebridge-hue, but not as a button

Please open an issue in the Homebridge Hue repo, attaching the debug dump file and capturing and attaching a debug log file.  See the Homebridge Hue FAQ on how to do this.

jonas240, 2021-03-25:
I paired the shortcut switch like @Knuspel (phoscon beta) which worked as expected. However the switch does not report the correct battery percentages. It's still stating 0% in Homekit (through Homebridge). 
@ebaauw Is this a bug or did I miss something? 
Thanks!

ebaauw, 2021-03-25:
> It's still stating 0% in Homekit (through Homebridge).

How long is "still"?  It could probably take a couple of hours after pairing before the switch send the first battery report to deCONZ.  Best check in the deCONZ GUI if _Battery Percentage Remaining_ in the _Power Configuration_ cluster is still 0.  If it remains 0 after several hours, the binding and attribute reporting haven't been setup properly.  Either do this manually (see the _User Manual_ in the GUI) or reset and re-pair the button (without deleting it from deCONZ first).

Elvys8, 2021-03-25:
> It's still stating 0% in Homekit (through Homebridge).

It took a couple of days for me to show up properly. Same with Aqara buttons and sensors.

jonas240, 2021-03-25:
Thanks for the fast reply @ebaauw and @Elvys8 

It's been about two days since pairing. I'll wait a few days and that try the suggested approaches.

davewatson91, 2021-04-13:
This still doesn't appear to be working for me - I can't add it through the Phoscon interface. In deCONZ, it shows up as connected, but in HA it only provides a battery status and nothing else..

Lars-B84, 2021-04-13:
Have you tried using the shortcut button Blueprint thats available?

davewatson91, 2021-04-13:
> […](#)

Thankyou - that does work! - Why can't the shortcut button be treated like any other button in HA?

psaggiomo, 2021-04-25:
Hi. Any news about integration?

Mimiix, 2021-04-25:
@psaggiomo not sure why you are asking as it is integrated.

If you don't see it in phoscon, check their repository.

sieren, 2021-05-17:
Seems to work overall pretty "ok" on my end with the most recent deconz beta.
However it's now showing the battery level at 16% (after only 7 days of use with the stock battery). Is anyone else experiencing this?

jonas240, 2021-05-17:
@sieren I experienced the same. Just ignore it. A couple of uses later it will change to about 80%. I guess IKEA guesses the battery over provided voltages and that fluctuates sometimes.

sieren, 2021-05-17:
Thanks for the info! :) Good to know!

mrclschstr, 2021-06-26:
Since phoscon beta 2.12.01 it finally works. Just give it a shot!

[6 more replies]