dresden-elektronik/deconz-rest-plugin #3500: Segmentation fault during upgrade from RaspBee I to RaspBee II

fubar-coder, 2020-10-24:
## Describe the bug

I get a `Segmentation fault (core dumped)` when I try to upgrade from RaspBee I to RaspBee II following the instructions in issue #1441.

## Steps to reproduce the behavior

1. Stop the Raspberry PI 4B 8GB
2. Open the case
3. Replace the RaspBee I with a RaspBee II and start the network recovery procedure as described in the release notes for deconz V2.05.59.
4. Close the case
5. Start the Raspberry PI

## Expected behavior

No segmentation fault/error.

## Screenshots

## Environment

 - Host system: Raspberry Pi 4B 8GB
 - OS: Raspbian arm64
 - Running method: Marthoc Docker container
 - Firmware version: 26660700
 - deCONZ version: 2.05.84 / 14.9.2020
 - Device: RaspBee II

## deCONZ Logs

[deconz-crash-log.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5433664/deconz-crash-log.txt)

## Additional context

SwoopX, 2020-10-24:
I feel this is closely related to #3133

fubar-coder, 2020-11-24:
I tried all three approaches: Restore a backup, load a recent config and editing the database. The error was always the same.

SwoopX, 2020-11-24:
Well, as you haven't provided any further information as I've requested, don't wonder. I would have had a potential solution available taking only 5 mins... 🤷‍♂️

fubar-coder, 2020-11-24:
@SwoopX Ok, and what would've been that solution? Would be interesting for all other persons having the same problem.

Mimiix, 2020-11-24:
> There is a log and I gave all the information as required by the bug report template.

@SwoopX asked for :

Can you describe in more detail what happened directly after device switch? Apparently, deconz ran for a while. Have you tried to restore a Phoscon backup or just to load a recent config via the hidden page?

fubar-coder, 2020-11-24:
@Mimiix What else can I tell you other than telling you that it gave a segmentation fault? And the other information is just pointless. A segmentation fault is always an application bug and the log should be enough. Except when the log is either incomplete or useless.

Mimiix, 2020-11-24:
@fubar-coder For what it's worth: I did read, but i have no knowledge at all. I am just standing with Swoop. He's a dev and he knows what's up.
 
Without you, we can't fix your issue. I'm just trying to see what i can do to help here with either asking the right people or forwarding it to DE. So let us help you and see where we can bring you. Don't worry, the toolbox isn't empty but we have to start somewere.

fubar-coder, 2020-11-24:
@Mimiix Yeah, it was frustrating to see that @SwoopX didn't utilize the provided information and telling me later that he has a potential solution **without even telling what it is**! XKCD refence: [#979](https://xkcd.com/979/)

SwoopX, 2020-11-26:
@fubar-coder Listen, I very well read what you wrote, but you were obviously totally indifferent to my reply. I even gave you two simple questions back on this, but yes, it's my fault having not taken your answers into account. I deeply apologize that my clairvoyant abilities are still underdeveloped. And wait, have I indicated a possible solution here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3500#issuecomment-716045220? Again, my bad that I overlooked you database entries.

Sorry mate, but you're not gonna put this on me. There wasn't any update on this for 3 weeks, but it's always delightful to see how people react when you tickle them. I offered help which you didn't want.

fubar-coder, 2020-11-27:
@SwoopX No, you didn't really help. You asked questions that could've been answered by reading the issue, but you were too lazy. However, I'm closing this issue, as there is no will to resolve this issue.

closed 2020-11-27 by fubar-coder

Mimiix, 2020-11-27:
@fubar-coder I'd like you to read the following post. https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3765

After that, I hope you just have bad day and tommorow will be better for you.

 Talking to people like you just did to Swoop is not done.

fubar-coder, 2020-11-27:
@Mimiix I could've understood when I should've helped with debugging the segmentation fault, and I really would've helped in that case. It would also be nice to see, if a developer would've tried to reproduce the issue. It would even be OK, if the developer would've said that he/she cannot reproduce the issue, and I'm willing to help in that case. That's what I do as developer. That's what I expect from other developers. It might be a very high bar to meet, but I set this bar for me too. I could've answered the questions from SwoopX and nothing would've changed. This is what frustrates me to no end.

Mimiix, 2020-11-27:
> @Mimiix I could've understood when I should've helped with debugging the segmentation fault, and I really would've helped in that case. It would also be nice to see, if a developer would've tried to [...]

I hear you. But please, stay civil. That's all I ask from you.

fubar-coder, 2020-11-27:
Will do

SwoopX, 2020-11-27:
@fubar-coder You know, if you get such situations as here like 5 times a day when you ask some questions and get no feedback, that is also incredibly frustrating. If you then combine that with 2 fubar weeks of work and regardless some of your anyway almost non-existant leasure time and then the perception remains that this is just ignored, you just end up with sarcasm and a pretty reluctant attitude.

So, there's a reason behind my questions and it's the details that count. I also gave reference to a very similar (if not the identical) issue in this regard. I personally don't see too much benefit in just throwing solutions at people who later on say "Yeah, probably a solution, but for a different problem". Folks in my world do not survive long with just throwing stuff at people. Please consider this explanation.

> It would even be OK, if the developer would've said that he/she cannot reproduce the issue, and I'm willing to help in that case.

Oh, I can still reproduce my crashes any time like a clockwork. And as you can derive from the issue raised, we don't know why it occurs, especially since it appears to be originating from a QT function based on the coredump.

> That's what I expect from other developers. It might be a very high bar to meet...

I started with deconz from a coding perspective because somebody was so kind to provide some base support for a smoke sensor I wanted to use. However, since I wanted to have the "full package", I dived into the code and while feeling more and more comfortable, started others to help and integrate lots of new devices. So, I'd hope that gets me close to your bar.

Now, we'd be more than happy to get ideas or support in pinpointing the root cause. However, all I can potentially offer is a workaround, which made my test network usable again. There was not yet put much effort since from a cost/benefit perspective, it just affected myself (and maybe a handful of others).

"_Please, first help us a bit to help you!_ "

fubar-coder, 2020-11-27:
> @fubar-coder You know, if you get such situations as here like 5 times a day when you ask some questions and get no feedback, that is also incredibly frustrating. If you then combine that with 2 [...]

I can understand this.

> So, there's a reason behind my questions and it's the details that count. I also gave reference to a very similar (if not the identical) issue in this regard. I personally don't see too much benefit [...]

Yes, I did consider it. My first guess - according to the log file - was, that a network change also changes several pointers/objects in the application and that a different thread tries to access some just recently released objects. In other words: This will most likely be a use-after-free or a null pointer problem. That's the reason why doing the same thing (even though in three different ways), didn't seem like a helpful idea to me.

> Oh, I can still reproduce my crashes any time like a clockwork. And as you can derive from the issue raised, we don't know why it occurs, especially since it appears to be originating from a QT [...]

Something I'd like to validate is: Does this happen on every platform or is it a Linux ARMv8 (64 bit) problem only? Is there a way to get a debug build with a PDB file on Windows?

> "_Please, first help us a bit to help you!_ "

Will do.

SwoopX, 2020-11-27:
> Yes, I did consider it. My first guess - according to the log file - was, that a network change also changes several pointers/objects in the application and that a different thread tries to access [...]

Yeah, we fished quite some stuff out over the last months. It's definitively related to loading/applying the network settings and reminds me of a stupid mistake I made long time ago while using the index for a vector instead of the interator.
Mentioning that because adding an additional line in the table where the network settings were stored got it working again after some fiddling around. Only noticed that because I had a 2nd system which was still operational and compared the network settings.

> Something I'd like to validate is: Does this happen on every platform or is it a Linux ARMv8 (64 bit) problem only? Is there a way to get a debug build with a PDB file on Windows?

So far, I can confirm it happens under Buster and Ubuntu x64.

SwoopX, 2021-02-20:
Root cause in #3133