dresden-elektronik/deconz-rest-plugin #3501: Can't connect bulbs

cryptolukas, 2020-10-24:
## Describe the question or issue you are having

I started the docker image get always these message in my logfile
```
22:49:39:661 device disconnected reason: 4, index: 0
22:49:40:521 wait reconnect 15 seconds
22:49:40:624 dev /dev/ttyAMA0
22:49:41:520 wait reconnect 14 seconds
22:49:42:521 wait reconnect 13 seconds
22:49:43:520 failed to reconnect to network try=3
22:49:43:521 wait reconnect 12 seconds
22:49:43:633 dev /dev/ttyAMA0
22:49:44:520 wait reconnect 11 seconds
22:49:45:520 wait reconnect 10 seconds
22:49:46:520 wait reconnect 9 seconds
22:49:47:521 wait reconnect 8 seconds
22:49:48:520 failed to reconnect to network try=4
22:49:48:520 wait reconnect 7 seconds
22:49:49:520 wait reconnect 6 seconds
22:49:50:520 wait reconnect 5 seconds
22:49:51:521 wait reconnect 4 seconds
22:49:52:521 wait reconnect 3 seconds
22:49:53:520 failed to reconnect to network try=5
22:49:53:521 wait reconnect 2 seconds
22:49:53:636 dev /dev/ttyAMA0
22:49:54:520 wait reconnect 1 seconds
22:49:54:621 dev /dev/ttyAMA0
22:49:54:667 device disconnected reason: 4, index: 0
```

docker-compose.yml
```
version: '3'
services:
  deconz:
    image: marthoc/deconz
    # This makes for more convienient naming, but only allows one instance of the image
    container_name: deconz
    restart: unless-stopped
    network_mode: "host"
    devices:
      - "/dev/ttyAMA0:/dev/ttyAMA0"

```

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS) Raspberry 4
 - Running method: (Raspbian PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
 / Marthoc Docker container)
 - Firmware version: (26xxyy00)
```
22:43:43:969 Announced to internet https://phoscon.de/discover
22:43:43:969 discovery server date: Sat, 24 Oct 2020 22:43:43 GMT
22:43:43:969     local time seems to be ok
22:43:43:969 discovery found version 2.04.35 for update channel stable
```
 - deCONZ version: (2.xx.yy) docker-compose use latest
 - Device: (CRaspBee II)
 - Do you use an USB extension cable: ( no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?
NOPE

## deCONZ Logs

## Additional context

Mimiix, 2020-10-26:
Missing some info here. 

But, i think you need the full serial name under "devices" in the compose file

cryptolukas, 2020-10-26:
What information is missing?
In the documention is the device mentioned as /dev/ttyAMA0

Did I understand something wrong?

Mimiix, 2020-10-26:
I just seen that aswell. 

Can you try to find the device serial? Then put that there? IT should look like this:
/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_XXXXXXXX-if00

cryptolukas, 2020-10-26:
I have just these serial devices
```
root@raspberrypi:~# ls -lah /dev/serial0
lrwxrwxrwx 1 root root 5 Oct 24 22:55 /dev/serial0 -> ttyS0
root@raspberrypi:~# ls -lah /dev/serial1
lrwxrwxrwx 1 root root 7 Oct 24 22:55 /dev/serial1 -> ttyAMA0
root@raspberrypi:~# cat /dev/serial1
root@raspberrypi:~#
```

Mimiix, 2020-10-26:
So it doesn't see the stick at all. Is it attached? Do you use extension cable? Raspbee? Conbee?  Is it new? Old? Firmware updated?

cryptolukas, 2020-10-26:
I don't use an extension cable as mention initially. I use the GPIO module Raspbee II
Firmware as mentioned initially too.

cryptolukas, 2020-10-26:
I hope that's correct

Mimiix, 2020-10-26:
But did you buy it brand new? It doesn't see it, so i am not sure if it works. Can you install clean raspbian and follow this guide:
https://phoscon.de/en/raspbee2/install#raspbian

cryptolukas, 2020-10-26:
I do it later.

cryptolukas, 2020-11-05:
I could fix it.
The docker container seems to have issues [issue-3526](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3526)

Installed deconz locally. I offer my help to make the docker image better. Seems broken at the moment.

closed 2020-12-19 by stale[bot]