dresden-elektronik/deconz-rest-plugin #3503: Aurora AONE Smart PIR sensor

aberkeley, 2020-10-25:
## Device

Hi would it be possible to add support for the follow device: 

- Product name: Auora AONE™ PIR SENSOR
- Manufacturer: Aurora
- Model identifier: MotionSensor51AU
- Device type :   - Sensor
  
https://aone.auroralighting.com/ProductDetails/AU-A1ZBPIRS

## Screenshots
[image]

[image]

[image]

SwoopX, 2020-10-25:
That's a rebranded Develco motion sensor. Integration should be fairly easy.

ebaauw, 2020-10-25:
Three _Occupancy Sensor_ and an _IAS Zone_ endpoint.  How many motion sensors does it have?

SwoopX, 2020-10-25:
That's actually correct. You can potentially realize complex scenarios on illuminance and such according to the manual. However, that "power" is supposedly only unleashed with their gateway...

aberkeley, 2020-10-25:
It just has a single motion sensor, i agree with SwoopX - be good to just get the IAS working to start with. The luminance/occupancy stuff would be fun to use, but basic motion sensing is a start!

closed 2020-11-14 by manup