dresden-elektronik/deconz-rest-plugin #3505: Again and again ghost devices.

salopette, 2020-10-25:
Hello, I keep getting ghost devices.
I've already restarted Pi, but it doesn't work.
And when I let Alexa search for devices, it also finds which I haven't had for half a year.
Can you empty the cashe?
[image]
[image]

closed 2020-10-25 by close-issue-app[bot]

salopette, 2020-10-25:
Ich habe bereits alle Infos zur Verfügung gestellt, warum macht dieser scheiß BOT alles zu?

salopette, 2020-10-25:
Wenn ich die Geräte mit DEL lösche, kommen die nach einer Zeit wieder

[image]

Mimiix, 2020-10-25:
You need to fill in the form and keep the headers. Please provide all information. And keep it English.