dresden-elektronik/deconz-rest-plugin #3506: Again and again ghost devices.

salopette, 2020-10-25:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

[image]
[image]

salopette, 2020-10-25:
Hello, I keep getting ghost devices.
I've already restarted Pi, but it doesn't work.
And when I let Alexa search for devices, it also finds which I haven't had for half a year.
Can you empty the cashe?

Mimiix, 2020-10-25:
Really? First ignoring the template, now tricking the bot?

closed 2020-10-25 by Mimiix

Smanar, 2020-10-25:
Hello, but why you have so much ghost device ?
You have all recently ? You havent some sensor not used ATM but already included one time ?

Mimiix, 2020-10-25:
@Smanar this issue is closed.

Smanar, 2020-10-25:
Seriously the issue is correct, it s not "tricking the bot", all informations are gived.

Mimiix, 2020-10-25:
The information is not provided as per request in the template. Thus not correct.

Please check your pm on discord.