dresden-elektronik/deconz-rest-plugin #3508: Commands to Aqara Curtain Motor B1 are not working

wvuyk, 2020-10-25:
## Describe the bug

We are trying to get the Aqara Curtain Motor working. But commands sent with updating `"bri"=nnn`, `"open"=true/false` or `"lift"=50` are not executed by the device. 

I am not sure if the device is fully implemented as #1654 has been closed by the bot and it seems to have an open end.
We tried sending commands directly through deCONZ using the cluster window. When sending any command "success" is returned, but the device does not show any response, like opening or closing. There is one exception to that, if you start an open or close by pressing the button on the curtain motor. Sending a "Stop" through the cluster will then stop the device as only working command.

Rejoining the device does not make a difference

## Steps to reproduce the behavior

Open the Window Covering cluster and execute either "Up/Open" or "Down/Close". The device will not respond.
Press the open/close button on the physical device and execute "Stop" on the cluster. This works

## Expected behavior

Using the buttons on the Window Covering Cluster works. We then hope the commands through REST API will start working too.
## Screenshots

[image]

[image]

## Environment

 - Host system:  PC 
 - Running method:  Windows
 - Firmware version: 26650700
 - deCONZ version: 2.05.86
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - no

## deCONZ Logs

## Additional context

ebaauw, 2020-10-25:
This device uses the _Analog Output_ cluster instead of the _Window Covering_ cluster.  In the GUI, you need to write the inverted percentage tilt to the _Present Value_ attribute.  0 is fully closed and 100 is fully open (it's reversed from the API).  I'm pleasently surprised it actually reacts to _Stop_.

The API should translate setting `open` or `tilt` to the appropriate _Write Attributes_ command.  There should be a debug message in the deCONZ log:
```
write attribute of 0x%016llX ep: 0x%02X cluster: 0x000D: 0x0055
```
Looking at the code, `{"stop": true}` should result in a _Stop_ command alright.

ebaauw, 2020-10-25:
Just to be sure, is the _Model Identifier_ `lumi.curtain.hagl04`?

SwoopX, 2020-10-25:
The float values now work via GUI!

wvuyk, 2020-10-25:
@ebaauw 

> The API should translate setting open or tilt to the appropriate Write Attributes command. There should be a debug message in the deCONZ log

My challenge so far that I have not been able to find any logging in windows. do you know where I could find this?

> Looking at the code, {"stop": true} should result in a Stop command alright.

My guess is that it would work too. But during testing we went for the Window covering cluster and never checked back.
We will test the _Analog output cluster_ and when I am able to test with the user again, we will double check on the logging

Will be back with the results.

ebaauw, 2020-10-25:
> The float values now work via GUI which wasn't the case back in the days!

GUI was broken for less common data types, but that got fixed.

wvuyk, 2020-10-25:
And, sorry missed that one, it is`lumi.curtain.hagl04`

[image]

wvuyk, 2020-10-28:
I did some further testing and it seems all commands are working. Here is the REST API returned. The only thing that seems to miss is the battery information, which could be useful? I copied a copy of power configuration. It is seems to miss the voltage information, but has a battery percentage of 186 (was 196 originally). We can load it to maximium to see that 100% would be.

```
{
"etag": "a6d45ebda596a59710f94b9cd9f6e2fa",
"hascolor": false,
"lastannounced": "2020-10-28T10:28:00Z",
"lastseen": "2020-10-28T14:47Z",
"manufacturername": "LUMI",
"modelid": "lumi.curtain.hagl04",
"name": "Window covering device 4",
"state": {
"bri": 254,
"lift": 100,
"on": true,
"open": false,
"reachable": true
},
"swversion": "20190329",
"type": "Window covering device",
"uniqueid": "04:cf:8c:df:3c:75:68:f3-01"
}
```

[image]

ebaauw, 2020-10-28:
> I did some further testing and it seems all commands are working.

That's good to know.  Thanks for reporting that.  I tested on my `lumi.curtain` (by hacking the code to treat it like the B1), but never got confirmation.

> The only thing that seems to miss is the battery information, which could be useful? I copied a copy of power configuration. It is seems to miss the voltage information, but has a battery percentage of 186 (was 196 originally). We can load it to maximium to see that 100% would be.

This is the first Xiaomi device that actually seems to support the _Power Configuration_ cluster.  _Battery Percentage Remaining_ is in 0.5% as per the ZCL spec, so 200 should be the maximum value.  We probably just need to whitelist the B1 to create a ZHABattery sensor, as we do for the IKEA Fyrtur and Kadrilj.

wvuyk, 2020-10-28:
I also noticed that the stop command cannot be done with sending a `"bri-inc"=0` it needed a `"stop"=true` as only command, could not be combined. So I guess the `"stop"=true` is now the way to go.

Whitelisting is all that is needed for the battery to be shown?

ebaauw, 2020-10-28:
`bri_inc` is explicitly disabled, but `stop` isn't.  I think the latter is an oversight on my part, assuming the B1 woudn't accept any _Window Covering_ command.  I'll skip the check for `bri_inc` (although it should be considered deprecated in favour of `stop`).

Indeed, the creation of the ZHABattery `/sensors` resource should be a simple whitelist.  Now sure whether `state.battery` will be updated, though.  We would need some logging/sniffing to see if battery is reported through the Xiaomi special attribute report, or whether the B1 sends reports to the coordinator by default (as most Xiaomi sensors do).  I don't suppose it supports binding the _Power Configuration_ cluster and setting up attribute reporting for the _Battery Percentage Remaining_?

There's also some workaround in the code for a deCONZ bug, which should have been fixed by now, but I better test that before making the change.  That'll have to wait until the weekend.

closed 2020-10-29 by manup

kevinvugts, 2021-05-21:
@ebaauw 

Hi Eric,

I just received my Aqara Xiaomi B1 motor and installed it.
I added it to Deconz via home assistant but cannot get it to work properly.

The state does not seem to work, neither does up/down works.
Any advice what to do? 

Best,