dresden-elektronik/deconz-rest-plugin #3516: Devices no longer responsive after a light bulb swap

ericvandamme, 2020-10-26:
## Describe the bug
I had a few light bulbs that were ending end of life (they were flickering) so they were deleted from deCONZ using the web interface and replaced with new Ikea Tradfri LED bulbs (904.084.86 - E26). Since, various devices have been falling off the network and becoming unresponsive and requiring to be repaired. This included three Ikea Tradfri relays, two Ikea Fyrtur blinds, a Danalock, and a Philips Hue bulb.

## Steps to reproduce the behavior
It is not certain if this is reproducible, but I will be having upcoming similar light bulb replacements necessary and will monitor the situation. This situation did not occur when a similar replacement was done using Osram/LEDVANCE lights.

## Expected behavior
Replacement of light bulbs should not result in other devices also removing themselves from the Zigbee network.

## Screenshots
There is no apparent effects in GUI. The devices appear to remain active and controllable in the interfaces; however, reading or sending commands directly through the cluster will timeout.

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian Linux 10
 - Firmware version: 26390500
 - deCONZ version: 2.05.84 / 2020-09-14
 - Device: ConBee I
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
Will be willing to set appropriate debug log levels in upcoming replacements

## Additional context
52 Devices (30 are considered routers)
120V
Consists of devices manufactured by:
- Philips (Lights, Remote, Motion Sensors)
- Ikea (Covers, Lights, Relays)
- LEDVANCE (Lights)
- Osram (Lights)
- Danalock (Lock)
- Schlage (Lock)
- Sengled (Lights)
- Sinope Technologies (Dimmer)
- RunLessWire (Remote)

manup, 2020-10-26:
Normally the only way to dropped off devices after a replacement should be Xiaomi battery powered devices which often stick to a specific parent. Doesn't seem to be the case here.

How long after the replacement did you wait for these tests? At least routers and the coordinator should adapt to the changes after a while. Have you tried power-cycling the devices? Repairing almost always shouldn't be needed (with the exception of Xiaomi).

ericvandamme, 2020-10-26:
I had some initial devices that no longer responded instantly afterwards (October 23), and then more devices began being unresponsive afterwards, though it cannot be certain that it didn't occur at the same time - like the Ikea blinds and the Danalock. After a few power cycles and waiting two days, devices were just repaired.

The exception being the one lone Philips Hue bulb. It was responsive until last night (October 25). Now it looks to need to be repaired as well, but maybe with the Hue Thief as normal pairing isn't doing the trick.

ericvandamme, 2020-10-28:
To add additional context, two additional devices (Ikea TRÅDFRI Wireless control outlet) fell off the network and needed to be repaired to the network.

closed 2020-12-19 by stale[bot]