dresden-elektronik/deconz-rest-plugin #3518: deCONZ 2.05.86 Windows 10 Gateway

UweRP, 2020-10-27:
Hallo.

## Describe the bug

Wenn ich deCONZ von 2.05.84 auf 2.05.86 unter Windows 10 x64 update wird der Conbee II Stick nicht mehr von der Phoscon App gefunden. deConz selbst zeigt den Stick und die Geräte an aber ohne Namen. Gehe ich zurück auf die 2.05.84 funktioniert es wieder wie erwartet. 
## Steps to reproduce the behavior

## Expected behavior

## Screenshots

Version 2.05.86 -- fehlerhaft
[image]
[image]

Version 2.05.84 -- ist ok
[image]
[image]

## Environment

 - Host system: PC
 - Running method: Windows 10 x64
 - Firmware version: 0x26660700
 - deCONZ version: 2.05.86
 - Device: ConBee II
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? If so: KVM (HDMI/USB Switch)

## deCONZ Logs

## Additional context

Mimiix, 2020-10-27:
@UweRP Hi! Can you translate the written text in English? This would help with you issue as not everyone speaks German. Thanks!

UweRP, 2020-10-27:
Hello Mimiix.
Sorry, I do not speak English. With the help of the Google translator here is the English text. I hope this is understandable.
--
When I update deCONZ from 2.05.84 to 2.05.86 under Windows 10 x64, the Conbee II stick is no longer found by the Phoscon app. deConz itself shows the stick and the devices but without a name. If I go back to the 2.05.84 it works again as expected.
--

Mimiix, 2020-10-27:
@UweRP Danke!

The translation is fine. I've asked @manup to look at this, as nothing has changed.

manup, 2020-10-27:
This looks a bit like loading the database fails in .86, could you please send the zll.db file which works in .84 to mpi@dresden-elektronik.de for investigation?

SwoopX, 2020-10-27:
Just as a side note: I've seen that previously while testing/doing some changes, some of them hilariously simple. I my case, it was definitively not related to the DB, but you never know.

UweRP, 2020-10-29:
Danke für die Antworten.
@manup die Mail mit der zll.db ist verschickt.

- Google translation
Thank you for the answers.
@manup the mail with the zll.db is sent.

manup, 2020-10-31:
Thanks, so I tested to load the zll.db file on my Windows PC with 2.05.86 here there were no issues, nodes are loaded fine and also have the names set in the database. I don't now yet why it fails on your machine yet. The next deCONZ Version will contain a little debug message window, hopefully this gives more insights where the error happens.

Rolf56, 2020-11-01:
Hi all
I had the same issue. After updating to version .86 under Windows 10, I lost all my devices in deCONZ and was not able to log into the Phoscon App web site. The message was the password ist wrong. When I tryed to change it to the default password another error massage was shown.  I can't remember the content of it.
After reinstallation version .84 everything is working fine as before.

manup, 2020-11-01:
Thanks I'll look into it, my last test was on a Windows 7 pc, I don't think this should make a difference but will test this on Windows 10 again.

brunoz78, 2020-11-07:
I have the same problem. Back to version 84 and it works fine again.

manup, 2020-11-13:
Does the issue remain with v2.5.88?

brunoz78, 2020-11-14:
Unfortunately the display problem is still present with the Verison 88. Back to version 84 and it will display everything correctly again.
[image]

UweRP, 2020-11-14:
Yes, I still have the problem with v2.5.88.

What else I noticed. A different installation path is suggested depending on the rights I have to start the installation.
The installation file does not automatically request admin rights to install.
The installation is in the directory c:\Program Files (x86)\deCONZ
If I start the installation without admin rights, c:\Users\Uwe\AppData\Local\deCONZ is suggested as the target.

Rolf56, 2020-11-14:
@UweRP This was always that way in the past. You have to right click and run as administrator. I'm always confused where to place the OTA files. I store them allway on both directories, but think Deconz is using C:\Users\<user>\otau instead of C:\Program Files (x86)\deCONZ\otau.

brunoz78, 2020-11-21:
Unfortunately the same problem with the new version 2.06.00. Firmware update was also done. With version 2.05.84 everything is still working.
[image]

Rolf56, 2020-11-28:
Hi all
I made another trial with version 2.6.0 under Windows 10. It has the same reaction, I lost the names of all my devices in deCONZ and was not able to log into the Phoscon App web site. 
[image]

After reinstallation version .84 everything is working fine as before.

I sent you an extract, out of the new debug view as a file to mpi@dresden-elektronik.de for investigation?
In that view, I can see a lot of HTTP/1.1 404 Errors. Especially when reading the sensors but also with other objects.

SwoopX, 2020-12-04:
Hey folks, could you please try something out? with regard to the lost node names, select one of the routers (yellowish), right click on it and then select reading the simple descriptors.

Would be interested in hearing if they get their names back. Thanks!

brunoz78, 2020-12-05:
The same result with simple descriptors, no difference. Tested with v2.07.00 and latest firmware 26680700. 
[image]

Rolf56, 2020-12-08:
Hi all, I still think it has to do with this HTTP Errors 404 Not Found.
I installed Version 2.07.00 now. No info for all devices, see the node list on the right too.
[image]

I now tried to connect with Postman and get following error:

[image]

The node itself gives an answer:

[image]

but it does not find /api

Rolf56, 2020-12-08:
Back to version 2.05.84 and everthing is OK-
I also get values and a time from ConBee. In the post before time was 01.01.2000 00:00

[image]

UweRP, 2020-12-12:
I would also like to refer you to the bug report https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3740
Presumably it's the same mistake.

Rolf56, 2020-12-12:
@brunoz78 : Hey Bruno. your right after copying the file de_rest_plugin.dll it runs also under 2.07.01
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3740#issuecomment-743002564

As I mention before it is the REST API what is bugy.
Hope a fix is available very soon!

brunoz78, 2020-12-16:
With version 2.08.00 everything works fine again. In addition, the display now also shows the battery level. Thank you very much for the repair.

Mimiix, 2020-12-16:
@UweRP Can you try on the latest version? It should work again.

UweRP, 2020-12-17:
Update from 2.05.84 to 2.08.00 seems to fix the problem. The names of the devices are displayed, the Phoscon app finds the ConBee II stick.
Many thanks.

Mimiix, 2020-12-17:
Case closed!

closed 2020-12-17 by Mimiix