dresden-elektronik/deconz-rest-plugin #3524: [Device Support Request] Zemismart 3 Gang Remote Switch  (2nd attempt)

stevewull, 2020-10-27:
## Device

- Product name: Zemismart Wireless Switch Work with Tuya Hub One Two Three Gangs Touch Switches Sticker
- Manufacturer: _TZ3400_key8kk7r
- Model identifier: TS0043
- Device type:
  - Switch

Listed here: https://zigbee.blakadder.com/Zemismart_ZM-ZS3.html

Purchased here: https://www.aliexpress.com/item/4000460643839.html?aff_platform=link-c-tool&cpt=1579460387259&sk=_sPA71V&aff_trace_key=1ea588879adf4a71b69472fcc8469bea-1579460387259-01187-_sPA71V&terminal_id=5ab3b3bf703c438dac102dcfef054d7e

I have the three gang version

## Screenshots

[image]

[image]

Switch 01:

[image]

[image]

[image]

[image]

[image]

Switch 02:
[image]

[image]

Switch 03:
[image]

[image]

Smanar, 2020-10-28:
Hello, so I have made a mistake on PR and removed a part of code, can you try this code pls.

To test it, you need an Unix OS, and take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only with a modification for step 1

```
git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

stevewull, 2020-10-31:
Thanks Smanar,

I will try to set up a test environment.  I have done this in the past. This has caused problems as I only have one Conbee and have to bring down the production environment. Some of the devices get confused and have trouble pairing properly when I try to restore them to production.

Smanar, 2020-11-01:
If you have an unix machine, no danger, just make a backup of the file lib (in other folder), and you just need 10s to rollback, the code don't change anything to c,onfig.

stevewull, 2020-11-07:
Hi Smanar,
I now have a Pi4  running the stable version of deConz that I can use for testing. You provided a link with instructions but I am not quire sure where to start.

Should I uninstall the stable version of deconz?
Should I start with the line that installs deconz-dev?

Smanar, 2020-11-08:
Hello, no depend of the deconz version you have, but there is a major modification on deconz core for button gestion (use now a json file), so I think it s better if you can install the last beta (if you have no issue with it)

Then after yes https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only

So it will be for you

```
sudo apt install deconz-dev
git clone --branch tuya https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you want to avoid re-installation, you can make a backup of the file "libde_rest_plugin.so", in another folder, so if you have problem, the rollback just take 30s.

stevewull, 2020-11-09:
Thanks for the instructions. Test environment is working using your branch for the plug-in.

The Zemismart switch now appears in the Phoscon App as three "On/Off lights."  

I think they are supposed to act like a switch with each of the three buttons having single press, long press, and hold.

Smanar, 2020-11-09:
It s not a wired switch ?

Smanar, 2020-11-09:
Sorry my bad, was wired device in my mind.
I have corrected the code to use them as wirelesss device now

Pls make a new try, will be faster

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

stevewull, 2020-11-10:
Ok. I loaded your new code.  Not sure exactly what to check for testing.

Each of the three buttons still show up in Phoscon App as an on/off light.

Smanar, 2020-11-10:
You have deleted them, before the inclusion ?

Smanar, 2020-11-11:
Ok I think I have found the problem, the tuya hack was working if you have only 1 endpoint, can you make another try pls ?

stevewull, 2020-11-13:
Hi Sameer,
I performed a new git pull and compiled your code. I launched deConz and, using Phoscon app, removed the switch from deConz.

I then put the switch in pairing mode. The Phoscon app does not detect or display when I choose add new lights, switches, or sensors. 

However, the switch is visible as a node in the deConz gui, not connected to other nodes. The light on the switch continues to flash indicating that it did not leave pairing mode. I see the cluster info for three different on/off switches.

Separately, I connected the switch to a non-deConz controlled Zigbee network. This also detected the three clusters as on/off switches. I didn't see a way to detect single press, long press, hold.

Smanar, 2020-11-13:
Can you make a try.
- Set phoscon in permit join (add new sensor)
- Read again basic attribute on deconz, cluster 0x0000, button "read" probalby need to make the device awake in same time

The device will to be visible in sensor part but invisible in phoscon.

Smanar, 2020-11-14:
I m sorry but I need to finalize the code sunday if I want it be in next beta.
There is no one that can try the code ?
On "tuya" branch or "tuya_pr" (based on last code possible)

I just need to know if the "switches group" is working (mean all buttons are working in the same entry), else I will make a roolback to have 1 sensor device by endpoint.

Mimiix, 2020-11-14:
@Smanar release will be today.

stevewull, 2020-11-14:
testing now

stevewull, 2020-11-14:
When you say "device will be visible in sensor part".  Do you mean in the API?

stevewull, 2020-11-15:
I performed a git pull from the Tuya branch listed above. The switch shows up in the sensors section of the API. Three entries.

    "2": {
      "config": {
        "battery": 95,
        "on": true,
        "reachable": false
      },
      "ep": 1,
      "etag": "a3a55398dd4805c633e6d0d75e201e70",
      "lastseen": null,
      "manufacturername": "_TZ3400_key8kk7r",
      "mode": 1,
      "modelid": "TS0043",
      "name": "TS0043 2",
      "state": {
        "buttonevent": 1002,
        "lastupdated": "2020-11-10T04:31:17.116"
      },
      "type": "ZHASwitch",
      "uniqueid": "60:a4:23:ff:fe:a0:f1:30-01-0006"
    },
    "3": {
      "config": {
        "battery": 95,
        "on": true,
        "reachable": false
      },
      "ep": 2,
      "etag": "a3a55398dd4805c633e6d0d75e201e70",
      "lastseen": null,
      "manufacturername": "_TZ3400_key8kk7r",
      "mode": 1,
      "modelid": "TS0043",
      "name": "TS0043 3",
      "state": {
        "buttonevent": 2002,
        "lastupdated": "2020-11-10T04:31:21.146"
      },
      "type": "ZHASwitch",
      "uniqueid": "60:a4:23:ff:fe:a0:f1:30-02-0006"
    },
    "4": {
      "config": {
        "battery": 95,
        "on": true,
        "reachable": false
      },
      "ep": 3,
      "etag": "a3a55398dd4805c633e6d0d75e201e70",
      "lastseen": null,
      "manufacturername": "_TZ3400_key8kk7r",
      "mode": 1,
      "modelid": "TS0043",
      "name": "TS0043 4",
      "state": {
        "buttonevent": 3003,
        "lastupdated": "2020-11-10T04:31:48.870"
      },
      "type": "ZHASwitch",
      "uniqueid": "60:a4:23:ff:fe:a0:f1:30-03-0006"
    }
  }
}

Smanar, 2020-11-15:
@Mimiix I have ask manup, I have until tonight.

@stevewull It s strange, on other issue the user have the 3 entries grouped with the same code, BTW if it s working it s the principal, thx.

petosak, 2020-12-19:
Anybody knows how to change and use as a button?
[image]
HA recognize as switch but never switched anything

Smanar, 2020-12-19:
You have a battery device ?
I don't know HA, the 3 button No 1" are "light" ? If yes, there is a problem, don't delete your device but remake inclusion, you need to have a field that use "buttonevent"

petosak, 2020-12-19:
I have exactly same 3button controller as in this case:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3220
Each button can do different things. I thought that i'll can use push/hold/click for each button for action in HA....
What you mean "remake inclusion"? How can i do that?

Smanar, 2020-12-19:
Ok is the battery powered one, you nned to have a sensor entry with "buttonevent".
Just re-pair (or re-include) the device but without deleting the old entry.

petosak, 2020-12-20:
I still can't see events as in Aqara. Still see only On/Off for every 3 buttons. Is it right? On right side is one button 3state Aqara.
[image]

Smanar, 2020-12-21:
Not normal at all, you haven't 0x0006 cluster so to have on/off ....
Can you find the complete device JSON somewhere ?

petosak, 2020-12-21:
do you mean this?:
`{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "bdeab4a89df6691c0dd5acbe0d8b38b0",
    "lastseen": "2020-12-21T19:33Z",
    "manufacturername": "_TZ3000_bi6lpsew",
    "mode": 1,
    "modelid": "TS0043",
    "name": "TS0043",
    "state": {
        "buttonevent": 1002,
        "lastupdated": "2020-12-21T09:57:57.397"
    },
    "type": "ZHASwitch",
    "uniqueid": "5c:02:72:ff:fe:c2:43:78-01-0006"
}`

Smanar, 2020-12-21:
Yep exactly.
And I don't understand how you can have this display on HA !
There is no on/off entries, it s not you that have configured something ?

To use this device, you need to use the field "buttonevent", atm it s 1002, if you press another button, the value will change.

petosak, 2020-12-21:
I understand you. But HA gives me 3x switch which is not meaningfull
[image]
I can't make automation on buttonevent....(via GUI)
Only when switch is changed
[image]

stevewull, 2020-12-21:
I have the exact same issue using the battery device. Plan to use this in Home Assistant. Each of the three buttons shows up as an on/off light in HA.  

I had given up. Will make another attempt to detect button events.

Steve

Smanar, 2020-12-22:
You can have 3 lights created by error during inclusion ( and you can remove them in API) But they are not usable, you need to use the device with name "TS0043" for you ? (it s the one it have on json)

clovisdivay, 2021-01-03:
I have the same issue with this product I just purchased (it seems to be a rebrand of the same switch) :
https://fr.aliexpress.com/item/1005001687942333.html?spm=a2g0s.9042311.0.0.4e3d6c375gipch

I hope you will be able to make it work :)

Smanar, 2021-01-03:
Do you have the basic info panel, and the basic attribute ? Tuya stuff can be different

clovisdivay, 2021-01-04:
Hi. I took screen captures of everything I could find. 
It's the first time I open the deCONZ app to find info so maybe I don't have everything you need, sorry. I also won't be able to test any beta versions as I use an integrated version of deCONZ in Home Assistant (https://www.home-assistant.io/integrations/deconz/) so I hope it will be enough.

Thanks for your help !

[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2021-01-04:
Manufact name being empty? That's a new one.

Smanar, 2021-01-04:
Yep this device seem present on the code 
>    { VENDOR_EMBER, "TS0043", silabs7MacPrefix }, // Tuya wireless switch

Iike have said Mimix, you can check the manufacture name, you NEED to have one of those
```

                    else if ((sensorNode->manufacturer() == QLatin1String("_TZ3000_bi6lpsew")) ||
                        (sensorNode->manufacturer() == QLatin1String("_TZ3400_keyjhapk")) ||
                        (sensorNode->manufacturer() == QLatin1String("_TYZB02_key8kk7r")) ||
                        (sensorNode->manufacturer() == QLatin1String("_TZ3400_keyjqthh")) ||
                        (sensorNode->manufacturer() == QLatin1String("_TZ3400_key8kk7r")) ||
                        (sensorNode->manufacturer() == QLatin1String("_TZ3000_vp6clf9d")) ||
                        (sensorNode->manufacturer() == QLatin1String("_TYZB02_keyjqthh")))
```

If you haven't it, deconz can reconize it, but if you can see it in deconz, the API can use deconz value to help for inclusion.

But your problem is a problem to include it ?

petosak, 2021-01-04:
> Yep this device seem present on the code
> 
> > { VENDOR_EMBER, "TS0043", silabs7MacPrefix }, // Tuya wireless switch
> 
> Iike have said Mimix, you can check the manufacture name again (it faild sometime), you NEED to have one of those
> 
> ```
> 
>                     else if ((sensorNode->manufacturer() == QLatin1String("_TZ3000_bi6lpsew")) ||
>                         (sensorNode->manufacturer() == QLatin1String("_TZ3400_keyjhapk")) ||
>                         (sensorNode->manufacturer() == QLatin1String("_TYZB02_key8kk7r")) ||
>                         (sensorNode->manufacturer() == QLatin1String("_TZ3400_keyjqthh")) ||
>                         (sensorNode->manufacturer() == QLatin1String("_TZ3400_key8kk7r")) ||
>                         (sensorNode->manufacturer() == QLatin1String("_TZ3000_vp6clf9d")) ||
>                         (sensorNode->manufacturer() == QLatin1String("_TYZB02_keyjqthh")))
> ```
> 
> If you haven't it, the API can't reconize it, but if you can see it in deconz, the API can use deconz value to help for inclusion.
> 
> But your problem is a problem to include it ?

My battery powered has this:
[image]
But still is as 3x on/off switch that doesn't make sense....

clovisdivay, 2021-01-04:
@Smanar I once managed to get a manufactor name, which was `_TZ3400_key8kk7r`. But now every time I try to apair it, it shows nothing...

Smanar, 2021-01-05:
@petosak yes can be normal, this device create normally 3 switches (visibles as lights), deconz create a sensor device and try to prevent the switches creation.
But It can faild so you will have the 3 switches + the sensor, only the sensor is usable.

The procedure can need 2 inclusion, the first one to have the manufacture name to reconize the device, and the second one to create the sensor and block the switches creations.

Take a look in the sensors part if you haven't a new device ?

@clovisdivay where are you looking ? in deconz , phoscon or third app ?
If you are in deconz, you can't faild.
- set phoscon in permit join (add new sensor)
- On deconz ask for basic attribute, (the button read that ask for manufacture name)
- Repeat asking for basic attribute

You can avoid the creation of the 3 bad switches if you set phoscon in permit join beetween the 2 attributes asking.

Remember you need to wake up the device when sending command.

petosak, 2021-01-05:
@Smanar i'm not sure what you mean:
Take a look in the sensors part if you haven't a new device ?
But i can catch events in HASSIO:
`Event 9 fired 9:37 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "ts0043",
        "unique_id": "5c:02:72:ff:fe:c2:43:78",
        "event": 2002,
        "device_id": "c46838e8661a564b792a8c11de56b9f2"
    },
    "origin": "LOCAL",
    "time_fired": "2021-01-05T20:37:13.331299+00:00",
    "context": {
        "id": "fb89f6a405497727d3a5d847bb250966",
        "parent_id": null,
        "user_id": null
    }
}
Event 8 fired 9:37 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "ts0043",
        "unique_id": "5c:02:72:ff:fe:c2:43:78",
        "event": 2002,
        "device_id": "c46838e8661a564b792a8c11de56b9f2"
    },
    "origin": "LOCAL",
    "time_fired": "2021-01-05T20:37:12.323096+00:00",
    "context": {
        "id": "fee5a52c82f85c68fd4a70dbceacc849",
        "parent_id": null,
        "user_id": null
    }
}`
I don't know why two times, but it's usable now.
In Phoscon i can see 3x light:
[image]

[image]
I really appreciate your help.

Smanar, 2021-01-06:
But light are notworking if I m right you need to use the event to use this device.
> "event": 2002,

2002 mean you have used the second button. But I suspect you have the double event problem ?

petosak, 2021-01-06:
Yes. You're right. Lights not working.
I know how to use events. I meant that i see 2 events on one push button. On every one push/click/hold i get 2 events.
I can see only diffent id. I don't know why, but it works.

Smanar, 2021-01-06:
It s not your fault ^^.
I m still searching why this double event, in my mind was the pooling, but if you have the problem for other event, for me it s realy 2 command send by the device.
And I don't understand why some users don't have the problem.

clovisdivay, 2021-01-06:
@Smanar Okay I did what you asked, now I have the manufacturer's name.
I don't have the 3 switches in Phoscon (I removed them before doing what you told me), but I don't have anything now... Maybe I should start over ? Should I reset the switch to factory settings and start the process over ?

[image]

Smanar, 2021-01-06:
All seem fine.
If you realy have nothing in sensor (not in light), in your third app, the sensor can be invisible in phoscon.
Just set phoscon in permit join (add new sensor/light), and read again the basic attribute (a second time), deconz already know the manufacture name, so you will not have the light creation, but you need have the sensor one.
Else can you show the log during the inclusion ?

clovisdivay, 2021-01-06:
Okay it seems to work now, though it is still invisible in Phoscon (not a big deal for me).

I also have the "double event" problem

Smanar, 2021-01-07:
@clovisdivay I don't think it will change something, but can you show me the long during the double event too.
On deconz / help with debug log 'info" and "info_l2"

clovisdivay, 2021-01-07:
INFO : 
[image]

For INFO_l2, there is too much going on in the log, I don't know what would be of interest for you. What should I be looking for ?

Smanar, 2021-01-07:
Right, hard to see something usefull.
If I add some debug line in code, are you able to recompile it to test it ?

clovisdivay, 2021-01-08:
Sorry, no :( I use deconz as a plugin in Home Assistant (I think it uses pydeconz) and I can't chose the version.

Flow-tech-lab, 2021-01-15:
> Yep this device seem present on the code
>
>
> Iike have said Mimix, you can check the manufacture name again (it faild sometime), you NEED to have one of those
>
> ```
>
>                     else if [...]

Hello @Smanar, first of all, thank you all the work on these Tuya switches!

I have a Zemismart 3gang switch and tried everything you described, but never saw the sensor in the Rest API.

Might it be because of the manufacturer name? Mine is "_TZ0300_qzjcsmar", which is not in your list. Model identifier is correct, TS0043.

Smanar, 2021-01-15:
If is a battey one, yes.
I can add it, but be sure of the manufacture name ^^, I think there is a typo in "_TZ0300_qzjcsmar".

Flow-tech-lab, 2021-01-15:
It is a battery one, yes. I think, I got the name right:
[image]

Smanar, 2021-01-15:
Just added here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4157

Flow-tech-lab, 2021-01-15:
Awesome, that was quick!
I already made a git pull to try it, everything looks fine in deconz, but API still doesnt show it as a sensor. Must be missing something

Smanar, 2021-01-15:
Yep, you probably need to make 2 inclusions, without deleting the old one.

Flow-tech-lab, 2021-01-18:
I dont really understand, how you mean the 2 inclusions, but I got it to work, at least partly:

- Phoscon shows the device under "switches"
- My Rest API Client shows the following:

[image]

So, battery state shows correct,  but **"buttonevent": null** is not correct, right?

Smanar, 2021-01-18:
Depend of the version you are using.
This device was added 3 days ago > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3524#issuecomment-760974335

So if you use this branch it will work, (no need to remake the inclusion)

Edit :
Haaaaaa, yes I remember, you need too edit the json file, recompiling the code is not enough, sorry.
You need to edit the file  button_maps.json with the modification visible here https://github.com/dresden-elektronik/deconz-rest-plugin/pull/4157/files

Flow-tech-lab, 2021-01-18:
Really appreciate your help, thanks a lot! I already use this branch with the changes in button_maps.json and  de_web_plugin.cpp.  This is how I did it:

`
[image]
`
Included it two times in Phoscon, see it as a switch (not as light, not as sensor).
Was this the correct way? I am not sure what else you mean with recompiling the code.

It seems, that everything is recognized correctly, just the buttonevent has no value at all.. Here a screenshot from my iobroker:

[image]

Smanar, 2021-01-18:
Tuya are boring to include, it s ok now ^^, stop to include it.

Your method is good, the too, else you will be not able to see it in the API, you are not using docker ?
If you press a button this value need to change.

Else if you have the GUI, can you take a look in log when pressing a button ? in deconz/help/log with "info" and "info_l2"

Flow-tech-lab, 2021-01-19:
Ok, so in "info" I can see:

 [INFO] - No button map for: TS0043 endpoint: **0x01** cluster: ONOFF (0x0006) command: LIDL (0xFD) payload[0]: **002**

endpoint 0x01, 0x02 or 0x03, for each of the three buttons.
payload[0]: 000 for short press or 002 for long press.

Is there a way to solve the "No button map" issue? button_maps.json is already the right one..

[5 more replies]

closed 2021-03-12 by github-actions[bot]