dresden-elektronik/deconz-rest-plugin #3526: Is my RaspBee II broken ?

stijink, 2020-10-28:
## Describe the question or issue you are having
I just got my RaspBee 2 today. But it seems to be broken. I already tried to update the firmware. But i run into errors here as well. At least the RTC seems to be working. :-/

I'm running out of ideas how to debug this problem and how to proceed. Also i don't want to spend hours/days debugging the issue.

Here's the output of my upgrade attemps. (I tried the local and the docker version)
```
home-pi ➜  ~ docker run -it --rm --entrypoint "/firmware-update.sh" --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz

-------------------------------------------------------------------

             marthoc/deconz Firmware Flashing Script

                       Version: 0.6

-------------------------------------------------------------------

Listing attached devices...

GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee

Enter the full device path, or press Enter now to exit.

Device Path : /dev/ttyAMA0

-------------------------------------------------------------------

Firmware available for flashing:
deCONZ_ConBeeII_0x26580700.bin.GCF
deCONZ_ConBeeII_0x26660700.bin.GCF
deCONZ_RaspBeeII_0x26660700.bin.GCF
deCONZ_Rpi_0x26350500.bin.GCF
deCONZ_Rpi_0x26390500.bin.GCF

Enter the firmware file name from above, including extension.
Alternatively, you may enter the name of a firmware file to download
from http://deconz.dresden-elektronik.de/deconz-firmware/
or press Enter now to exit.

File Name : deCONZ_RaspBeeII_0x26660700.bin.GCF

-------------------------------------------------------------------

Device: /dev/ttyAMA0

Firmware File: deCONZ_RaspBeeII_0x26660700.bin.GCF

Are the above device and firmware values correct?
Enter Y to proceed, any other entry to exit: y

Flashing...

GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyAMA0 (RaspBee)

867: Error: (assert) n == CMD_deconzVersion.size()

Flashing Error! Please re-run this script...

home-pi ➜  ~
```

## Screenshots
[image]
[image]

## Environment
 - Host system: Raspberry Pi 3B+
 - Running method: Marthoc Docker container
 - Firmware version: (nicht verbunden)
 - deCONZ version: 2.05.86 / 15.10.2020
 - Device: RaspBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
```
home-pi ➜  ~ docker logs 2465b6778bc4012fe456b5328a5091b70eace8b8be5deefd86d91f30c74f93ad --follow
[marthoc/deconz] Starting deCONZ...
[marthoc/deconz] Current deCONZ version: 2.05.86
[marthoc/deconz] Web UI port: 80
[marthoc/deconz] Websockets port: 443
[marthoc/deconz] VNC Disabled
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()

16:07:59:700 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
16:07:59:728 CTRL. 3.27.216:07:59:892 dev /dev/ttyAMA0
16:07:59:892 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
16:08:00:216 parent process /bin/sh
16:08:00:217 gw run mode: docker
16:08:00:217 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
16:08:00:217 sd-card cid: 035344534e36344780ffffffff013281
16:08:00:219 DB sqlite version 3.27.2
16:08:00:219 DB PRAGMA page_count: 0
16:08:00:219 DB PRAGMA page_size: 4096
16:08:00:219 DB PRAGMA freelist_count: 0
16:08:00:219 DB file size 0 bytes, free pages 0
16:08:00:220 DB PRAGMA user_version: 0
16:08:00:220 DB upgrade to user_version 1
16:08:00:558 DB write sqlite user_version 1
16:08:00:570 DB PRAGMA user_version: 1
16:08:00:570 DB upgrade to user_version 2
16:08:00:615 DB write sqlite user_version 2
16:08:00:626 DB PRAGMA user_version: 2
16:08:00:626 DB upgrade to user_version 6
16:08:00:681 DB write sqlite user_version 6
16:08:00:692 DB PRAGMA user_version: 6
16:08:00:692 DB upgrade to user_version 7
16:08:00:731 DB write sqlite user_version 7
16:08:00:742 DB PRAGMA user_version: 7
16:08:00:742 DB cleanup
16:08:00:743 DB create temporary views
15:08:00:763 started websocket server at port 443
15:08:00:767 create default username and password
15:08:00:773 [INFO] - Found file containing button maps. Parsing data...
15:08:00:784 [ERROR] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
15:08:00:804 [INFO] - Button maps loaded.
15:08:00:808 dlg action: Read binding table
15:08:00:808 found node plugin: libde_rest_plugin.so - REST API Plugin
15:08:00:814 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
15:08:00:867 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
15:08:00:924 dev /dev/ttyAMA0
15:08:02:030 dev /dev/ttyAMA0
15:08:02:050 device disconnected reason: 4, index: 0
15:08:02:975 wait reconnect 15 seconds
15:08:03:094 dev /dev/ttyAMA0
15:08:04:025 wait reconnect 14 seconds
15:08:05:067 wait reconnect 13 seconds
15:08:06:440 wait reconnect 12 seconds
15:08:06:496 dev /dev/ttyAMA0
15:08:06:703 Announced to internet https://phoscon.de/discover
15:08:06:703 discovery server date: Wed, 28 Oct 2020 15:08:06 GMT
15:08:06:704 	 local time seems to be ok
15:08:06:704 discovery found version 2.04.35 for update channel stable
15:08:07:067 wait reconnect 11 seconds
15:08:07:868 New websocket 127.0.0.1:56984 (state: 3)
15:08:08:067 wait reconnect 10 seconds
15:08:09:068 wait reconnect 9 seconds
15:08:10:068 wait reconnect 8 seconds
15:08:11:068 wait reconnect 7 seconds
15:08:12:067 wait reconnect 6 seconds
15:08:13:067 wait reconnect 5 seconds
15:08:14:067 wait reconnect 4 seconds
15:08:15:068 wait reconnect 3 seconds
15:08:16:068 wait reconnect 2 seconds
(goes on like this ...)
```

Smanar, 2020-10-28:
Hello, just by curiosity, you haven't another raspberry but without docker ?

stijink, 2020-10-29:
@Smanar If you would ask my wife she would say "too many". So i tested the lot on another Pi (Model 2). And the result is positive. I got it working. This time without docker. Thanks for the inspiration.

MrXlebNick, 2022-04-18:
Does it mean that if I have the same error, I should be replacing Raspberry Pi and not the Raspbee Shield?