dresden-elektronik/deconz-rest-plugin #3527: Device name

pur3asdf, 2020-10-28:
## Device

- Product name: [SR-ZG2868](https://www.sunricher.com/dim-cct-rgb-3-in-1-zigbee-remote-sr-zg2868.html)
- Manufacturer: Sunricher
- Model identifier: ZG2868A
- Device type :
-Remote

## Screenshots

  Required screenshots:

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

SirMaggot, 2020-12-11:
Interested in this.

closed 2021-03-12 by github-actions[bot]

pimw1, 2021-04-20:
Hi, i'm also very interested in this. I've the same remote. If needed, i can provide more screenshots. There are actually 4 sections with clusters, not 2. 
[image]
[image]

pimw1, 2021-05-01:
How can I bring this to attention?

Smanar, 2021-05-01:
Lol, good question.

But this device seem classic, why do you don't try yourself to add it ?
You can just mimic the code for a device that have same cluster list, except this one have 2 endpoint.

SirMaggot, 2021-05-01:
I did add my remote directly to my lights, like described in the manual, works like a champ.

pimw1, 2021-05-01:
Thanks! I'll take a look tomorrow. 
Regarding directly connecting it: that's what i am currently doing. But i want to do more advanced stuff via Home Assistant (for example: activating scenes containing not only zigbee bulbs, but also wifi bulbs).

pur3asdf, 2022-09-04:
Any update on this?

pimw1, 2022-09-04:
I didn't continued with this, the procedure sounds to complicated (i.e. not worth the effort). My remote is idling at home for 1.5 year now.