dresden-elektronik/deconz-rest-plugin #3528: Node disappears after a reboot of the deconz container

Jaguaza, 2020-10-28:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

[image]

 - Host system: NUC INTEL
 - Running method: HassOS runing proxmox
 - Firmware version: (26660700)
 - deCONZ version: (2.05.84)
 - Device:  ConBee II
 - Do you use an USB extension cable: Yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context

Jaguaza, 2020-10-28:
@dn0sar - Thanks for your interest. For me it is frustrating that this happens with my xiaomi devices. I hope they find the cause and can correct it soon.

Jaguaza, 2020-10-29:
Nobody knows why it doesn't work?

Smanar, 2020-10-29:
You don't have a PC to make a test ? (windows or ubuntu)

Install deconz on it and:
First make the try with a empty database (with a clean instal)
And the second time with using the backup/restaure feature availalble in phoscon to transfert your database ?

Jaguaza, 2020-10-29:
> ¿No tienes una PC para hacer una prueba? (windows o ubuntu)
> 
> Instale deconz en él y:
> Primero intente con una base de datos vacía (con una instalación limpia) ¿
> Y la segunda vez con el uso de la función de respaldo / restauración disponible en phoscon para transferir su base de datos?

In windows it is the same, I already tested. With other software it does not happen to me, only with deconz, therefore I decline to a denocz error.

Mimiix, 2020-10-29:
Xiaomi sensors tend to behave odd  if they are connected to the coordinator. If the coordinator then powers off/ reboots they either:
- Drop off
- Come back after 1 hour.

Fix? Put some routers in there.

Ps: Bumping doesn't work on github 😉

Jaguaza, 2020-10-29:
> Los sensores de Xiaomi tienden a comportarse de manera extraña si están conectados al coordinador. Si el coordinador luego se apaga / reinicia:
> 
> * Bajar ó dejar algo
> * Vuelve después de 1 hora.
> 
> ¿Reparar? Ponga algunos enrutadores allí.
> 
> Ps: golpear no funciona en github 😉

I have routers there
Xiaomi Temperature Sensors: 6 Model - WSDCGQ11LM
Xiaomi Window & Door Magnetic Sensors: 11model - MCCGQ01LM
Xiaomi flood sensors: 2 model - SJCGQ11LM
Xiaomi smoke sensor: 1 model - JTYJ-GD-01LM / BW
Xiaomi motion sensors: 3 model - RTCGQ11LM
Xiaomi plug: 1 model - ZNCZ02LM
Philips hue bulbs: 6 model - 8718696598283
Philips color hue bulb: 1 model - 9290012573A

Jaguaza, 2020-10-29:
> Xiaomi sensors tend to behave odd if they are connected to the coordinator. If the coordinator then powers off/ reboots they either:
> 
> * Drop off
> * Come back after 1 hour.
> 
> Fix? Put some routers in there.
> 
> Ps: Bumping doesn't work on github 😉

Why doesn't this happen with zigbee2mqtt?

Mimiix, 2020-10-30:
> > Xiaomi sensors tend to behave odd if they are connected to the coordinator. If the coordinator then powers off/ reboots they either:
> > 
> > * Drop off
> > * Come back after 1 hour.
> > 
> > Fix? Put some routers in there.
> > Ps: Bumping doesn't work on github 😉
> 
> Why doesn't this happen with zigbee2mqtt?

I have no clue, as i don't use that 😅 . I created a internal ticket. Manuel should reply later on.

manup, 2020-10-30:
>The node of the xiomi window and door sensors is not maintained. The failure is random, sometimes the node of a sensor is lost and in another reboot after adding it, a different one is lost.

Hi can you please tell how "lost" is meant here:

1) The device in deCONZ GUI is not there anymore;
2) The device is there but doesn't communicate;
3) The device isn't visible anymore in Phoscon App.

When working are the sensors connected directly to the coordinator (green line) or through a router?

Jaguaza, 2020-10-30:
> > El nodo de los sensores de puertas y ventanas xiomi no se mantiene. La falla es aleatoria, a veces se pierde el nodo de un sensor y en otro reinicio después de agregarlo se pierde uno diferente.
> 
> Hola, ¿puedes decir qué significa "perdido" aquí?
> 
> 1. El dispositivo en deCONZ GUI ya no está allí;
> 2. El dispositivo está ahí pero no se comunica;
> 3. El dispositivo ya no está visible en la aplicación Phoscon.
> 
> En funcionamiento, ¿los sensores están conectados directamente al coordinador (línea verde) oa través de un enrutador?

1. The device in deCONZ GUI is not there anymore;

manup, 2020-10-30:
Hmm that's very odd, this would indicate that the database is not or could not saved properly.

How long did you wait after device pairing and restart? According to the code the database should be written within 1 minute (usually after 5 seconds) after adding a new sensor.

Can you show a screenshot of the faulty device Node Info and Endpoint list: example can be found in https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Jaguaza, 2020-11-02:
> How long did you wait after device pairing and restart?

Tnanks @manup 

How long did you wait after device pairing and restart?  - Several minutes from 5 to 10 minutes for 31 device.

Can you show a screenshot of the faulty device Node Info and Endpoint list:
[image]
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-11-02:
Hu ?
What is the model id of one of those device ?
I m not sure but it seem deconz have skipped some informations, manufacture code  = 0 ?

Jaguaza, 2020-11-02:
> Hu ?
> What is the model id of one of those device ?
> I m not sure but it seem deconz have skipped some informations, manufacture code = 0 ?

They are all window and door sensors from xiaomi. Higher up I put the model.

@manup 
In the last reboot I lost 6 nodes, I have been waiting two hours after the reboot of the container and they have not returned. (I'm talking about nodes missing in deconz, not unconnected devices).

Before reboot:
[image]

After reboot:
[image]

Smanar, 2020-11-02:
I don't see the model id (it s in the basic cluster panel 0x0000)
It s "lumi.sensor_magnet", if yes the device is not correctly included in deconz, it have missed some information for it.

Jaguaza, 2020-11-02:
> Xiaomi Window & Door Magnetic Sensors: 11model - MCCGQ01LM

These are @Smanar

Smanar, 2020-11-02:
You knwow there is 3 version of WXKG02LM and all are working differently ?
To be sure we are talking about the same device, the only method is the model id send by the firmware.

Jaguaza, 2020-11-02:
I don't understand, my English is terrible

Jaguaza, 2020-11-02:
> You knwow there is 3 versions of WXKG02LM for exemple and all are working differently ?
> To be sure we are talking about the same device, the only method is the model id send by the firmware, but for me there is a problem during the device inclusion, you have the same result when making the inclusion close the gateway ?

I don't understand, my English is terrible

SwoopX, 2020-11-03:
@Smanar Those devices disappearing are the round Mija ones.

Jaguaza, 2020-11-03:
These are.  Let's see if we can find out why it disappears.  If @manup  or someone found the problem.  Thanks for trying to help @Smanar and @SwoopX. 

[image]
17 hours later the nodes are still missing.  As I said, it is not that they do not connect, it is that they disappear.
[image]
Of 31 node 6 missing.

Smanar, 2020-11-03:
Have you tried to include them close the gateway ?

I can find the correct picture, but you need to have different result, for exemple
- A real manufacture number
- A real type, not FFFF
- More cluster

I have found a similar issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1224

manup, 2020-11-17:
One more question, were the sensors which are gone after reboot listed in the Phoscon App and in there with proper model id?

eserero, 2020-11-17:
I have the exact same problem! 
I have the same sensor exactly - and I tried to add about 5 or 6 of them and they they disappear from deconz after restart - sometime they stay (and I have one that is always there), but in most cases they just disappear from deconz.
couple more observations:
1. they disappear from deconz, but stay on the web app (but of course not working)
2. if I pair them again - they re-appear in deconz in the exact same place they were before - but with the default name and in exact same place in the list of sensors in the web app. to me it indicates that they are not completely forgotten
3. I also see manufacturer 0000 and other details exactly like the screeshots from @Jaguaza

Jaguaza, 2020-11-17:
> One more question, were the sensors which are gone after reboot listed in the Phoscon App and in there with proper model id?

Yes.

SwoopX, 2020-11-18:
We need to have the full log of a deconz restart here. The logs above have stripped the important part where devices are loaded from DB. So a log starting deconz + having it run for 2-3 mins where you also try to trigger your devices vanished should hopefully give us more insights.

eserero, 2020-11-18:
here is the log from my system. 
After restart of deconz I lost both door sensors. - i can see some "no button handlers" for them in the logs.
I also noticed I lost a couple more sensors (smoke detector and motion sensor) - 
in the logs below I can see all of them with the "no button handlers" message.

10:40:39:419 HTTP Server listen on address 0.0.0.0, port: 8081, root: /usr/share/deCONZ/webapp/
10:40:39:429 CTRL. 3.27.210:40:39:462 COM: /dev/ttyACM0 / serialno:
10:40:39:462 COM: --dev: /dev/ttyACM0 (ConBee II)
10:40:39:462 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
10:40:39:519 parent process /bin/sh
10:40:39:519 gw run mode: docker
10:40:39:519 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
10:40:39:520 DB sqlite version 3.27.2
10:40:39:520 DB PRAGMA page_count: 46
10:40:39:520 DB PRAGMA page_size: 4096
10:40:39:520 DB PRAGMA freelist_count: 2
10:40:39:520 DB file size 188416 bytes, free pages 2
10:40:39:520 DB PRAGMA user_version: 7
10:40:39:520 DB cleanup
10:40:39:521 DB create temporary views
10:40:39:546 started websocket server at port 8443
10:40:39:547 [INFO] - Found file containing button maps. Parsing data...
10:40:39:547 [ERROR] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
10:40:39:549 [INFO] - Button maps loaded.
10:40:39:550 dlg action: Read binding table
10:40:39:550 found node plugin: libde_rest_plugin.so - REST API Plugin
10:40:39:550 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
10:40:39:554 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
10:40:39:570 COM: /dev/ttyACM0 / serialno:
10:40:39:570 COM: --dev: /dev/ttyACM0 (ConBee II)
10:40:40:596 COM: /dev/ttyACM0 / serialno:
10:40:40:596 COM: --dev: /dev/ttyACM0 (ConBee II)
10:40:40:604 New websocket 10.0.0.111:62710 (state: 3)
10:40:40:611 COM check bootloader
10:40:40:621 COM detected application
10:40:40:636 Device firmware version 0x26660700 ConBee II
10:40:40:642 unlocked max nodes: 200
10:40:40:712 Device protocol version: 0x010C
10:40:40:723 new node - ext: 0x00212effff04dce2, nwk: 0x0000
10:40:40:768 LightNode 1: Living Room Center Light added
10:40:40:770 LightNode 3: Computer Light added
10:40:40:771 LightNode 2: Kids Light 1 added
10:40:40:772 LightNode 4: Parent Room Light 1 added
10:40:40:773 LightNode 5: Parent Room light 2 added
10:40:40:774 LightNode 6: Arbel Light 1 added
10:40:40:775 LightNode 7: Arbel Light 2 added
10:40:40:776 SensorNode 4 set node 0x00158d00029bf2e6
10:40:40:776 SensorNode 3 set node 0x00158d00029bf2e6
10:40:40:778 SensorNode 5 set node 0x00158d0001243f52
10:40:40:779 LightNode 8: Parents Light 3 (entry) added
10:40:40:780 SensorNode 6 set node 0x00158d00015b5e79
10:40:40:780 SensorNode 10 set node 0x04cf8cdf3c773ce1
10:40:40:781 SensorNode 7 set node 0x00158d0002720ca7
10:40:40:781 SensorNode 8 set node 0x00158d0002720ca7
10:40:40:781 SensorNode 9 set node 0x00158d0002720ca7
10:40:40:782 SensorNode 15 set node 0x00158d0002434415
10:40:40:782 SensorNode 16 set node 0x00158d0002386200
10:40:40:783 SensorNode 19 set node 0x00158d0001c19f9a
10:40:40:783 SensorNode 20 set node 0x00158d0001c19f9a
10:40:40:785 SensorNode 24 set node 0x04cf8cdf3c77c1a4
10:40:40:786 SensorNode 21 set node 0x04cf8cdf3c75b783
10:40:40:786 SensorNode 22 set node 0x04cf8cdf3c77b465
10:40:40:787 SensorNode 23 set node 0x04cf8cdf3c77b629
10:40:40:788 SensorNode 26 set node 0x00158d00027264b1
10:40:40:788 SensorNode 27 set node 0x00158d00027264b1
10:40:40:788 SensorNode 28 set node 0x00158d00027264b1
10:40:40:789 SensorNode 33 set node 0x00158d00024e8ee7
10:40:40:789 SensorNode 34 set node 0x00158d0002484b80
10:40:40:790 SensorNode 35 set node 0x00158d00015b5e60
10:40:40:791 SensorNode 37 set node 0x00158d00019dc16f
[... 235 lines cut ...]
10:44:34:181 ZCL attribute report 0x00158D0002720CA7 for cluster: 0x0403, ep: 0x01, frame control: 0x18, mfcode: 0x0000
10:44:39:067 [INFO] - No button handler for: lumi.sen_ill.mgl01 endpoint: 0x01 cluster: 0x0400 command: 0x0A payload[0]: 000
10:44:39:068 ZCL attribute report 0x04CF8CDF3C773CE1 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
10:44:39:562 Current channel 15
10:44:39:570 Device TTL 568 s flags: 0x7
10:44:39:570 Device reset watchdog 7200 s
10:44:41:377 ZCL attribute report 0x04CF8CDF3C7E508A for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F
10:44:41:378 0x04CF8CDF3C7E508A extract Xiaomi special attribute 0x00F7
10:44:41:378    64 on/off 0
10:44:41:378    03 temperature 25 °C
10:44:41:378    98 power 0.000000 (0)
10:44:41:378    95 consumption 1.697045 (1697)
10:44:41:378    96 voltage 2330.000000 (233)
10:44:41:378    97 current 0.000000 (0)
10:44:41:378    05 RSSI dB (?) 20 (0x0014)
10:44:41:378    9a unknown 0 (0x00)
10:44:41:378    08 unknown 278 (0x0116)
10:44:41:378    07 unknown 0 (0x0000000000000000)
10:44:41:378    09 unknown 2306 (0x0902)
10:44:41:378    0b unknown 0 (0x0000)
10:44:41:378    9b Consumer connected (yes/no) 1
10:44:44:053 [INFO] - No button handler for: lumi.sen_ill.mgl01 endpoint: 0x01 cluster: 0x0400 command: 0x0A payload[0]: 000
10:44:44:053 ZCL attribute report 0x04CF8CDF3C773CE1 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
10:44:44:570 GW firmware version: 0x26660700
10:44:44:570 GW firmware version is up to date: 0x26660700

Jaguaza, 2020-11-18:
> here is the log from my system.
> After restart of deconz I lost both door sensors. - i can see some "no button handlers" for them in the logs.
> I also noticed I lost a couple more sensors (smoke detector and motion sensor) -
> in the logs below I can see all of them with the "no button handlers" message.
> 
> 10:40:39:419 HTTP Server listen on address 0.0.0.0, port: 8081, root: /usr/share/deCONZ/webapp/
> 10:40:39:429 CTRL. 3.27.210:40:39:462 COM: /dev/ttyACM0 / serialno:
> 10:40:39:462 COM: --dev: /dev/ttyACM0 (ConBee II)
> 10:40:39:462 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
> 10:40:39:519 parent process /bin/sh
> 10:40:39:519 gw run mode: docker
> 10:40:39:519 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
> 10:40:39:520 DB sqlite version 3.27.2
> 10:40:39:520 DB PRAGMA page_count: 46
> 10:40:39:520 DB PRAGMA page_size: 4096
> 10:40:39:520 DB PRAGMA freelist_count: 2
> 10:40:39:520 DB file size 188416 bytes, free pages 2
> 10:40:39:520 DB PRAGMA user_version: 7
> 10:40:39:520 DB cleanup
> 10:40:39:521 DB create temporary views
> 10:40:39:546 started websocket server at port 8443
> 10:40:39:547 [INFO] - Found file containing button maps. Parsing data...
> 10:40:39:547 [ERROR] - Button map 'sunricherCCTMap' in JSON file has no assigned ModelIDs. Skip loading button map...
> 10:40:39:549 [INFO] - Button maps loaded.
> 10:40:39:550 dlg action: Read binding table
> 10:40:39:550 found node plugin: libde_rest_plugin.so - REST API Plugin
> 10:40:39:550 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin
> 10:40:39:554 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin
> 10:40:39:570 COM: /dev/ttyACM0 / serialno:
> 10:40:39:570 COM: --dev: /dev/ttyACM0 (ConBee II)
> 10:40:40:596 COM: /dev/ttyACM0 / serialno:
> 10:40:40:596 COM: --dev: /dev/ttyACM0 (ConBee II)
> 10:40:40:604 New websocket 10.0.0.111:62710 (state: 3)
> 10:40:40:611 COM check bootloader
> 10:40:40:621 COM detected application
> 10:40:40:636 Device firmware version 0x26660700 ConBee II
> 10:40:40:642 unlocked max nodes: 200
> 10:40:40:712 Device protocol version: 0x010C
> 10:40:40:723 new node - ext: 0x00212effff04dce2, nwk: 0x0000
> 10:40:40:768 LightNode 1: Living Room Center Light added
> 10:40:40:770 LightNode 3: Computer Light added
> 10:40:40:771 LightNode 2: Kids Light 1 added
> 10:40:40:772 LightNode 4: Parent Room Light 1 added
> 10:40:40:773 LightNode 5: Parent Room light 2 added
> 10:40:40:774 LightNode 6: Arbel Light 1 added
> 10:40:40:775 LightNode 7: Arbel Light 2 added
> 10:40:40:776 SensorNode 4 set node 0x00158d00029bf2e6
> 10:40:40:776 SensorNode 3 set node 0x00158d00029bf2e6
> 10:40:40:778 SensorNode 5 set node 0x00158d0001243f52
> 10:40:40:779 LightNode 8: Parents Light 3 (entry) added
> 10:40:40:780 SensorNode 6 set node 0x00158d00015b5e79
> 10:40:40:780 SensorNode 10 set node 0x04cf8cdf3c773ce1
> 10:40:40:781 SensorNode 7 set node 0x00158d0002720ca7
> 10:40:40:781 SensorNode 8 set node 0x00158d0002720ca7
> 10:40:40:781 SensorNode 9 set node 0x00158d0002720ca7
> 10:40:40:782 SensorNode 15 set node 0x00158d0002434415
> 10:40:40:782 SensorNode 16 set node 0x00158d0002386200
> 10:40:40:783 SensorNode 19 set node 0x00158d0001c19f9a
> 10:40:40:783 SensorNode 20 set node 0x00158d0001c19f9a
> 10:40:40:785 SensorNode 24 set node 0x04cf8cdf3c77c1a4
> 10:40:40:786 SensorNode 21 set node 0x04cf8cdf3c75b783
> 10:40:40:786 SensorNode 22 set node 0x04cf8cdf3c77b465
> 10:40:40:787 SensorNode 23 set node 0x04cf8cdf3c77b629
> 10:40:40:788 SensorNode 26 set node 0x00158d00027264b1
> 10:40:40:788 SensorNode 27 set node 0x00158d00027264b1
> 10:40:40:788 SensorNode 28 set node 0x00158d00027264b1
> 10:40:40:789 SensorNode 33 set node 0x00158d00024e8ee7
> 10:40:40:789 SensorNode 34 set node 0x00158d0002484b80
> 10:40:40:790 SensorNode 35 set node 0x00158d00015b5e60
> 10:40:40:791 SensorNode 37 set node 0x00158d00019dc16f
[... 237 lines cut ...]
> 10:44:39:068 ZCL attribute report 0x04CF8CDF3C773CE1 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
> 10:44:39:562 Current channel 15
> 10:44:39:570 Device TTL 568 s flags: 0x7
> 10:44:39:570 Device reset watchdog 7200 s
> 10:44:41:377 ZCL attribute report 0x04CF8CDF3C7E508A for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F
> 10:44:41:378 0x04CF8CDF3C7E508A extract Xiaomi special attribute 0x00F7
> 10:44:41:378 64 on/off 0
> 10:44:41:378 03 temperature 25 °C
> 10:44:41:378 98 power 0.000000 (0)
> 10:44:41:378 95 consumption 1.697045 (1697)
> 10:44:41:378 96 voltage 2330.000000 (233)
> 10:44:41:378 97 current 0.000000 (0)
> 10:44:41:378 05 RSSI dB (?) 20 (0x0014)
> 10:44:41:378 9a unknown 0 (0x00)
> 10:44:41:378 08 unknown 278 (0x0116)
> 10:44:41:378 07 unknown 0 (0x0000000000000000)
> 10:44:41:378 09 unknown 2306 (0x0902)
> 10:44:41:378 0b unknown 0 (0x0000)
> 10:44:41:378 9b Consumer connected (yes/no) 1
> 10:44:44:053 [INFO] - No button handler for: lumi.sen_ill.mgl01 endpoint: 0x01 cluster: 0x0400 command: 0x0A payload[0]: 000
> 10:44:44:053 ZCL attribute report 0x04CF8CDF3C773CE1 for cluster: 0x0400, ep: 0x01, frame control: 0x18, mfcode: 0x0000
> 10:44:44:570 GW firmware version: 0x26660700
> 10:44:44:570 GW firmware version is up to date: 0x26660700

Thanks for your log.  Sorry I can't contribute right now, I'm out.

Jaguaza, 2020-12-03:
@manup  knows something about this.  In other systems like zha or zigbee2mqtt it doesn't happen.

Jaguaza, 2020-12-04:
😭

Jaguaza, 2020-12-10:
I did not expect that a company like Dresden Elektronik Ingenieurtechnik gmbh would let us down and not care about their customers that we have paid for their products.

Mimiix, 2020-12-10:
@Jaguaza Stuff's been crazy apperantly. I'll ask Manuel to check into this issue 👍

Jaguaza, 2020-12-10:
> @Jaguaza Stuff's been crazy apperantly. I'll ask Manuel to check into this issue 👍

Thanks @Mimiix.

manup, 2020-12-11:
Hi, sorry for the long wait time. I think I've found why the nodes don't load properly for these in the restore node from database code:

[image]

deCONZ only restores nodes from the database when they have a valid Node Descriptor, if I remember correctly some older Xiaomi Mija devices aren't very responsive when queried for some standard descriptors, so perhaps here the Node Descriptor is missing. The screenshot looks a bit like that since these are all default values, tricky.

I'll make some tests, one workaround could be to create a valid fake Node Descriptor from deCONZ for this specific model Id.
There is also another point where this might fail, as the above checks rely on the order of descriptors in the database this is fixed now in the SQL query by ordering by type. Should be fixable for the upcoming beta.

accelle17, 2020-12-11:
> some older Xiaomi Mija devices aren't very responsive when queried for some standard descriptors, so perhaps here the Node Descriptor is missing

Where do I see the node descriptor? I'd like to verify if this is the same with me. I have some xiaomi mijia devices that often dropped during reboot. My workaround was to at least do a reset on the node before I finish the joining.

manup, 2020-12-11:
In the deCONZ UI you can see it in the Node Info panel.
In the database file `zll.db` it's stored in the `device_descriptors` table with type 2.

manup, 2020-12-11:
Small update, I've found one of these round sensors in the company will do some tests with it later on.

manup, 2020-12-11:
So the problem appears to be indeed the Node Descriptor. When pairing the sensor it doesn't react to queries from the coordinator at all. But fortunately the sensor can be woke up to be responsive.

With the following workaround retrievees the missing data when pairing the sensor again:

## Pairing sensor again to show the node in deCONZ GUI

1. Open the Phoscon App &rarr; Menu &rarr; Sensors;
2. Click "Add new Sensor" &rarr; Other;
3. Press the button in the small pin hole like 10 seconds until the sensor LED blinks blue;
4. Open / close the sensor a few times, the sensor should now be added and the node shown in deCONZ GUI.

Nothing new yet as some data is missing.

[image]

## Retrieve missing descriptors in deCONZ GUI

5. Right click on the node and select **Read node descriptor**;
6. Quickly after that press the the pin hole button on the sensor;
7. Right click on the node and select **Read simple descriptors**;
8. Quickly after that press the the pin hole button on the sensor again.

[image]

When everything is queried it should like in the last screenshot and the sensor node survives a deCONZ reboot.
Note you can repeat the last steps if it doesn't work the first time.

eserero, 2020-12-14:
@manup  you are the king!!!! I fixed it on my system and no node disappear when restarting deconz.
BTW - I saw this issue on xiaomi door sensors and motion sensors - issue fixed in both. I will add more sensors tomorrow and see if this solve the problem. 
Many thanks again!

Mimiix, 2020-12-17:
@Jaguaza Any update on this :)?

Jaguaza, 2020-12-17:
> @Jaguaza Any update on this :)?

I'm out all month.

pluusla, 2020-12-17:
Got the same problem. But the workaround from manup doesn't work for me.. A fix would be much appreciated!

Mimiix, 2020-12-17:
@pluusla Can you open a seperate issue on it? Complete with logs and all?

wilfredsmit, 2021-02-02:
I had the same issue with Xiaomi Mi doorsensors and Xiaomi Mi buttons (round ones). The solution by @manup works for me. The sensors survives a restart of deconz. Thanx!

closed 2021-03-12 by github-actions[bot]

manosioa, 2021-08-11:
> So the problem appears to be indeed the Node Descriptor. When pairing the sensor it doesn't react to queries from the coordinator at all. But fortunately the sensor can be woke up to be responsive.
> 
> With the following workaround the missing data can be queried after pairing the sensor again:
> 
> ## Pairing sensor again to show the node in deCONZ GUI
> 1. Open the Phoscon App → Menu → Sensors;
> 2. Click "Add new Sensor" → Other;
> 3. Press the button in the small pin hole like 10 seconds until the sensor LED blinks blue;
> 4. Open / close the sensor a few times, the sensor should now be added and the node shown in deCONZ GUI.
> 
> Nothing new yet as some data is missing.
> 
> [image]
> 
> ## Retrieve missing descriptors in deCONZ GUI
> 1. Right click on the node and select **Read node descriptor**;
> 2. Quickly after that press the pin hole button on the sensor;
> 3. Right click on the node and select **Read simple descriptors**;
> 4. Quickly after that press the pin hole button on the sensor again.
> 
> [image]
> 
> When everything is queried it should look like in the last screenshot and the sensor node survives a deCONZ reboot.
> Note you can repeat the last steps if it doesn't work the first time.

This SOLVED the issue!
Thank you SO MUCH!

iroQuai, 2022-02-15:
I'm not really sure: if I use this workaround, does it work with ZHA afterwards too?