dresden-elektronik/deconz-rest-plugin #3529: Debug end device losing connection

dergreg, 2020-10-28:
## Describe the question or issue you are having

I have a Xiaomi Aqara Door Contact sensor mounted on my garage door, which used to work flawlessly the last few months, but suddenly stopped working a few weeks ago (didn't notice). I updated deCONZ two days ago to the latest stable version and re-paired the sensor.

However, it doesn't seem to be able to keep a connection established. After pairing, it was connected to the corrdinator, but lost connection in the garage.

What can be done to debug the issues? Distance wise, it should be able to connect to the coordinator and the "Osram Diele Repeater".

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi 3B
 - Running method: Raspbian
 - Firmware version: 26390500
 - deCONZ version: (2.05.84
 - Device: RaspBee I
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

I can provide some depending on what you think is relevant for drubbing the issue.

## Additional context

timovos, 2020-10-28:
I'm having the same issues. Xiaomi Aqara door sensor is always fighting with the zigbee network. Everytime I open DeCONZ, there's no line between the sensor and whatever parent.

Though, in my case, the sensor always works.

dergreg, 2020-11-18:
I repaired the sensor to the nearest repeater and verified it in the deconz application. Let's see if the connection remains

Smanar, 2020-11-18:
What is the osram device ?
For information Xiaomi have some issue with Osram plug.

dergreg, 2020-11-18:
It's an Osram plug indeed. What are the issues?

Smanar, 2020-11-18:
Xiaomi end device leave ^^
https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623

Long story, but the problem is from Xiaomi.

closed 2020-12-25 by stale[bot]