dresden-elektronik/deconz-rest-plugin #3531: Empty list of devices

MeyerDaniel, 2020-10-28:
## Describe the bug
I bought a ConBee II device and followed the instruction within the following document to get it to work with Wireshark: https://gzhls.at/blob/ldb/d/6/5/6/b3b2aa3e919352ec8d76c5d9ac1995a49917.pdf
For some reasons the list of devices when starting ZShark is empty.

## Steps to reproduce the behavior
  1. Plug in ConBee and let Windows 10 install the drivers.
  2. Download and install ZShark from here https://www.phoscon.de/de/conbee/software#zshark
  3. Restart Windows 10
  4. Start ZShark
[image]

## Expected behavior
The device should appear in the list as described in the referenced manual
## Screenshots

## Environment
Windows 10
Zshark V1_02
ConBee II

## deCONZ Logs
Nothing

## Additional context
Nothing

Smanar, 2020-10-28:
Hello,
From my memory Zshark work only (yet) for conbee and raspbee 1.

MeyerDaniel, 2020-10-29:
Hi
Many thanks for your response. This is a kind of bad news since according to the documentation it should work. You also get only the second edition offered when you want to buy one. So, is the support for ConBee II on your roadmap? I really regret that, having a useless device at the moment.

Regards

Mimiix, 2020-10-29:
@MeyerDaniel 

I've just spoke with Manuel. This is very old documentation which is not easily obtainable anymore. As far as i know, @ChrisHae is starting to work on this in the next few weeks. I am waiting for this too, so don't worry i'll be pushing 😉

closed 2020-12-19 by stale[bot]