dresden-elektronik/deconz-rest-plugin #3532: Bticino L4027C (Legrand)

dpmworld, 2020-10-28:
## Device

- Product name: Bticino Living Light L4027C
- Manufacturer:  Legrand
- Model identifier:  Shutter SW with level control
- Device type : Please remove all unrelated device types. 
  - Other: Window Cover

## Screenshots
[image]

[image]

[image]

[image]

Smanar, 2020-10-29:
Thx, easier for me than the discord for me ^^

So I have added the device in bind list, for position report, and used the same trick than for the other legrand device. The other device use too a whitelist for sensor, but it s useless for me.

To test it, you need an Unix OS (if you have one ?), and take a look here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only with a modification for step 1

```
git clone --branch legrand https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

dpmworld, 2020-10-29:
Hi Smanar, thenk you for taking it over so quickly.
I'm running deConz within a docker container, so I have to understand how to apply changes....

Yes the module does have a led in between the two buttons and I can make it blink while identifying it, but as you said the legrand cluster is all grayed out after trying to read it.

Smanar, 2020-10-30:
You haven't Unix to make test ? I can make the PR in blind mode, but without warranty.

for the LED I have find that
>Some devices requires have the functionality on the 3rd attribute and not 1rst (BTicino sub-brand)

So can you tried to search in the xml file
```
				<attribute id="0x0002" type="bool" name="Option 3" required="m" access="rw" default="0">
					<description>Option 2</description>
				</attribute>
```

and add just after
```

				<attribute id="0x0003" type="bool" name="Option 4" required="m" access="rw" default="0">
					<description>Option 4</description>
				</attribute>
```

dpmworld, 2020-11-09:
Hi Smanar, I'm setting up an ubuntu test machine in order to compile and test. Sry I took a while ;)
I'll let you know

dpmworld, 2020-11-09:
Ok, It seems everything is working fine.
I can use on and off commands as well as the %
The only problem is that I do not know how to calibrate it, so % is not working really, but I can use it to open/close the cover.

Here are the Windows Cover attributes:
[image]

[image]

[image]

[image]

Is there any place where I can find their meaning?

fdpm02, 2020-11-09:
Hi @Smanar 
just tested the change in the global.xml
[image]

[image]
All the 4 options are grayed out

Smanar, 2020-11-09:
@dpmworld this device don't support calibration, from the documentation it s just a switch, like the one from Legrand. Or you have different documentation than me ?
For lift you can use only 0% and 100 %

@fdpm02 arf, bad luck, will check if I found other value for different btcino devices to compare.

fdpm02, 2020-11-09:
@Smanar I suppose you know this site better than me: https://developer.legrand.com/documentation/product-cluster-list/
but according to https://developer.legrand.com/documentation/compatible-product/ these are "Roller Shutter Switch GEN2 (with Level Detection)" and using with the original gateway you can configure %.
Unfortunately I cannot find more info on cluster IDs and parameters...
I really don't understand why they do not release proper documentation: support on products like deConz would allow them to sell more and more.

BTW, thx for your time and support!

Smanar, 2020-11-09:
Ha nice, but haven't see this one yet.
But to work this kind of device don't depend of switch but shutter, I haven't the device in my hand but I think yours is just a switch that send the current in the "up" wire or the "down" wire, and it s the "limit switch" integrated in the shutter that cut the power when you reach the limit.
It s not the same technology for set level feature, or some devices mimic this feature using a timer.

From netatmo size, they are rather "open" for me, I have already ask them some informations, and for exemple, we will probably have their firmware to donwload on their site.

fdpm02, 2020-11-09:
This is from file "Roller-Shutter-Switch-GEN2-with-Level-Detection-ref-list.xlsx":
[image]
I think these are supposed to be the same device.

Smanar, 2020-11-09:
But I don't find thoses devices, not available in Legrand site, and on Btcino one I can find for exemple RG4027 (without the C) but it s not a connected device.

closed 2020-11-15 by manup

testorp, 2020-12-09:
[91735.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/5665423/91735.pdf)

attached a list of all devices currently available from Bticino in the Living Now and Living Light series

Smanar, 2020-12-11:
@dpmworld I have another user that have the device working with lift reversed.
Not the same version than you, It  just to be sure yours is still working with last deconz version ?

He have "Sw Build ID" = 16 and you 0B if I m right.

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3876

Smanar, 2021-02-24:
Hello, I have some question about this device (the L4027C)
Someone have already see it with LED "enabled on light", on deconz, the attribute to change this setting is grayed, mean not available.
Or this device use another attribute > 0x0002 ?

eternum79, 2022-06-27:
has anyone managed to get the percentage to work?

Smanar, 2022-06-27:
You have the long story here https://github.com/home-assistant/core/issues/71691

Have just see you on this issue, will continue on it ^^

lambu76, 2022-09-19:
hI all.
I read one more time again this post and the other mentioned, but I'm not getting if we have now a solution for managing percentage or not.

thx

lambu76, 2022-09-19:
Hi Smanar,
my problem is that I can only use Open/Close/Stop.

I cannot set any precentage.

I remember in the past (begin 2021) it was not possible to manage the percentage w/o Bticino/Legrand gateway, but, if I didn't misunderstood, it seems now there is a workaround....but I'm not getting which is and how to do it.

Smanar, 2022-09-19:
Ok I remember all now.
I have put a hack in the code for old version of this hardware.

So first bad new I go in holiday today for 1 week so can't do something during this time.
I m seing 2 solutions
1 - Something to try here https://github.com/home-assistant/core/issues/71691#issuecomment-1148902946
It' s a new code that permit to use DDF for covering, you have a DDF to test too
2 - Correct the code with bypassing the hack, but not sure manup will validate it

For the solution 2
Need to replace in code
```

            // Legrand invert bri and don't support other value than 0
            bool bStatus = false;
            uint nHex = taskRef.lightNode->swBuildId().toUInt(&bStatus, 16);
            if (bStatus && nHex < 28)
            {
                targetLiftZigBee = targetLift == 0 ? 100 : 0;
            }
            else 
            {
                targetLiftZigBee = targetLift == 100 ? 100 : 0;
            }
```

By
```

            // Legrand invert bri and don't support other value than 0
            bool bStatus = false;
            uint nHex = taskRef.lightNode->swBuildId().toUInt(&bStatus, 16);
            if (bStatus && nHex < 28)
            {
                targetLiftZigBee = targetLift == 0 ? 100 : 0;
            }
            else if (bStatus && nHex < INSERT_YOUR_SWBUILD_ID_HERE)
            {
                targetLiftZigBee = targetLift == 100 ? 100 : 0;
            }
```

But you will need to compile the code to test ir too.

lambu76, 2022-09-19:
You mentioned "old version of this hardware".

does it mean that BTicino/Legrand released a new version of HW even if the product code was left unchanged ?

thx

Smanar, 2022-09-19:
I think it's the device with the more changements during is life, there is so much (too much) in the deconz core for it.

I think they can have improved the hardware , from some resellers the switch without neutral have been improved too, the first versions was realy weak.

But for the covering, can be just a firmware upgrade, on the code I m checking the SWBuild written on the firmware, it s the version written on the firmware. I had issue with this device because some users with the same device have the side reversed and not other, the side change according to some version, if you update the device firmware, it can reverse the side on HA.

lambu76, 2022-09-19:
Can you Plesee let me know which is the properties to check if hw version is the old one or not?

I could try to buy a new one and check if it works fine for me.

My current devices are around 2 years old

Smanar, 2022-09-19:
>My current devices are around 2 years old

Hurr, make the test using the GUI, Go in the cluster 0x0102, use the command "Go To Lift Value", if this command don't work (or the covering go only 100% or 0 %), changing the code will do nothing for you.

The user with the position feature working have the SWBuildid = 0x002b https://github.com/home-assistant/core/issues/71691

lambu76, 2022-09-19:
All my devices have Firmware: 00b, if I look inside the Home Assistant Device Info panel.

If I check on deconz app in Basic cluster I can see SW Build ID = 00b.

I will try to buy on Amazon (and in case return it) a new one for test.

On Amazon it's pretty impossibile to find a gateway

eternum79, 2022-09-19:
> I tried command "Go To Lift Value" and I get back always the error "UNSUP_CLUSTER_COMMAND".
> 
> All my devices have Firmware: 00b, if I look inside the Home Assistant Device Info panel.
> 
> If I check on deconz app in Basic cluster I can see SW Build ID = 00b.
> 
> I will try to buy on Amazon (and in case return it) a new one for test.
> 
> On Amazon it's pretty impossibile to find a gateway

My firmware version is 001C

Smanar, 2022-09-19:
0x01C is 28

>if (bStatus && nHex < 28)

It's exactly the last one tested in the code (and this code is realy old), I think you don't have the position feature ?

It's possible too all devices sell ATM need the firmware update to support the position feature, so bying a new one will not help ..

lambu76, 2022-09-20:
So basically better try to find a gateway in order to update last firmawre our current devices instead hoping to find last firmware in a new device....correct?

lambu76, 2022-09-22:
Hi all,
I try to buy a Bticino kit with 3 L4027C and a gateway wifi plug in.

I tried to connect the L4027C to my zigbee network and I discovered that firmware preloaded is 0x01C, so v 28....and as expected if I try to use command "Go To Lift Value" I obtain the error "UNSUP_CLUSTER_COMMAND".

Then I reset this device and I tried to configure by using the BTIcino Gateway.

Once connected I checked the options of my L4027C and I found it can work in two different operational mode (I don't know the label in english...I'm using in italian)

- "Livello apertura" -> it can control the opening percentage.
- "Su/Giù/Stop" -> "Up/Down/Stop"

So my question are:

1) how can I update to latest firmware 001C ? I cannot find any option for checking new firmware
2) I found the option to switchon the Blu light and I enabled....is it possibile to mantain in my conbee nnetwork once I will disconnect from Bticino gateway and reconnect to conbee ?

thx

lambu76, 2022-09-25:
I foudn the way to udapte to last firmware....simply doing anything....just leave gateway and device swithced on and connected to Internet...in 2/3 hours the L4027C has been autoamtically updated to latest firmware 43.

Toimorrow I will try to connect to my Zigbee network.

Smanar, 2022-09-27:
And just by curiosity, are you skilled enouht to compile the code to test code modification if needed ?
I will give you all command to use, but you need a linux OS and other thing than the HA OS ?

lambu76, 2022-09-27:
unfortunately not

Smanar, 2022-09-27:
>I saw that there is a calibration option

It' s a on/off feature ? Can be too just a setting memorised on the application ?
If you have the GUI there is already some "calibration" command available in the GUI, I m searching on google but I don't find information on wich one is used by Legrand.

lambu76, 2022-09-27:
I'm a little bit confused now. 
So I will try to restart from my initial need.

L4027C can manage the percentage of opening/closing shutter by using the Legrand/BTicino gateway.

Is there an AUTOMATIC way to do the same if the L4027C is connected to Zigbee network managed by Deconz/Conbee in HASSIO ?

I wrote AUTOMATIC because for sure there is some tricks to reach the same goal by using some manual scripts/automation.

Smanar, 2022-09-27:
Nope because of the hack described here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3532#issuecomment-1251038883

Old version don't support it, so it s disabled on the code.
I think now your device support it, as you have the last firmware, can test using deconz (the application)
But the code need to be updated, the patch is fast and can be done in blind mode (without testing), but you know PR validation are not fast, so if t doesn't work need to wait for a next one.

lambu76, 2022-10-04:
I finally got the time to check my new L4027C connected to Zigbee network managed. by Deconz.

I can see, obviously, that the firmaware is the 02c as expected.

[image]

But if I try to use the function "Go To Lift Value" and I still get back the error "UNSUP_CLUSTER_COMMAND".

[image]

Any idea ?

Smanar, 2022-10-04:
Strange because 0x2C is the maximum I have see.
But on the doc here https://catalogue.bticino.com/app/webroot/low_res/518668_518688_LE11386_AB.pdf I can see 

> Attenzione: Il tempo di chiusura del contatto è di 2 minuti sia per apertura che per chiusura (non modificabile)
> Warning: Both the contact opening and closing time is 2 minutes (cannot be modified)
> Attention : Le temps de fermeture du contact est de 2 minutes, aussi bien pour ouverture que pour fermeture (non modifiable)
> Achtung: Die Schließzeit des Kontakts beträgt sowohl beim Öffnen als auch beim Schließen 2 Minuten (nicht veränderbar)
> Atención: El tiempo de cierre del contacto es 2 minutos para la apertura y para la cierre (no es modificable)
> Opgelet: Het contact heeft een sluittijd van 2 minuten, zowel tijdens het openen als het sluiten (deze tijd kan niet worden gewijzigd)

So I have read again the issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3876 and seem he have the K4027C version, sorry.

So it seem there is 2 devices, with similar look, same manufacture name, same model id, but differents features.

lambu76, 2022-10-04:
K4027c is apparently the same device w/o the front cap and it's used into new product lines.

Said so...is it anything you or me or we can do rather than using that K device?

Smanar, 2022-10-04:
0x2C is the highter firmware I know for this device, and yout model id is "Shutter SW with level control", so I realy don't understand why the level control is not working ....

On z2m we can read for this device

>         // Newer firmwares (e.g. 001f) Does support partial position reporting
>         // Old firmware of this device provides only three values: 0, 100 and 50, 50 means an idefinite position between 1 and 99.
>         // If you have an old Firmware set no_position_support to true
>         // https://github.com/Koenkk/zigbee-herdsman-converters/pull/2214 - 1st very basic support
>         zigbeeModel: [' Shutter SW with level control\u0000'],
>         model: 'K4027C/L4027C/N4027C/NT4027C',

And they are using the goToLiftPercentage request.

So it's strange ....

lambu76, 2022-10-05:
I tried "Go to Lift Percentage" with my actual L4027C installed (with fw 01b) and the new L4027C (with fw 02c).

The behavior is the same

- if shutter is 100% opened any value inserted lower than 100 will start the closing till 0%, despite the value
- if shutter is 0% opened (so closed) any value inserted greater than 0% will star the opening till 100%, despite the value

The question is .... for sure on new device what is missing is the calibration procedure that Legrand/BTcino gateway propose to execute at first usage.
How can perform the calibration in Homeassitant or Deconz/Phoscon ?

thx

Smanar, 2022-10-05:
You realy have the same reaction than device without the set position, only 0% , 100% and 50% is working.
I don't see the "calibration" command on other zigbee project, can you take a look in deconz on attribute "0x0017" on cluster 0x0102 ?

```
          <attribute id="0x0017" name="Mode" type="bmp8" default="0x0000" access="rw" required="o">
            <value name="Reversed" value="0"></value>
            <value name="Calibration Mode" value="1"></value>
            <value name="Maintenance Mode" value="2"></value>
            <value name="LED feedback" value="3"></value>
          </attribute>
```

lambu76, 2022-10-05:
this is the full list of attributes read from that cluster.

[image]
[image]

Smanar, 2022-10-06:
For tuya it s a value in second for 0xF003
But strange Legrand use this one

Attribute 0xF001 is just 1 / 0 for on / off, select the value, you will have a popup with more information.

Smanar, 2022-10-06:
I m sorry but I have checked all zigbee project I know and nothing about a calibration for Legrand.
I will ask to pipiche if he have an idea, but I don't have any way to explore.

lambu76, 2022-10-06:
Meanwhile you are checking if pipeche has any good news for me/us, I was able to find (thx Amazon free return of goods) a real  K4027C and fortunately it was sold with 02c firmware.

[image]

I tried the command "Go to Lift value" and I have same error "UNSUP_CLUSTER_COMMAND"

Same behavior also if I try to set some values in cluster "0102" for Calibrating values.

pipiche38, 2022-10-07:
From what I understood the calibration is done automatically when starting the device. Looks like nothing to configure and no hidden attribute/cluster.
If it doesn't work , the suggestions are:

Reset the device, it should start the calibration.

Smanar, 2022-10-07:
WTF what is this "Product code" on your capture ?

On this issue, the user have a firmware 0x02B and it seem his device have the setlevel featuyre.

```
2022-05-17 20:58:11 DEBUG (MainThread) [pydeconz.websocket] {"attr":{"id":"15","lastannounced":null,"lastseen":"2022-05-17T18:58Z","manufacturername":"Legrand","modelid":"Shutter SW with level control","name":"W
indow covering device 15","swversion":"002b","type":"Window covering device","uniqueid":"00:04:74:00:00:7d:ff:10-01"},"e":"changed","id":"15","r":"lights","t":"event","uniqueid":"00:04:74:00:00:7d:ff:10-01"}
```

> 
in deconz i can control the position of the cover with "go to lift percentage" command, and the current position is read correctly in the attributes of the device

Smanar, 2022-10-07:
Ok so pipiche have asked Legrand

The NLV need to work.
The calibration start when you power on the device, else need to press the "up" and the "down" during 3s up the led bink in orange.
Better to do that before inclusing the device.

Personnal notice :
Will be hard for us to that before the inclusion, because we need to reset the device to include it, and I think it reset the calibration too.

lambu76, 2022-10-07:
Guys...you are all super.

I will try to reset and check.

Regarding short press it was added now exactly in this way...after a reset...Quick click on reset button, led blinking Green and I started de pairing procedure on deconz.

I will keep you posted

lambu76, 2022-10-07:
Hi all.

I have some good news and probably one bad/strange news

### GOOD NEWS
I was able to start the calibration w/o resetting the device. I jsut pressed long UP/DOWN button, led start blinkinh orange and calibration mode started.
Once compelted the calibration (3/4 minutes with 3 full up/down of shutter) I'm able to use the command "GO to lift percentage" properly...0%, 100%, 50% 25% and so on....great

### BAD NEWS
Even if the UP/DOWN button trigger the expected and proper movment of the motor, the percentage read in HomeAssistant if different from the percentage used in the command.

I mean...

- if I use "Go to lift percentage" 25%, the shutter is moving to that percentage, but in H.A. the attribute current_position is 75
- similarly if I ask to to go 0% in Deconz, in H.A. I can see 100
- - similarly if I ask to to go 100% in Deconz, in H.A. I can see 0

it seems that Deconz consider open what if closed for H.A. and viceversa,

Any idea ?

Smanar, 2022-10-07:
Ok, so first @pipiche38 for the tips ^^. I think you are the first one to give it.

Now it's deconz turn.
There is a heck in deconz because this device initialy don't support this request, so fisrt need to remove it, so now you are in this situation too https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3532#issuecomment-1251038883

So choose your weapon ^^

lambu76, 2022-10-27:
> Ok, so first @pipiche38 for the tips ^^. I think you are the first one to give it.
> 
> Now it's deconz turn. There is a hack in deconz because this device initialy don't support this request, so fisrt need to remove it, so now you are in this situation too [#3532 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3532#issuecomment-1251038883)
> 
> So choose your weapon ^^

Hi all, sorry for the late response, but a couple of day after my HASSIo stopped recognized my Zigbee dongle due an update un DSM Synology where I was runnnig the VM with HASSIO....so I struggled looking a solution and then I decided to reinstall temporary HASSIO on a Raspberry Pi 4, waiting a solution...if it will ever arrive.

Said so.

I read the posts mentioned, but I'm not so confident to be able to follow those steps due my poor experince with LInux world.

My question now is the following.

Is there any chance that sooner or later the fix suggested will be officially implemented into Deconz firmware ?

Exactly today I updated it into HA to following realease, but nothing is changed for this device.

H.A. addon v. 6.17.0, with deCONZ to 2.19.1

Smanar, 2022-10-28:
I need someone recompile the code with 1 solution or the other to test it for I make a PR

lambu76, 2022-10-28:
who did support you with previous developments ?

Unfortuntaly I cannot help you on this point becasue I really don't have sufficient Linux skill to perfomr such task by myself.

Smanar, 2022-10-29:
I can give you the command line to use.
But too complicated on docker, you need a normal OS

lambu76, 2022-10-29:
let's try....

since I have to create a new VM, which linux flavour do you suggest ?

I can execute it within HYPERV on my laptop or in VMM within Diskstation..

I would suggest on HYPERV since currently Deconz stick is not usable on HASSIO vm running on Diskstation due last update of DSM from Synology.

Smanar, 2022-10-29:
Lol, you are more skilled than me for that.

But on HA you can use a distant deconz server, so you can HA on a machine and deconz on another.
And can make the tries using direclty the API with command line, so no need HA.

We will try the fast patch, and if you have time the second one.

So first need to create a simple OS and installing deconz with the conbee (for information, xiaomi/aqara device don't like to be without the coordinator too long time, so be fast for tests if you switch the conbee, don't worry you have hours)

Include the device in deconz.

For the API https://dresden-elektronik.github.io/deconz-rest-doc/
Remember you can find information in phoscon/help/api information

PORT and IP are the same used than for phoscon
ID is the device id

Get an API key
`curl http://IP:PORT/api/ -X POST -d '{"devicetype": "TestKey"}'`

Try the covering using API
curl -H 'Content-Type: application/json' -X PUT -d '{"lift": 10}' http://IP:PORT/api/APIKEY/light/ID/state

The device will go to 100% or 0 %, it's the actual issue.

Here the code I will use 

```
            // Legrand invert bri and don't support other value than 0
            bool bStatus = false;
            uint nHex = taskRef.lightNode->swBuildId().toUInt(&bStatus, 16);
            if (bStatus && nHex < 28)
            {
                targetLiftZigBee = targetLift == 0 ? 100 : 0;
            }
            else if (bStatus && nHex < 44) // Do nothing for last version
            {
                targetLiftZigBee = targetLift == 100 ? 100 : 0;
            }
            else // Do nothing for last version > 43
            {
                targetLiftZigBee = targetLift;
            }
```

As your device have 0x2C as swbuild the code will be not used

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing
So for you, after having installed deconz (The code is based on realy last one, so need a recent version):

```
sudo apt install deconz-dev
git clone --branch legrand_test_7 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry, it can freeze during compilation, my pi 3 do, so remove "-j2"

It don't impact the zigbee database (no problem for you as I think it will be a clean/empty system), but can use a lot the CPU (at least the first time, compilation will be faster after).

Reload deconz and test again 
`curl -H 'Content-Type: application/json' -X PUT -d '{"lift": 10}' http://IP:PORT/api/APIKEY/light/ID/state`

medy2, 2023-01-24:
> ### GOOD NEWS
> I was able to start the calibration w/o resetting the device. I jsut pressed long UP/DOWN button, led start blinkinh orange and calibration mode started. Once compelted the calibration [...]

I entered calibration mode, the led started blinking orange, but no automatic movements happened. I manually performed several full up/down cycles an the "goto x %" now works. but...  orange light keep flashing. It seems it never exits from calib mode...
@lambu76 did you do something to return in "normal" mode, or simply waited?

I tried on another switch and it behaves the same.

lambu76, 2023-01-24:
I experienced same problem and what I did was to reset the switch and restart.

Smanar, 2023-01-25:
Remember it's firmware dependent.

lambu76, 2025-07-06:
> Remember it's firmware dependent.

Hi there,
Which is the current level of support for this device in this scenario?
Is the percentage working fine inside homeassistant interface?

[25 more replies]