dresden-elektronik/deconz-rest-plugin #3535: Receiving commands from Sunricher Zigbee directly with ConBee II

Wim-N, 2020-10-29:
## Describe the question or issue you are having

For an automation system I would like to receive commands from a Sunricher ZGRC-KEY-013 (firmware 2.5.3_r20) Zigbee remote control using a ConBee II (version 2.05.84 / 9/14/2020, firmware 26660700) on a Raspberry Pi 4 using Python 3. So I'm not using deCONZ.

I have found 'Serial protocol' version 1.14 (2019-05-25), but that seems to be a little outdated according to other topics. Do you have a newer version of the protocol?

Using deCONZ the remote control is linked to the ConBee II. When you open deCONZ both are visible and when you press a key a green line appears to the left off both connecting them.

Using some demo code I have written a Python program that can communicate with the ConBee II via the SLIP protocol and also full CRC support is implemented.

So far so good.

But when I press a key on the remote control the ConBee II sends a message and the program responds:

Rx: 29-10-2020 15:47:51.099 1C-03-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225
Rx: 29-10-2020 15:47:51.102 0E-04-00-07-00-AA-00
Rx: Device state changed [0x0E]: Status: Success, Network: connected, APSE-DATA indication, APSE-DATA request free slots
Tx: APS_DATA_INDICATION request
Tx: C0-04-04-00-07-00-00-00-F1-FF-C0
Rx: 29-10-2020 15:47:51.105 1C-05-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225
Rx: 29-10-2020 15:47:51.107 04-04-05-08-00-01-00-2A
Rx: Query Send Data State Response, short [0x04]; Status: Error
Tx: Read Received Data Request request
Tx: C0-17-04-00-08-00-01-DC-FF-C0
Rx: 29-10-2020 15:47:51.171 1C-05-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225
Rx: 29-10-2020 15:47:52.083 0E-06-00-07-00-AA-00
Rx: Device state changed [0x0E]: Status: Success, Network: connected, APSE-DATA indication, APSE-DATA request free slots
Tx: APS_DATA_INDICATION request
Tx: C0-04-06-00-07-00-00-00-EF-FF-C0
Rx: 29-10-2020 15:47:52.086 1C-07-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225
Rx: 29-10-2020 15:47:52.089 04-06-05-08-00-01-00-2A
Rx: Query Send Data State Response, short [0x04]; Status: Error
Tx: Read Received Data Request request
Tx: C0-17-06-00-08-00-01-DA-FF-C0
Rx: 29-10-2020 15:47:53.090 1C-07-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225
Rx: 29-10-2020 15:47:54.193 0E-08-00-07-00-AA-00
Rx: Device state changed [0x0E]: Status: Success, Network: connected, APSE-DATA indication, APSE-DATA request free slots
Tx: APS_DATA_INDICATION request
Tx: C0-04-08-00-07-00-00-00-ED-FF-C0
Rx: 29-10-2020 15:47:54.195 1C-09-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225
Rx: 29-10-2020 15:47:54.197 04-08-05-08-00-01-00-2A
Rx: Query Send Data State Response, short [0x04]; Status: Error
Tx: Read Received Data Request request
Tx: C0-17-08-00-08-00-01-D8-FF-C0
Rx: 29-10-2020 15:47:55.115 1C-09-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225
Rx: 29-10-2020 15:47:56.217 1C-0A-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225
Rx: 29-10-2020 15:47:57.137 1C-0B-00-0C-00-05-00-02-0F-8B-FF-E1
Rx: MAC Poll Notification [0x1C]: Status: Success, NWK: 8B-0F, LQI = 255, RSSI = 225

The ConBee II reports an error at the [0x04] message so something is going wrong. And whatever key I press, I cannot find some info about what key was pressed on the remote control.

Is there a description about the sequence of messages needed while communicating with the ConBee II to get some info about the key being pressed?

## Screenshots

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian
 - Firmware version: (26660700) ConBee II (version 2.05.84 / 9/14/2020, firmware 26660700)
 - deCONZ version: (2.05.84)
 - Device: ConBee II and Sunricher ZGRC-KEY-013 (firmware 2.5.3_r20) Zigbee remote control
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Wim-N, 2020-11-12:
Anyone, please.....

Wim-N, 2020-12-07:
Anyone?????????

Wim-N, 2021-01-07:
Anyone????

ebaauw, 2021-01-07:
You’re unlikely to get a response on a serial protocol question in the repository for the REST API plugin.  I’d try https://github.com/dresden-elektronik/deconz-serial-protocol/issues.

Wim-N, 2021-01-07:
Thanks, I've done that

closed 2021-01-07 by Wim-N