dresden-elektronik/deconz-rest-plugin #3536: Hue Motion Outdoor Sensor SML002 don't report lightlevel

net1957, 2020-10-29:
## Describe the bug
I use Jeedom and a ConbeeII key with Deconz and Phoscon App on a Debian Linux
I try to use a Hue Motion Outdoor Sensor SML002
Presence and temperature are working well but lightlevel, and lux are always 0

The Phoscon App display: **Lightlevel 0  Lux** and i get the same in Jeedom

## Steps to reproduce the behavior
Add the  Sensor and display the status in Phoscon and Jeedom

## Expected behavior
Lightlevel and lux  should be different from 0

## Screenshots

## Environment
  Phoscon GW version: 2.05.71 17/11/2019
  Conbee II 26660700

 - Host system: PC
 - Running method: (Ubuntu in VM With Jeedom)
 - Firmware version: (26660700)
 - deCONZ version: (2.05.71)
 - Device:  ConBee II)
 - Do you use an USB extension cable: (no) 
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

net1957, 2020-10-29:
what parts should I update ?

net1957, 2020-10-29:
update to 2.05.84 
and it's worse, no more temperature : 0 in jeedom and in phoscon
in jeedom, the received json is:
<
{
{
    "54": {
        "config": {
            "alert": "none",
            "battery": 100,
            "delay": 0,
            "ledindication": false,
            "on": true,
            "pending": [],
            "reachable": true,
            "sensitivity": 2,
            "sensitivitymax": 4,
            "usertest": false
        },
        "ep": 2,
        "etag": "1781492d12092c9fbc51e80d10c70e90",
        "lastseen": "2020-10-29T21:18Z",
        "manufacturername": "Philips",
        "modelid": "SML002",
        "name": "Presence 54",
        "state": {
            "lastupdated": "2020-10-29T21:18:09.296",
            "presence": true
        },
        "swversion": "6.1.0.25261",
        "type": "ZHAPresence",
        "uniqueid": "00:17:88:01:06:47:fa:8b-02-0406"
    },
    "55": {
        "config": {
            "alert": "none",
            "battery": 100,
            "ledindication": false,
            "offset": 0,
            "on": true,
            "pending": [],
            "reachable": true,
            "usertest": false
        },
        "ep": 2,
        "etag": "993d7dad3eb59dfa15c7aae7a531ba58",
        "lastseen": "2020-10-29T21:18Z",
        "manufacturername": "Philips",
        "modelid": "SML002",
        "name": "Temperature 55",
        "state": {
            "lastupdated": "none",
            "temperature": null
        },
        "swversion": "6.1.0.25261",
        "type": "ZHATemperature",
        "uniqueid": "00:17:88:01:06:47:fa:8b-02-0402"
    },
    "56": {
        "config": {
            "alert": "none",
            "battery": 100,
            "ledindication": false,
            "on": true,
            "pending": [],
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000,
            "usertest": false
        },
        "ep": 2,
        "etag": "993d7dad3eb59dfa15c7aae7a531ba58",
        "lastseen": "2020-10-29T21:18Z",
        "manufacturername": "Philips",
        "modelid": "SML002",
        "name": "LightLevel 56",
        "state": {
            "dark": null,
            "daylight": null,
            "lastupdated": "none",
            "lightlevel": null,
            "lux": null
        },
        "swversion": "6.1.0.25261",
        "type": "ZHALightLevel",
        "uniqueid": "00:17:88:01:06:47:fa:8b-02-0400"
    }
}
>

cllatser, 2020-10-30:
Let's see if the same thing will happen as with the indoor Hue, what updates the lux when it detects movement. I am interested in this sensor as well. 😀

net1957, 2020-10-30:
It didn't catch the lux even if the detected a motion.
Last night I reverted back to 2.05.71 without any change on the results (got motion, temperature but no lux)
but this morning it seems to work as expected. It's strange and I have no explanation!
Perhaps it needed some time ?

cllatser, 2020-10-30:
So, it is necessary to leave it for a while to finish the configuration. The same thing happened to me with the indoor sensor. We have no patience and we want everything instantaneous. 😀

bee-con, 2020-11-01:
@net1957: If you lose light settings again, you might also want to update the sensor's firmware. See https://www.philips-hue.com/en-hk/support/release-notes/accessories and https://github.com/dresden-elektronik/deconz-ota-plugin#hue-firmware. I had one of my two outdoor hue sensors that would regurlary lose light information, and acquire it back again one or more hours or even days later. That problem disappeared after updating the sensor's firmware. HTH

net1957, 2020-11-01:
@bee-con Did upgrade the sensor firmware to 6.1.1.27575 and awaiting lux and temperature information (motion is ok). I hope it will be OK tomorrow. Thanks for the tips. Regards

net1957, 2020-11-04:
It was ok next morning. :)

jacekk015, 2020-11-16:
@net1957 Wheere did you get the firmware for SML002??
There's only available download for SML001

net1957, 2020-11-17:
It's the same

I forgot the link but this forum give some information: https://community.jeedom.com/t/mise-a-jour-firmware-des-capteurs-possible/41429

found here: http://fds.dc1.philips.com/firmware/ZGB_100B_010D/1107323831/Sensor-ATmega_6.1.1.27575_0012.sbl-ota

closed 2020-12-22 by net1957