dresden-elektronik/deconz-rest-plugin #3540: Websocket alternative?

OlwinFroon, 2020-10-30:
## Describe the question or issue you are having

Is there any other way than the websocket service to receive sensor events?

## Screenshots

n/a

## Environment

- Raspberry Pi 2B w/ latest Raspbian, headless
- ConBee II (V2.05.86, FW 26660700) w/ 1m ext cable

## deCONZ Logs

n/a

## Additional context

So far I have
- two Tint lights
- a remote control for those
- a LEDVANCE smart plug
- a Sonoff ZBR3 (modified to act as a dry relay)
- a Mi door sensor
- two Mi temperature/humidity sensors
working, and it was a walk in the park.

First tests showed that the REST API is working perfectly, allowing to control lights, smart plugs and so on.

Frequently polling the API's sensors resource is fine, too, to get the current status quo, temperatures and such.

However, polling is not an option when it's about the door sensor or the remote.

So I need a way to get sensor events delivered to my central backend application, and all I found so far is the websocket option.

I have a hard time implementing a webservice connection from my PHP backend to my ZigPi, and I wonder if it would be feasible to have push notifications instead.

It could be the same JSON snippets that show up on the websocket, but the would be sent to a defined port of a specific IP-address/URI, that you'd define in the deCONZ configuration section.

Smanar, 2020-10-30:
I don't think sorry.
What is your central backend application ?

OlwinFroon, 2020-10-30:
Home made PHP/MySQL stuff on a Synology NAS.

Okay, if there's nothing else but the websocket thingy, can I turn this into a feature request?

Smanar, 2020-10-30:
You can, but not sure there is something more easier, universal, and native to use better than websocket for me.

Smanar, 2020-10-30:
lol, right, haven't tried on my side with php, I have checked fastly on google, there is some short code sample, but it s just the theory ^^

OlwinFroon, 2020-10-31:
Even if it works reliably, it would still be a major kludge. A real disgrace to the craft.

So here's my petition to add some other sensor event outlet to this otherwise very fine piece of software.

<3

MattL0, 2020-11-01:
+1.

Isn’t @manup talked about mqtt one time?

Smanar, 2020-11-01:
You found a mqtt client is easier to code in php than a websocket client ?

manup, 2020-11-01:
Server Sent Events (SSE) might be an alternative in future.

https://kevinchoppin.dev/blog/server-sent-events-in-php

OlwinFroon, 2020-11-01:
That's still polling, right? Not anywhere in the vicinity of real time.

IMO the only way to achieve that with a PHP backend is actively nudging it when something happens.

There's lots of 
  "Alexa, do this" ... pause ... pause ... pause ... aaaand it happens
out there.

Need to do better than that.

manup, 2020-11-01:
No like Websockets SSE provides also server side push — but only unidirectional. A good description can be found in the MDN https://developer.mozilla.org/en-US/docs/Web/API/Server-sent_events/Using_server-sent_events

OlwinFroon, 2020-11-09:
Ya, I saw stuff like that all over the web: a (PHP) websocket _server_ sending to a JS/browser _client_. 
Won't fly with a PHP _client_, though, because PHP simply can't 'listen' to events.

Long story short: The way I see it, there's two approaches to get this up and running _properly_:

The kludgy one would be "chinese whispers", something like a node.js websocket client that'd connect to the deCONZ websocket service and relay all events to my PHP backend.
It might even work, but ... really? C'mon.

The cool one would be where you'd configure a target IP/URI and port, and deCONZ would then send those JSON events via HTTP POST to the configured target instead of currently connected websocket clients.

OlwinFroon, 2020-12-04:
Another thought: what are the chances of adding a bit to the current rules mechanism?

I mean, it's almost all there: 
action.address: path to a light, group or scene resource
action.body: JSON encoded parameters
action.method: PUT, POST or DELETE

My problem would be solved if action.address could be a HTTP address...

OlwinFroon, 2020-12-16:
Huh, who would've known. 
I just wrote a rule with a HTTP URI as action,address ... and it worked!
Problem solved :}

Mimiix, 2020-12-16:
CaseClosed?

OlwinFroon, 2020-12-16:
Yup.

Mimiix, 2020-12-16:
Case closed!

closed 2020-12-16 by Mimiix